LiberTEM/LiberTEM #1071: Word wrap for table cells

uellue, 2021-07-22:
The table cells in https://libertem.github.io/LiberTEM/sample_datasets.html are currently not wrapping.

This could be an issue with the theme?

sk1p, 2021-07-26:
> This could be an issue with the theme?

This is generally an issue with tables - it's hard to have a working default styling that covers all cases in a sane way. I think we should instead just not use a table. Could a list work better maybe? Similar but simpler than our [funding section](https://libertem.github.io/LiberTEM/acknowledgments.html#funding)? 

That way, we don't have to fit long dataset names into the horizontal space, while still displaying the other columns...

uellue, 2021-08-17:
> Could a list work better maybe? Similar but simpler than our [funding section](https://libertem.github.io/LiberTEM/acknowledgments.html#funding)?

Sounds good, would also be much easier to edit in the source code. I've made it a "good first issue" now. :-)

PR #1082 (2021-08-17): Misc docs fixes
* Drive-by: Missed changelog entry, type hints
* Update UDF diagram with drawing by @sk1p
* Restructure sample datasets, closes #1071 
* Restructure UDF reference docs, refs #638  

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x / N/A] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
* [N/A] I have added/updated [... cut]
commit da5206c4 (2021-08-17): Replace table with paragraph and labeled list

This is much more friendly to edit in the source code and solves the wrod wrap issue.

Closes #1071
closed 2021-08-17 by uellue