LiberTEM/LiberTEM #1085: Open a Dataset via URL parameters and possibly embed in other web applications.

uellue, 2021-08-20:
Starting and accessing LiberTEM through JupyterHub and JupyterLab #1074 makes sure that LiberTEM runs as the correct Unix user. It delegates authentication to JupyterHub/Lab and authorization to the operating system via Unix users and file permissions.

Since the URL for LiberTEM for a specific user is predictable, such as https://example.com/user/weber/libertem/, we could link to datasets from an electronic lab notebook (ELN) or other web application. On the ELN side we'd just have to know the username, a storage path and possibly parameters. On LiberTEM side we just need to accept the storage path and dataset parameters somehow, for example through URL parameters.

Possibly that would even allow embedding the GUI resp. GUI elements?

uellue, 2021-08-24:
Moved to the 0.9 milestone since there is specific demand from University of Queensland users and it looks like a relatively simple feature with large benefit for users. :-)

sk1p, 2021-08-25:
No need to include the username in the URL, this works, too: https://your-jupyter-hub/user-redirect/libertem/

uellue, 2021-08-31:
Refs #867

uellue, 2021-09-28:
Integration work by @shabihsherjeel depends on this feature. If I understood correctly, the timeline is flexible for now? In any case, an important feature for the next release and should definitely be shortlisted. :-)

shabihsherjeel, 2021-09-29:
Yeah. I had the cli option setup on a local branch. It's nice to see this being implemented. I will keep an eye out for this and use it to run it in other projects that load LiberTEM. Thanks for letting me know. :)

uellue, 2021-12-06:
Discussion with @sk1p: We should also provide an interface to start a cluster or connect to one without user interaction for integrations like this.

uellue, 2021-12-06:
Moved to the 0.10 milestone so that we get the executor and Dask integration work out first.

uellue, 2024-05-16:
It seems like this is solved in #1518? For the data management at ER-C we figured that linking is better than embedding since it introduces less dependencies, is way more general and works nearly as well.

closed 2024-05-16 by uellue