LiberTEM/LiberTEM #1090: Possible confusion between current GUI parameters and notebook download

uellue, 2021-08-24:
Not sure how severe this issue is, but listing it here at least for documentation and discussion purposes.

## How to reproduce

* Add an analysis in the GUI with selectors or similar and run it
* Change the parameters/selectors in the GUI without re-running
* "Download" -> copy notebook and inspect the parameters for the analysis

## Observed behavior

The notebook contains the parameters from the last run

## Expected behavior

Not sure if a user would expect the current GUI parameters or the last run. In any case, it is an ambiguous situation.

## Possible solutions

* Warning or other indication of ambiguity/invalid state/requiring re-run, for example different icons or color change of buttons?
* Disable download until run?
* Run upon downloading?
* Ignore and leave as-is?

uellue, 2021-08-25:
Since CoM now (#1087) allows quick re-calculation without hitting "Apply", this is probably where my confusion came from.

sk1p, 2021-08-25:
Idea: style the apply button differently depending on whether the parameters are "dirty" or not

sk1p, 2022-01-26:
> The notebook contains the parameters from the last run

FYI with #1172 having landed, this situation changed, at least for the CoM analysis: parameters are updated as they are changed in the GUI, because the parameter guess should _definitely_ not based on the values from the last run (because sometimes, there even isn't a "last run" when performing the guess...). This could be adapted for the other analyses as well.