LiberTEM/LiberTEM #1093: Release 0.8

sk1p, 2021-08-25:
# Release checklist

Issues to be considered for this release:

* #1070 - finishing touches on jupyter integration? release on pypi?

## Before (using a release candidate package)

* [ ] Review open issues and pull requests
* [ ] Run full CI pipeline, including slow tests, on [Azure DevOps](https://dev.azure.com/LiberTEM/LiberTEM/_build?definitionId=3)
* [ ] Handle deprecation, search the code base for `DeprecationWarning`
      that are supposed to be removed in that release.
* [ ] GUI dependency update with `npm install`
* [ ] Review https://github.com/LiberTEM/LiberTEM/security/dependabot and update dependencies
* [ ] Full documentation review and update, including link check using
      ``sphinx-build -b linkcheck "docs/source" "docs/build/html"``
* [ ] Run complete test suite, including slow tests that are deactivated by default
      and tests that require sample files.
* [ ] Update the expected version in notes on changes, i.e. from `0.3.0.dev0`
      to `0.3.0` when releasing version 0.3.0.
* [ ] Update and review change log in `docs/source/changelog.rst`, merging
      snippets in `docs/source/changelog/*/` as appropriate.
* [ ] Update the JSON files in the ``packaging/`` folder with author and project information
* [ ] Edit `setup.cfg` to exclude flaky tests temporarily from release builds
* [ ] Create a release candidate using `scripts/release`. See `scripts/release --help` for details.
* [ ] `Confirm that wheel, tar.gz, and AppImage are built for the release candidate on
      GitHub <https://github.com/LiberTEM/LiberTEM/releases>`_
* [ ] Confirm that a new version with the most recent release candidate is created in the
      `Zenodo.org sandbox <https://sandbox.zenodo.org/record/367108>`_ that is ready for submission.
* [ ] Install release candidate packages in a clean environment
      (for example:
      `python -m pip install -i https://test.pypi.org/simple/ --extra-index-url https://pypi.org/simple 'libertem==0.2.0rc11'`)
* [ ] For the GUI-related items, open in an incognito window to start from a clean slate
* [ ] Correct version info displayed in info dialogue?
* [ ] Link check in version info dialogue
* [ ] Make sure you have test files of all supported types available
    * [ ] Include floats, ints, big endian, little endian, complex raw data
* [ ] Open each test file
    * [ ] Are parameters recognized correctly, as far as implemented?
    * [ ] Any bad default values?
    * [ ] Does the file open correctly?
    * [ ] Have a look at the dataset info dialogue. Reasonable values?
* [ ] Perform all analyses on each test file.
    * [ ] Does the result change when the input parameters are changed?
    * [ ] All display channels present and looking reasonable?
    * [ ] Reasonable performance?
    * [ ] Use pick mode.
* [ ] Re-open all the files
    * [ ] Are the files listed in "recent files"?
    * [ ] Are the parameters filled from the cache correctly?
* [ ] Try opening all file types with wrong parameters
    * [ ] Proper understandable error messages?
* [ ] Pick one file and confirm keyboard and mouse interaction for all analyses
    * [ ] Correct bounds check for keyboard and mouse?
* [ ] Check what happens when trying to open non-existent files or directories in the GUI.
    * [ ] Proper understandable error message?
    * [ ] Possible to continue working?
* [ ] Shut down libertem-server while analysis is running
    * [ ] Shut down within a few seconds?
    * [ ] All workers reaped?
* [ ] Check what happens when trying to open non-existent files by scripting.
    * [ ] Proper understandable error message? TODO automate?
* [ ] Check what happens when opening all file types with bad parameters by scripting
    * [ ] Proper understandable error message? TODO automate?
* [ ] Run libertem-server on Windows, connect to a remote dask cluster running on Linux,
  open all file types and perform an analysis for each file type.
* [ ] Use the GUI while a long-running analysis is running
    * [ ] Still usable, decent response times?
* [ ] Confirm that pull requests and issues are handled as intended, i.e. milestoned and merged
  in appropriate branch.

## After releasing on GitHub

* [ ] Confirm that all release packages are built and release notes are up-to-date
* [ ] Install release package
* [ ] Confirm correct version info
* [ ] confirm package upload to PyPI
* [ ] Publish new version on zenodo.org
* [ ] Update documentation with new links, if necessary
    * [ ] Add zenodo badge for the new release to Changelog page
* [ ] Send announcement message on mailing list
* [ ] Edit `setup.cfg` to include flaky tests again
* [ ] Bump version in master branch to next .dev0 (`./scripts/release bump v0.X.0.dev0 --commit`)
* [ ] Add to institutional publication databases
* [ ] Add the current LiberTEM version to [CVL](https://github.com/Chasdfracterisation-Virtual-Laboratory/CharacterisationVL-Software>) - add both the singularity and the .desktop file!

uellue, 2021-08-31:
- [ ] If possible, clarify which format we support in #805 and #1077. Refs https://github.com/LiberTEM/LiberTEM/pull/1077#issuecomment-909383527
   - [ ] Amend changelog
   - [ ] Include in SEQ documentation

uellue, 2021-09-07:
During release testing, I ran into the memory issues again and couldn't even complete half the GUI tests on a machine with 32 GB RAM. With that outcome I'd tend to classify these issues as a blocker for the release, since objectively LiberTEM is not able to complete the release checklist in its current state. I'm adding the corresponding Issues to the list above.

uellue, 2021-09-07:
Since the Jupyter integration and CoM are nice features that people are waiting for and the memory issues are not new, we could also release 0.8 with a note on known bugs, and fast-track a 0.8.1 release that adresses the memory issues. What do you think?

sk1p, 2021-09-27:
As discussed, I'll have an in-depth look at the memory problem, and if possible fix this for 0.8

sk1p, 2021-10-04:
For post-release work, see also https://github.com/LiberTEM/LiberTEM-jupyter-proxy/issues/6

closed 2021-10-07 by uellue