LiberTEM/LiberTEM #1112: Integrate CoM parameter guess in GUI

uellue, 2021-09-20:
Follow-up to #1111: Integrate guess to GUI, for example with an "auto" button.

It turns out that finding the correct alignment between scan and detector is critical for correct results, in particular with ptychography, and at the same time error-prone.

With this function one can quickly check if the parameters distributed with a dataset make sense and/or find a good guess if the required metadata is not available.

sk1p, 2021-12-14:
How this could work: 

1) User presses button in the GUI
2) GUI makes a "RPC" to the Web API
3) Either take existing CoM results, or run CoM with default parameters on the data (should we just run without radial cut-off mask, or do we need to guess that one too? Maybe run with a very large radial mask)
4) Parameter guess is serialized to JSON and sent as RPC results (if we have to re-run, include the results 

I think this can be prototyped quite easily - and can be used to gain use-cases and examples for this aspect to be integrated into the plugin API later.

sk1p, 2021-12-14:
About point 3) - for running a new CoM analysis from the "rpc method", the job running needs to be refactored a bit:

- The "job running" part of the web API needs to be put into a service object which is accessible from different HTTP handlers
- The "rpc method" then needs to have access to this service object, and also access to all data relevant to the compound analysis
- It needs to be able to handle creating jobs without the usual GUI interaction (so: server-side UUID generation, then the GUI needs to be informed that there actually is a new job running)
- On the GUI side, the state handling for the parameters might need slight changes
  - For now, we just implement this for the concrete use case, I think.

PR #1172 (2022-01-12): Integrate CoM parameter guess into web GUI
Fixes #1112 - adds a "Guess parameters" button to the CoM analysis.

Also includes a bunch of new type annotations. @uellue: most important part to review is the RPC itself, in `libertem.analysis.com`, and maybe trying this on a few data sets.

## TODO

- [x] Sometimes, the guess still diverges - need to find out if this is in the new code, or in the guess itself
- [x] Add a few integration tests for the new API endpoint
- [x] Run analysis if we don't have a result yet
- [x] Add feedback: spinner while the API is still guessing

## Contributor Checkli [... cut]
closed 2022-01-12 by uellue