LiberTEM/LiberTEM #1120: Autodetection of SEQ dark, gain and dead pixel maps apparently not working with our reference dataset

uellue, 2021-10-01:
## How to reproduce

* Open our reference SEQ dataset like this and inspect the corrections:

```python

params = {'path': '/Users/weber/LargeData/ER-C-1/groups/data_science/data/reference/SEQ/2020-05-04 8x8 8fps 128x128 streampix 8.seq.seq',
          'nav_shape': [8, 8], 'sig_shape': (128, 128), 'sync_offset': 0}

ds = ctx.load("SEQ", **params)

ds.get_correction_data().have_corrections()

# False
```

Listing of `/Users/weber/LargeData/ER-C-1/groups/data_science/data/reference/SEQ/`:

```
2020-05-04 8x8 8fps 128x128 streampix 8.seq.Config.Metadata.xml
2020-05-04 8x8 8fps 128x128 streampix 8.seq.dark.mrc
2020-05-04 8x8 8fps 128x128 streampix 8.seq.gain.mrc
2020-05-04 8x8 8fps 128x128 streampix 8.seq.metadata
2020-05-04 8x8 8fps 128x128 streampix 8.seq.seq
```

The GUI also shows patterns consistent with uncorrected gain and dark.

## What should happen

Following [our documentation](https://libertem.github.io/LiberTEM/reference/dataset.html#seq), the correction data should be loaded from the available files.

uellue, 2021-10-01:
In our data management reference folder, the file is named `*.seq.seq`, while in the folder for unit tests it is called `*.seq`. Both file sets have exactly the same size each, so I figure it is the same source? @sk1p I'll just assume that the file got renamed by accident? Can you perhaps fix that in the unit test data and rename it `default.seq`-> `default.seq.seq` since I don't have write permission there? I'll fix the discovery code.

sk1p, 2021-10-04:
> Can you perhaps fix that in the unit test data and rename it default.seq-> default.seq.seq since I don't have write permission there? I'll fix the discovery code.

I'm not convinced that this is the right course of action - the other SEQ dataset I have available doesn't have the ".seq.seq" ending, so I think `2020-05-04 8x8 8fps 128x128 streampix 8.seq.seq` is the one that is wrong.

uellue, 2021-10-04:
Maybe I can just make it work with both `.seq` and `.seq seq` then? 😊

sk1p, 2021-10-04:
> Maybe I can just make it work with both `.seq` and `.seq.seq` then? 

I guess that's in line with the robustness principle ("be conservative in what you do, be liberal in what you accept from others"), even if we can't support every possible way people rename their files. I guess in the long run, #807 is the correct way to fix this.

PR #1121 (2021-10-04): Support both `.seq` and `.seq.seq` as extension for the main SEQ file
to find `.seq.*` with correction data.

Closes #1120

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [N/A] I have added/updated documentation for all user-facing changes
* [N/A] I have added/updated test cases
* [N/A] I have included the [rebuilt production build of the clien [... cut]
commit 66501da8 (2021-10-04): Support both `.seq` and `.seq.seq` as extension for the main SEQ file
to find `.seq.*` with correction data.

Closes #1120
closed 2021-10-04 by uellue