LiberTEM/LiberTEM #1134: Documentation for offline installation

uellue, 2021-10-15:
Since microscope computers are often behind a firewall without internet access, we should document how to install LiberTEM packages on such a system.

Rough outline:

* Install Conda from stand-alone installer on target system
* Create empty environment in offline mode on target system
* Create a staging environment with the same OS, architecture and Python version on a system with internet access
* `pip wheel ...` to create wheels
  - Additional info on how to install a development version from GitHub
* Copy wheels to target system
* Install wheels with pip in offline mode

uellue, 2022-03-07:
Bumped to next release since it keeps cropping up frequently, and even for @sk1p and I it is annoying to figure out each time.

sk1p, 2024-05-16:
Fixed by #1627

closed 2024-05-16 by sk1p