LiberTEM/LiberTEM #1155: Abstract for SEEM 2021

uellue, 2021-11-18:
An invited talk at the Sino-European Early Career Researchers Workshop on Emerging Techniques and Applications in Electron Microscopy where I'll be presenting LiberTEM. :-)

Credited are the LiberTEM creators, i.e. you. Looking forward to your feedback! The deadline for submission is tomorrow, btw.

The conference is free to participate and has several interesting talks and speakers, see below!

[Abstract-Weber_Sino-European-Workshop-2021.docx](https://github.com/LiberTEM/LiberTEM/files/7563613/Abstract-Weber_Sino-European-Workshop-2021.docx)

## More info on the conference

[Poster_SEEM-2021_20211104.pdf](https://github.com/LiberTEM/LiberTEM/files/7563625/Poster_SEEM-2021_20211104.pdf)

[Program_SEEM-2021_20211104.pdf](https://github.com/LiberTEM/LiberTEM/files/7563632/Program_SEEM-2021_20211104.pdf)

## Free registration

https://fau.zoom.us/webinar/register/WN_AzS0A3CcQNy4Z7K1GEACGw

AnandBaburajan, 2021-11-18:
Congrats everyone and best of luck @uellue! Looking forward to attending the sessions!

uellue, 2021-12-06:
Here's also the slides for the talk: https://iffcloud.fz-juelich.de/s/qkEwEDc7saT53Db

Looking forward to your feedback! :-)

AnandBaburajan, 2021-12-09:
I really liked the session, @uellue. Seeing you so happy and excited while presenting LiberTEM made me happy too. : )

sk1p, 2021-12-14:
Thank you for presenting LiberTEM @ SEEM 2021, @uellue! As there have been no further comments, I'm going to close the issue.

closed 2021-12-14 by sk1p