LiberTEM/LiberTEM #1166: cupy v10 support?

sk1p, 2021-12-09:
Of special note in the [upgrade guide](https://docs.cupy.dev/en/latest/upgrade.html) is a [change in `cupy.cuda.Device` behavior](https://docs.cupy.dev/en/latest/upgrade.html#change-in-cupy-cuda-device-behavior) - does this affect our cupy use?

uellue, 2021-12-09:
> Of special note in the [upgrade guide](https://docs.cupy.dev/en/latest/upgrade.html) is a [change in `cupy.cuda.Device` behavior](https://docs.cupy.dev/en/latest/upgrade.html#change-in-cupy-cuda-device-behavior) - does this affect our cupy use?

Good point! The code should not be affected since we don't use it as a context manager: https://github.com/LiberTEM/LiberTEM/blob/7009b3c6f1563e77505e19ec7f47f28c893908ae/src/libertem/udf/base.py#L1641-L1667

uellue, 2021-12-09:
Tests in `tests/test_analysis_masks.py` fail for a different reason that is independent of the Cuda version. I believe it is related to re-using the slices from the tiling scheme. The CuPy device Tile is a cupy.ndarray and not a `DataTile`.

I'm not sure how to handle the DataTile when supporting different kind of objects that implement the ndarray interface. I can see the following options:

* Treat `DataTile` like a template<ndarray-like>, i.e. allow creating different DataTiles for different base types, for example through multiple inheritance
* Deprecate the extra attributes of `tile` and move everything to UDFMeta. I believe the tile attributes predate UDFMeta?

sk1p, 2021-12-09:
Good catch!

> * Treat `DataTile` like a template, i.e. allow creating different DataTiles for different base types, for example through multiple inheritance

That would be a prime example for using a [`typing.Protocol`](https://docs.python.org/3/library/typing.html#typing.Protocol), but...

> * Deprecate the extra attributes of `tile` and move everything to UDFMeta. I believe the tile attributes predate UDFMeta?

I think something like this is more reasonable. The `DataTile` has not really been a public interface anyways, at least looking through our documentation. `UDFTileMixin.process_tile` specifies `tile` as `np.ndarray` - so  instead of deprecating the extra attributes (they are the reason why `DataTile` exists - otherwise, `get_tiles` could just yield `np.ndarray`s...), we add the necessary attributes to `UDFMeta` and are already done. We can easily add `UDFMeta.tile_scheme_idx` and `UDFMeta.sig_slice` to support common use cases. If we don't want `DataTile` to leak through to UDFs, we could even "shed" the extra attributes before passing it to `process_*`. Thoughts?

sk1p, 2021-12-09:
Milestoning for 0.9 since this is a recent regression.

PR #1167 (2021-12-09): Add `UDFMeta.sig_slice` and `UDFMeta.tiling_scheme_idx`
They are meant as a direct replacement of accessing the `DataTile` attributes directly, and serve as an official interface for this functionality, which also works with cupy.

In addition, removes a now-redundant `UDFMeta` recreation in the `UDFRunner`, which was needed before to set the tiling scheme.

Fixes #1166

## Contributor Checklist:

* [N/A] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.co [... cut]
commit 1aeb834b (2021-12-09): Add `UDFMeta.sig_slice` and `UDFMeta.tiling_scheme_idx`

They are meant as a direct replacement of accessing the `DataTile`
attributes directly, and serve as an official interface for this
functionality, that also works with cupy.

In addition, removes a now-redundant `UDFMeta` recreation in the
`UDFRunner`, which was needed before to set the tiling scheme.

Fixes #1166
closed 2021-12-09 by uellue