LiberTEM/LiberTEM #1175: Docs build failure with recently released nbsphinx 0.8.8

uellue, 2022-01-10:
See https://dev.azure.com/LiberTEM/LiberTEM/_build/results?buildId=1542&view=logs&jobId=2e37e63e-629f-5d60-4506-5615d0d227d3&j=2e37e63e-629f-5d60-4506-5615d0d227d3&t=580c1cad-5a9c-5ef1-d5a7-d2e4b5865820. I can reproduce this locally.

A band-aid that blacklists nbsphinx 0.8.8 is contained in #1174.

For a proper fix we should create a minimal reproducer and fix our notebook or follow up with upstream.

uellue, 2022-01-10:
Maybe we can postpone this until it (possibly) breaks again when the next nbsphinx release is out or we run into other issues with using 0.8.7 for now?

sk1p, 2022-01-10:
As a note, the concrete problem is fixed in https://github.com/spatialaudio/nbsphinx/commit/28184d2d952411922dbd6f157c99880b41f0f34b

uellue, 2022-01-10:
> As a note, the concrete problem is fixed in [spatialaudio/nbsphinx@28184d2](https://github.com/spatialaudio/nbsphinx/commit/28184d2d952411922dbd6f157c99880b41f0f34b)

Nice, so waiting out the next release is the correct strategy I guess? :-)

sk1p, 2022-01-10:
Yeah, blacklisting the broken release is the right move

uellue, 2022-01-10:
Considered fixed, then! :-)

closed 2022-01-10 by uellue