LiberTEM/LiberTEM #1183: Release 0.9

sk1p, 2022-01-13:
## TODO: Issues and pull requests to be fixed/finished before release

- [ ] TVIPS #1179 
- [ ] `DelayedJobExecutor` #1170 
- [ ] Build a release candidate and add release checklist below

@uellue @matbryan52 feel free to add additional tasks in this list

sk1p, 2022-01-25:
CI build w/ slow tests and recreating tox environments: https://dev.azure.com/LiberTEM/LiberTEM/_build/results?buildId=1648&view=results

sk1p, 2022-01-25:
`tests/udf/test_memleak.py::test_executor_memleak[dask]` fails on Mac OS X for some reason, log here: https://dev.azure.com/LiberTEM/LiberTEM/_build/results?buildId=1648&view=logs&j=7f5f2969-2f30-5453-4239-97ef5e9a1fd4&t=abbc44e5-2855-5024-6cb4-fa218810c41b

Might need some tweaks to make the test more robust, or we really do have a memory leak?

sk1p, 2022-01-25:
Another `-m slow` run here: https://dev.azure.com/LiberTEM/LiberTEM/_build/results?buildId=1655&view=results

sk1p, 2022-01-25:
[v0.9.0rc1](https://github.com/LiberTEM/LiberTEM/releases/tag/v0.9.0rc1) published.

sk1p, 2022-01-26:
Docker image runs fine with both docker itself and singularity, as documented.

sk1p, 2022-01-26:
As does the AppImage.

sk1p, 2022-01-26:
Looking at [the issues and PRs closed with the 0.9 milestone](https://github.com/LiberTEM/LiberTEM/milestone/14?closed=1), I think the changelog fragments are pretty much complete.

A minor issue missing from the changelog is #753, which was implemented as part of #1129

uellue, 2022-01-28:
@sk1p @matbryan52 in https://github.com/LiberTEM/LiberTEM/pull/1197/commits/b0a5a0a230cbf84405985378a856b372837229ba I have added new contributors. Am I missing anyone?

sk1p, 2022-01-31:
> Docs say that only 1-bit and 6-bit work for MIB quad data. However, opening /Users/weber/LargeData/ER-C-1/groups/data_science/data/reference/MIB/quad/Au_focused/12-bit_raw-20210728_095715/12-bit_raw.hdr works, apparently. 

Indeed, sorry about that - 12bit quad raw support was added a bit later than the others, and is indeed supported! Could you fix this in #1197? Good catch!

>  I have added new contributors. Am I missing anyone?

Looks good to me.

uellue, 2022-01-31:
> Could you fix this in #1197?

Sure, will do! :-)

uellue, 2022-02-14:
Discussion with @sk1p about the CI situation: So far there wasn't any substantial progress to get Azure Pipelines running again. In order to get the release out, @sk1p will migrate all Linux-based jobs to the data pipeline. We'll run Windows and OS X tests by hand once all pending PRs are merged, and continue with the release checklist.

For the time after, we'll deploy a dedicated machine to run all Linux CI jobs in-house. For OS X and Windows jobs we'd migrate to GitHub Actions. At the same time we'll wrap them in such a way that they are easy to execute locally with all Python versions. That way we have a fallback at hand. Should GitHub actions run into the same issues, we can think about actually deploying our own CI workers for that as well.

In conclusion, an expensive lession how depending on cloud services can lead to issues. I'm now more and more in favor to run open source solutions on own hardware with minimum exposure to the whims of third parties.

sk1p, 2022-02-15:
I'll re-open this until the release is actually out there.

uellue, 2022-02-17:
From my side the RC3 is good to go. :-)

sk1p, 2022-02-17:
I'll go ahead and tag the release, then!

sk1p, 2022-02-17:
https://github.com/Characterisation-Virtual-Laboratory/CharacterisationVL-Software/pull/136 update on CVL is in the pipeline

uellue, 2022-02-17:
Congrats and thanks, everyone! Release 0.9 is officially finished. :-) Lots of work, nice result!

closed 2022-02-17 by uellue