LiberTEM/LiberTEM #1187: Test failures with current version of dependencies

sk1p, 2022-01-24:
Running the tests with current versions of numba, numpy, pytest-asyncio fails and/or times out. The following is an incomplete summary:

```
FAILED tests/template/test_com_template.py::test_com_default - nbclient.exceptions.CellTimeoutError: A cell timed out while it was being executed, after 600 seconds.
FAILED tests/template/test_sum_template.py::test_sum_default - nbclient.exceptions.CellTimeoutError: A cell timed out while it was being executed, after 600 seconds.
FAILED tests/template/test_com_template.py::test_com_rotflip - nbclient.exceptions.CellExecutionError: An error occurred while executing the following cell:
FAILED tests/template/test_sum_template.py::test_sum_roi - nbclient.exceptions.CellExecutionError: An error occurred while executing the following cell:
FAILED tests/template/test_sumsig_template.py::test_sum_sig_default - nbclient.exceptions.CellExecutionError: An error occurred while executing the following cell:
FAILED tests/template/test_disk_template.py::test_disk_default - nbclient.exceptions.CellTimeoutError: A cell timed out while it was being executed, after 600 seconds.
FAILED tests/template/test_fem_template.py::test_fem_analysis - nbclient.exceptions.CellExecutionError: An error occurred while executing the following cell:
FAILED tests/template/test_fft_template.py::test_sum_fft_default - nbclient.exceptions.CellTimeoutError: A cell timed out while it was being executed, after 600 seconds.
ERROR tests/server/test_browse.py::test_browse_localfs_fail - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_cancel.py::test_cancel_udf_job - RuntimeError: Event loop is closed
ERROR tests/server/test_cancel.py::test_cancel_udf_job - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_cluster_status.py::test_local_cluster_details - RuntimeError: Event loop is closed
ERROR tests/server/test_cluster_status.py::test_local_cluster_details - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_dataset.py::test_load_raw_success - RuntimeError: Event loop is closed
ERROR tests/server/test_dataset.py::test_load_raw_success - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_dataset.py::test_load_raw_with_backends - RuntimeError: Event loop is closed
ERROR tests/server/test_dataset.py::test_load_raw_with_backends - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_dataset.py::test_load_raw_fail - RuntimeError: Event loop is closed
ERROR tests/server/test_dataset.py::test_load_raw_fail - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_dataset.py::test_dataset_delete - RuntimeError: Event loop is closed
ERROR tests/server/test_dataset.py::test_dataset_delete - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_dataset.py::test_initial_state_after_reconnect - RuntimeError: Event loop is closed
ERROR tests/server/test_dataset.py::test_initial_state_after_reconnect - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_dataset.py::test_prime_cache - RuntimeError: Event loop is closed
ERROR tests/server/test_dataset.py::test_prime_cache - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_download.py::test_download_hdf5 - RuntimeError: Event loop is closed
ERROR tests/server/test_download.py::test_download_hdf5 - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_startup.py::test_get_config - RuntimeError: Event loop is closed
ERROR tests/server/test_startup.py::test_get_config - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_startup.py::test_conn_is_disconnected - RuntimeError: Event loop is closed
ERROR tests/server/test_startup.py::test_conn_is_disconnected - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_startup.py::test_conn_connect_local - RuntimeError: Event loop is closed
ERROR tests/server/test_startup.py::test_conn_connect_local - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_browse.py::test_browse_localfs - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_startup.py::test_cluster_connect_error - RuntimeError: Event loop is closed
ERROR tests/server/test_job.py::test_run_job_delete_ds - RuntimeError: Event loop is closed
ERROR tests/server/test_startup.py::test_cluster_connect_error - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_job.py::test_run_job_delete_ds - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_startup.py::test_initial_state_empty - RuntimeError: Event loop is closed
ERROR tests/server/test_job.py::test_cancel_unknown_job - RuntimeError: Event loop is closed
ERROR tests/server/test_startup.py::test_initial_state_empty - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_startup.py::test_initial_state_w_existing_ds - RuntimeError: Event loop is closed
ERROR tests/server/test_job.py::test_cancel_unknown_job - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_job.py::test_run_with_all_zeros_roi - RuntimeError: Event loop is closed
ERROR tests/server/test_startup.py::test_initial_state_w_existing_ds - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_job.py::test_run_with_all_zeros_roi - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_startup.py::test_initial_state_analyses - RuntimeError: Event loop is closed
ERROR tests/server/test_job.py::test_run_job_update_analysis_parameters - RuntimeError: Event loop is closed
ERROR tests/server/test_startup.py::test_initial_state_analyses - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_job.py::test_run_job_update_analysis_parameters - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_state.py::test_job_state_remove - RuntimeError: Event loop is closed
ERROR tests/server/test_job.py::test_analysis_removal - RuntimeError: Event loop is closed
ERROR tests/server/test_state.py::test_job_state_remove - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_job.py::test_analysis_removal - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_token_auth.py::test_missing_token[GET-/] - RuntimeError: Event loop is closed
ERROR tests/server/test_token_auth.py::test_missing_token[GET-/] - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_job.py::test_create_compound_analysis - RuntimeError: Event loop is closed
ERROR tests/server/test_job.py::test_create_compound_analysis - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_token_auth.py::test_missing_token[GET-/api/datasets/detect/] - RuntimeError: Event loop is closed
ERROR tests/server/test_job.py::test_rerun_com_analysis - RuntimeError: Event loop is closed
ERROR tests/server/test_token_auth.py::test_missing_token[GET-/api/datasets/detect/] - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_token_auth.py::test_missing_token[GET-/api/datasets/schema/] - RuntimeError: Event loop is closed
ERROR tests/server/test_job.py::test_rerun_com_analysis - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_job.py::test_com_guess_rpc - RuntimeError: Event loop is closed
ERROR tests/server/test_token_auth.py::test_missing_token[GET-/api/datasets/schema/] - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_token_auth.py::test_missing_token[POST-/] - RuntimeError: Event loop is closed
ERROR tests/server/test_job.py::test_com_guess_rpc - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_token_auth.py::test_missing_token[POST-/] - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_token_auth.py::test_missing_token[POST-/api/datasets/detect/] - RuntimeError: Event loop is closed
ERROR tests/server/test_job.py::test_com_guess_rpc_need_to_run_job - RuntimeError: Event loop is closed
ERROR tests/server/test_job.py::test_com_guess_rpc_need_to_run_job - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_token_auth.py::test_missing_token[POST-/api/datasets/detect/] - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_shutdown.py::test_shutdown - RuntimeError: Event loop is closed
ERROR tests/server/test_token_auth.py::test_missing_token[POST-/api/datasets/schema/] - RuntimeError: Event loop is closed
ERROR tests/server/test_shutdown.py::test_shutdown - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_token_auth.py::test_missing_token[POST-/api/datasets/schema/] - RuntimeError: Cannot close a running event loop
ERROR tests/server/test_startup.py::test_start_server - RuntimeError: Event loop is closed
ERROR tests/server/test_startup.py::test_start_server - RuntimeError: Cannot close a running event loop
```

I suspect this is the pytest-asyncio release from last week - I'll dig a bit deeper now.

sk1p, 2022-01-24:
This is probably https://github.com/pytest-dev/pytest-asyncio/issues/271 with the suggested fix being a session-scoped event loop. This simple fix makes it _almost_ work:

```patch
diff --git a/conftest.py b/conftest.py
index f01a2a47..f4834496 100644
--- a/conftest.py
+++ b/conftest.py
@@ -475,6 +475,13 @@ def pytest_fixture_post_finalizer(fixturedef, request):
         adjust_event_loop_policy()
 
 
+@pytest.fixture(scope="session")
+def event_loop():
+    loop = asyncio.get_event_loop_policy().new_event_loop()
+    yield loop
+    loop.close()
+
+
 @pytest.fixture(autouse=True)
 def auto_ctx(doctest_namespace):
     ctx = lt.Context(executor=InlineJobExecutor())
```

It then errs out some time after ("sometimes" after the last test case - if I run the last test case in isolation, it runs fine...)

sk1p, 2022-01-24:
Getting closer - `pytest tests/server/test_ds_detect.py tests/server/test_token_auth.py -x` triggers the issue on cleanup, without `test_ds_detect` tests run successfully.

sk1p, 2022-01-24:
Further tracked down to the `local_cluster_url` fixture. Easy reproducer:

```python
@pytest.mark.asyncio
async def test_stuff(local_cluster_url):
    pass
```

This fails in isolation.

sk1p, 2022-01-24:
Further evidence: the above doesn't fail in isolation anymore, once I override the `event_loop` fixture with one that has session scope, _and_ move [the loop cleanup](https://github.com/LiberTEM/LiberTEM/blob/4e2a1402e954ff939ce131b3b93e435d0f04ce21/conftest.py#L310) introduced in https://github.com/LiberTEM/LiberTEM/pull/565 completely into this overridden fixture. It seems to me that our own cleanup code conflicts with the code recently introduced in pytest-asyncio, as it's also not clear which of the two runs first etc (as they are using the same hook, as far as I can see) - moving that into the fixture means we know our code runs first.

The asyncio-based tests in tests/server/ run through now, but there are still failures in the notebook/nbconvert-based tests in tests/template/, which also take quite long to time out. I'll now try to figure out what the issue is with these (possibly something with `nest_asyncio` or similar...)

sk1p, 2022-01-24:
In summary, this looks like something that could have severe impact on the stability of our test suite - maybe the best course of action is to pin pytest-asyncio until 0.9 is out of the door.

uellue, 2022-01-24:
@sk1p should I add the pin to #1170?

sk1p, 2022-01-24:
Yes, go ahead.

PR #1188 (2022-05-02): Fixes for the current pytest-asyncio
The changes here at least make `tests/server` run through; the ones in `tests/template` ~still hang indefinitely somewhere, after
the main test code times out. Possibly the hang is in the added code for event loop cleanup, like `loop.shutdown_asyncgens` and `loop.shutdown_default_executor`?~ Updating to current `pytest` and `pytest-asyncio` fixed the hang in there, too.

Fixes #1187

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [ ] I  [... cut]
closed 2022-05-02 by uellue