LiberTEM/LiberTEM #1194: DelayedExecutor / CuPy (+others) support

matbryan52, 2022-01-27:
In checking the documentation for 0.9.0 (#1193), I realized that we haven't considered CuPy / CUDA support with the DelayedExecutor, except the convenience method `get_resources_from_udfs()` mentioned here [CUDA and Scheduling](https://libertem.github.io/LiberTEM/dask.html#cuda-and-scheduling).

I'm not sure we've ever tested CUDA-based execution (CuPy or otherwise) with `DelayedJobExecutor`. Certainly there is no capacity in the `DaskBufferWrapper` implementation to allocate the result arrays on the device (the `lib` kwarg to `.allocate` is ignored).

Dask does have some capacity to integrate with CuPy at the array level (https://docs.dask.org/en/stable/gpu.html), but I haven't tested it in the `DelayedJobExecutor` context. I'm putting this issue out so that there is a trace of that.

matbryan52, 2022-02-02:
Another note, I pushed a prototype branch with the capacity to allocated chunked, cupy-backed Dask arrays as buffers:

https://github.com/matbryan52/LiberTEM/compare/83179d9..delayed_cupy

One thing I hadn't quite appreciated is that even when a UDF runs with cupy as backend and 'device' buffers, the results buffers are always allocated as numpy arrays. (I haven't had much opportunity to use GPU-backed LiberTEM because it's hard to access a GPU here). So, for my initial comment I think CuPy-backed UDFs will work with the Delayed Executor because inside the UDFTask itself we run normal (dask-free) implementations for the Buffers, and only on the main node will we create the Dask-chunked results. (which would never have been cupy anyway).

I'll try to test this assertion on ptycho or another system, however.

uellue, 2025-03-27:
Since this the focus has shifted to live processing and the DelayedExecutor didn't really catch on, sadly. Closing for now until there's pull.

closed 2025-03-27 by uellue