LiberTEM/LiberTEM #1196: Default nav shape for our reference TVIPS dataset; opening with Nav shape (0, x) or (y, 0) in the GUI.

uellue, 2022-01-28:
## How to reproduce

* Open `/LargeData/ER-C-1/groups/data_science/data/reference/TVIPS/rec_20200623_080237_000.tvips` for the first time or reset parameters.
* Click "Load dataset" with the detected parameters.

I've tested it with Singularity on Moellenstedt, just FYI.

## What happens

LiberTEM detects the nav shape as `(10, 0)`. Clicking "Load dataset" with nav shape parameter doesn't trigger any action. The dialogue just remains as-is. After changing to (10, 1), the file can be loaded.

I've tested a nav shape of (0, x) or (y, 0) with MIB as well and get the same behavior.

Not getting any reaction confused me as a user.

## What should happen

First, the autodetected shape should not contain a zero. If the nav shape can't be detected properly for 4D STEM, it should default to (y, 1).

Second, attempting to load with an invalid shape should trigger an error or give some other feedback to the user, instead of just not doing anything.

Since that actually confused me even as an experienced user, users will likely encounter this at least for TVIPS data, and it should be straightforward to fix, I'd propose to include it in 0.9.

uellue, 2022-01-28:
After a bit of digging, it seems that the TVIPS dataset detects the correct shape `(10, )`, but the web GUI doesn't handle the missing parameter well. @sk1p could you perhaps have a look? :-)

AnandBaburajan, 2022-01-31:
@sk1p I can take this up in case you haven't started working on this already.

sk1p, 2022-01-31:
> @sk1p I can take this up in case you haven't started working on this already.

Thanks! I haven't started on it yet, so feel free to take over. Let me know if you need assistance 😊

sk1p, 2022-01-31:
@AnandBaburajan FYI our CI is having some issues right now, but feel free to create a PR anyways, I'll manually build and test if needed. Thanks!

closed 2022-02-14 by sk1p