LiberTEM/LiberTEM #1198: Permanent archival of Docker or Singularity images

uellue, 2022-01-31:
Docker or Singularity images could be useful to preserve a specific software version with its run-time environment for long-term archival and reproducibility in scientific computing.

[Docker Hub may delete images and change access policies at their discretion.](https://www.docker.com/blog/docker-hub-image-retention-policy-delayed-and-subscription-updates/)

For that reason we should look into a registry that guarantees longer retention. Perhaps the [GitHub container registry](https://docs.github.com/en/packages/working-with-a-github-packages-registry/working-with-the-container-registry) could be an option? Optimal would be a service like Zenodo.

sk1p, 2022-01-31:
> Optimal would be a service like Zenodo

We could directly generate and upload the Singularity image (SIF file) to Zenodo - that should be "straight forward"™.

uellue, 2025-03-27:
Switched to GitHub container registry since then.

Probably it is better to prepare Singularity images for particular applications/analyses, for example to accompany a paper? Revisiting this issue now, the tests in CI against current versions seem more important to me since that's what most people actually use, not Singularity images. I'll close here.

closed 2025-03-27 by uellue