LiberTEM/LiberTEM #1200: New K2IS dataset fails to load

uellue, 2022-02-01:
## How to reproduce

We acquired a 4D STEM dataset at the Holo with the K2IS and I tried to open it in LiberTEM, bot on the command line and scripting. The file is in `/storage/holo/clausen/testdata/ER-C-1/groups/data_science/data/k2is 4dstem test 2022-02-01`.

## What happens in the GUI

I get an eternal spinner. On the command line it prints

```
[2022-02-01 17:11:50,624] ERROR [asyncio.default_exception_handler:1738] Exception in callback _chain_future.<locals>._set_state(<Future pendi...7c0fd7580>()]>, <Future at 0x...StopIteration>) at /Users/weber/.conda/envs/libertem39/lib/python3.9/asyncio/futures.py:378
handle: <Handle _chain_future.<locals>._set_state(<Future pendi...7c0fd7580>()]>, <Future at 0x...StopIteration>) at /Users/weber/.conda/envs/libertem39/lib/python3.9/asyncio/futures.py:378>
Traceback (most recent call last):
  File "/Users/weber/.conda/envs/libertem39/lib/python3.9/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/Users/weber/.conda/envs/libertem39/lib/python3.9/asyncio/futures.py", line 380, in _set_state
    _copy_future_state(other, future)
  File "/Users/weber/.conda/envs/libertem39/lib/python3.9/asyncio/futures.py", line 356, in _copy_future_state
    dest.set_exception(_convert_future_exc(exception))
TypeError: StopIteration interacts badly with generators and cannot be raised into a Future
[2022-02-01 17:11:51,666] ERROR [asyncio.default_exception_handler:1738] Task was destroyed but it is pending!
task: <Task pending name='Task-1278' coro=<RequestHandler._execute() running at /Users/weber/.conda/envs/libertem39/lib/python3.9/site-packages/tornado/web.py:1704> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /Users/weber/.conda/envs/libertem39/lib/python3.9/asyncio/futures.py:384, <TaskWakeupMethWrapper object at 0x7f37c0fd7580>()]> cb=[_HandlerDelegate.execute.<locals>.<lambda>() at /Users/weber/.conda/envs/libertem39/lib/python3.9/site-packages/tornado/web.py:2326]>
```

## What happens via scripting

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-10-1ea58fe8dc3e> in <module>
      1 params = {'path': '/Users/weber/LargeData/ER-C-1/groups/data_science/data/k2is 4dstem test 2022-02-01/Capture34_.gtg',}
----> 2 ds = ctx.load("K2IS", **params)

/cachedata/users/weber/LiberTEM-uellue/src/libertem/api.py in load(self, filetype, io_backend, *args, **kwargs)
    228         """
    229         # delegate to libertem.io.dataset.load:
--> 230         return load(
    231             filetype,
    232             *args,

/cachedata/users/weber/LiberTEM-uellue/src/libertem/io/dataset/__init__.py in load(filetype, enable_async, executor, *args, **kwargs)
    126     else:
    127         ds = cls(*args, **kwargs)
--> 128         ds = ds.initialize(executor)
    129         workers = executor.get_available_workers()
    130         scheduler = Scheduler(workers)

/cachedata/users/weber/LiberTEM-uellue/src/libertem/io/dataset/k2is.py in initialize(self, executor)
    786 
    787     def initialize(self, executor):
--> 788         return executor.run_function(self._do_initialize)
    789 
    790     def _set_skip_frames_and_nav_shape(self):

/cachedata/users/weber/LiberTEM-uellue/src/libertem/executor/dask.py in run_function(self, fn, *args, **kwargs)
    426         fn_with_args = functools.partial(fn, *args, **kwargs)
    427         future = self.client.submit(fn_with_args, priority=1, pure=False)
--> 428         return future.result()
    429 
    430     def map(self, fn, iterable):

~/.conda/envs/libertem39/lib/python3.9/site-packages/distributed/client.py in result(self, timeout)
    218         if self.status == "error":
    219             typ, exc, tb = result
--> 220             raise exc.with_traceback(tb)
    221         elif self.status == "cancelled":
    222             raise result

/cachedata/users/weber/LiberTEM-uellue/src/libertem/io/dataset/k2is.py in _do_initialize()
    763     def _do_initialize(self):
    764         self._files = self._get_files()
[... 86 lines cut ...]
    481         self._curGroupNameAtLevelX = oldGroupName

~/.conda/envs/libertem39/lib/python3.9/site-packages/ncempy/io/dm.py in _readTagGroup()
    433         oldTotalTag = self._curGroupNameAtLevelX
    434         for ii in range(0, nTags):
--> 435             self._readTagEntry()
    436 
    437         # Go back down a level after reading all entries

~/.conda/envs/libertem39/lib/python3.9/site-packages/ncempy/io/dm.py in _readTagEntry()
    455 
    456         if lenTagLabel > 0:
--> 457             tagLabelBinary = self.fromfile(self.fid, dtype='<u1', count=lenTagLabel)  # read as binary
    458             tagLabel = self._bin2str(tagLabelBinary)
    459             if self._v:

~/.conda/envs/libertem39/lib/python3.9/site-packages/ncempy/io/dm.py in fromfile()
    288             kwargs["offset"] = self._buffer_offset
    289             self._buffer_offset += int(dtype.itemsize) * count
--> 290             return np.frombuffer(*args, **kwargs)
    291         else:
    292             return np.fromfile(*args, **kwargs)

ValueError: buffer is smaller than requested size
```

sk1p, 2022-02-01:
Interesting - this looks like it's failing when reading the tags from the .gtg file. I'll have a look ASAP. Is this reproducible with different acquisitions, or is it only happening with this specific file?

sk1p, 2022-02-01:
I can reproduce this with:

```python
from libertem.io.dataset.k2is import _get_nav_shape
_get_nav_shape("./data/Capture34_.gtg")
```

uellue, 2022-02-01:
> Is this reproducible with different acquisitions, or is it only happening with this specific file?

So far we have only acquired that one file. We'll check another one later.

sk1p, 2022-02-01:
> So far we have only acquired that one file. We'll check another one later.

Thanks, that would be helpful! In the meantime, you could try to catch exceptions from/in `_get_nav_shape` and manually specify the `nav_shape`. Let me know if you need help with that!

[image]

So it tries to access beyond the file limits here... maybe an issue with an unknown (new?) tag, or just a corrupted file...

sk1p, 2022-02-01:
Oh, updating to latest `ncempy` version fixed this for me! Latest being `1.9.0`

sk1p, 2022-02-01:
Ok, so I see two issues: 

1) We need to guard the event loop from `StopIteration`, in all possible places (for example, `executor.run_function`, which may be the culprit here). That would mean the exception is properly forwarded to the caller, and the user at least gets some kind of feedback (even though in this case, a `MyStopIteration` exception would not be very user-friendly. I'm still not sure where the `StopIteration` really comes from, but I'll look into it.

2) `_get_nav_shape` for K2IS should be resilient to faults in its usage of `ncempy`, so it at least degrades gracefully in case of an exception.

sk1p, 2022-02-01:
This is the source of `StopIteration`:

```
In [5]: ds = ctx.load("auto", path="/storage/holo/clausen/testdata/ER-C-1/groups
   ...: /data_science/data/k2is 4dstem test 2022-02-01/Capture34_.gtg")         
distributed.worker - WARNING -  Compute Failed
Function:  _do_detect
args:      ()
kwargs:    {}
Exception: DataSetException("OSError('Unable to open file (file signature not found)')")

distributed.worker - WARNING -  Compute Failed
Function:  check_valid
args:      ()
kwargs:    {}
Exception: DataSetException('invalid magic number: 0 != 102')

distributed.worker - WARNING -  Compute Failed
Function:  _get_num_frames
args:      ()
kwargs:    {}
Exception: StopIteration()

---------------------------------------------------------------------------
StopIteration                             Traceback (most recent call last)
<ipython-input-5-910274c2623d> in <module>
----> 1 ds = ctx.load("auto", path="/storage/holo/clausen/testdata/ER-C-1/groups/data_science/data/k2is 4dstem test 2022-02-01/Capture34_.gtg")

~/source/LiberTEM/src/libertem/api.py in load(self, filetype, io_backend, *args, **kwargs)
    234             executor=self.executor,
    235             enable_async=False,
--> 236             **kwargs,
    237         )
    238 

~/source/LiberTEM/src/libertem/io/dataset/__init__.py in load(filetype, enable_async, executor, *args, **kwargs)
    105     """
    106     if filetype == "auto":
--> 107         return _auto_load(*args, executor=executor, enable_async=enable_async, **kwargs)
    108 
    109     cls = get_dataset_cls(filetype)

~/source/LiberTEM/src/libertem/io/dataset/__init__.py in _auto_load(path, executor, *args, **kwargs)
     52             "please specify the `path` argument to allow auto detection"
     53         )
---> 54     detected_params = detect(path, executor=executor)
     55     filetype_detected: typing.Optional[str] = detected_params.get('type', None)
     56     if filetype_detected is None:

~/source/LiberTEM/src/libertem/io/dataset/__init__.py in detect(path, executor)
    170         try:
    171             cls = get_dataset_cls(filetype)
--> 172             params = cls.detect_params(path, executor)
    173         except (NotImplementedError, DataSetException):
    174             continue

~/source/LiberTEM/src/libertem/io/dataset/k2is.py in detect_params(cls, path, executor)
    884             return False
    885         s = executor.run_function(_get_syncer_for_detect_params, files)
--> 886         num_frames = executor.run_function(_get_num_frames, s)
    887         num_frames_w_shutter_active_flag_set = executor.run_function(
    888             _get_num_frames_w_shutter_active_flag_set, s

~/source/LiberTEM/src/libertem/executor/dask.py in run_function(self, fn, *args, **kwargs)
    426         fn_with_args = functools.partial(fn, *args, **kwargs)
    427         future = self.client.submit(fn_with_args, priority=1, pure=False)
--> 428         return future.result()
    429 
    430     def map(self, fn, iterable):

/cachedata/users/clausen/conda/envs/libertem/lib/python3.7/site-packages/distributed/client.py in result(self, timeout)
    220         if self.status == "error":
    221             typ, exc, tb = result
--> 222             raise exc.with_traceback(tb)
    223         elif self.status == "cancelled":
    224             raise result

~/source/LiberTEM/src/libertem/io/dataset/k2is.py in _get_num_frames()
    302 
    303 def _get_num_frames(syncer):
--> 304     syncer.sync_sectors()
    305     with syncer.sectors[0] as s:
    306         num_frames = (

~/source/LiberTEM/src/libertem/io/dataset/k2is.py in sync_sectors()
    353                 lambda b: b.header['block_count'] == block_with_max_idx.header['block_count']
    354             )
--> 355             for s in self.sectors
    356         ]
    357         for b in start_blocks:

~/source/LiberTEM/src/libertem/io/dataset/k2is.py in <listcomp>()
    353                 lambda b: b.header['block_count'] == block_with_max_idx.header['block_count']
    354             )
--> 355             for s in self.sectors
    356         ]
    357         for b in start_blocks:

~/source/LiberTEM/src/libertem/io/dataset/k2is.py in first_block_with()
    541 
    542     def first_block_with(self, predicate=lambda b: True):
--> 543         return next(b for b in self.get_blocks()
    544                     if predicate(b))
    545 

StopIteration: 

In [6]:                                                                                                                                            
```

sk1p, 2022-02-01:
> This is the source of `StopIteration`:

Which, btw., is still happening with the current `ncempy`, triggered by `K2ISDataSet.detect_params`, so actually two independent bugs, nice!

uellue, 2022-02-01:
We now have a second dataset in the same folder with ID 35. The second I can open in the GUI, the first runs into that StopIteration issue

sk1p, 2022-02-01:
> ... i.e. after upgrading ncempy

Thanks for the confirmation! Looks like ID 34 runs into a weird edge case, I'm currently looking into the details.

sk1p, 2022-02-01:
Sounds good, we'll have to wait for the response from Microsoft anyways before we can release...

sk1p, 2022-02-01:
I have a suspicion about this: in the raw data stream from the camera, sometimes the frame ID resets from some high number to 0 before the actual data is acquired. This was not happening in any of the data sets that we acquired before, so the code doesn't handle this case. I'm still debugging if this is really true, though...

sk1p, 2022-02-01:
Something is going on with the synchronization here:

[image]

This is plotting all "block ids" for the blocks in each sector.

For the top sectors, it's possible to synchronize by "shifting the x axis" (IIRC), but the outlier sector is just completely off. This is a scatter plot, zoomed in to the top, which looks fine:

[image]

Maybe the physical synchronization cable is wonky for that one sector?

sk1p, 2022-02-01:
I think the conclusion from this issue is that we can't really load the data (except if one really wanted, one could implement some kind of try-hard data rescue mode, that ignores sectors that can't be synced to the others...). It's bad that the GUI runs into the non-responsiveness issue, which should be fixed by #1201 - should we put this issue on the backlog, or would you be happy closing it once #1201 is merged?

sk1p, 2022-02-15:
@uellue could you verify that the non-responsiveness issue mentioned above is fixed in current master? Then we can milestone the rest of the issue for 0.10

uellue, 2022-02-15:
The eternal spinner is fixed. It now just closes the dataset opening dialogue. I'll just close this, then, since the dataset seems to be broken.

closed 2022-02-15 by uellue