LiberTEM/LiberTEM #1211: Support sparse arrays for ROI in run_udf

matbryan52, 2022-02-15:
With the release of Scipy 1.8.0 there is an array interface for sparse matrices, allowing operations such as `array.nonzero()` and `array.count_nonzero()`, as well as slicing. (https://docs.scipy.org/doc/scipy/reference/sparse.html) 

This would be good to support as a way of reducing a bit of overhead when using ROIs. I think the modifications should be small, though the interface doesn't seem fully complete (for example `np.nonzero(sparse_array)` fails for me, but `sparse_array.nonzero()` works).

Something to look at.

matbryan52, 2022-02-15:
This should possibly just be added to #1207 , I didn't see how wide in scope that was!

uellue, 2022-02-15:
@matbryan52 interesting idea to use sparse ROIs! This should make quite a difference for frame pickingas well. The scipy.sparse arrays are currently limited to two dimensions, which means they won't work as a ROI at this time. I also tried a sparse.pydata.org COO matrix which can be n-dimensional, and dispatching `np.count_nonzero` failed there as well.

The ROI is also used deep in the internals of LiberTEM, including functions compiled with Numba: https://github.com/LiberTEM/LiberTEM/blob/778c7b18220fc689fad102daa6acf71dbaec1503/src/libertem/io/dataset/base/tiling.py#L177

Making them support sparse ROIs would be not trivial since Numba doesn't transparently support both NumPy and sparse arrays at this time. For that reason I'd recommend to only tackle this if we can show very significant performance benefits in a real-world use case. Picking frames from a very large nav dimension could be an example. :-)

uellue, 2022-02-15:
> Making them support sparse ROIs would be not trivial since Numba doesn't transparently support both NumPy and sparse arrays at this time

It could already help to transfer the ROI as sparse array and convert it to dense locally, actually. :-)

uellue, 2023-04-19:
Seems to be implemented, closing!

closed 2023-04-19 by uellue