LiberTEM/LiberTEM #1221: Release v0.9.1

sk1p, 2022-02-22:
We are planning to release a stable point release v0.9.1 soon - currently only for one issue (but important for our deployment):

- #1219 / #1220

If any other important bugs arise, please comment here.

sk1p, 2022-04-26:
CI doesn't appear to trigger correctly for `vX.Y.Z` branch names, this needs to be fixed before the point release can be published.

uellue, 2022-04-28:
Fix confirmed on Moellenstedt! Should be ready for release.

sk1p, 2022-04-28:
There's a [last-minute issue with our hyperspy sample notebook](https://dev.azure.com/LiberTEM/LiberTEM/_build/results?buildId=1763&view=logs&j=4d6f0f97-9d62-53bc-79d2-96c85fc226be&t=b5ec37f2-4755-55f4-9304-bcf3d588aca8), which we need to fix - 0.9.2 incoming.

PR #1244 (2022-04-28): Change log for 0.9.2

Fixes #1221

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
* [N/A] I have added/updated test cases
* [N/A] I have included the [rebuilt production build of the client](https://libertem.github.io/LiberTEM/contributing.html?#building-the-client) (only if cha [... cut]
closed 2022-04-28 by sk1p