LiberTEM/LiberTEM #1225: Allow spawning multiple CUDA workers on a single device when using cluster_spec

matbryan52, 2022-03-10:
In some cases when it is difficult to saturate a GPU with a single CUDA worker, it is beneficial to spawn multiple workers on the same device. This compensates for the fact that I/O to GPUs is currently sequential with processing in LiberTEM.

This is already possible by manually modifying the return dictionary from `cluster_spec` before passing it to the `DaskJobExecutor.make_local` method, but the change to support this use case is minimal, in fact all we have to do is ensure that the dictionary keys are unique when we iterate over the `cudas` argument to `cluster_spec`.

See https://github.com/matbryan52/LiberTEM/commit/9645bf2db5c4ecf66d6d9ae6b53ff0acf698f554 for a possible implementation.

In this case, the argument `cudas` is still an iterable of ints corresponding to CUDA device ids, but the implementation will spawn multiple workers on the device if the device id is present more than once. In the current release this silently fails by creating the same worker spec multiple times, each overwriting the previous entry in the worker spec dictionary.

uellue, 2022-06-20:
If several workers per GPU work better one could also make this the default! @matbryan52 did you try out what number of workers is best?

matbryan52, 2022-06-20:
> If several workers per GPU work better one could also make this the default! @matbryan52 did you try out what number of workers is best?

I am not sure there is a general answer to this, it depends on how heavy the UDF(s) computations is and the frame/tile size. The motivating example for this was a UDF where the only heavy compute step was a full-frame FFT on a reasonably small frame, which was leaving a lot of the GPU resources underutilised (between CPU => GPU transfer time and general LiberTEM UDF overheads).

PR #1270 (2022-06-20): Ensure unique CUDA worker names
Fixes the case where a CUDA worker spec with the same name squashes the previous worker spec in the `workers_spec` dictionary. This enables spawning multiple CUDA workers on the same GPU via `cluster_spec`, which currently has to be done by manually by editing the returned dictionary (now more challenging due to tracing setup!!).

The first cuda worker created on each device has an unmodified name ('default-cuda-0'), subsequent workers on the same device are named ('default-cuda-0-0', 'default-cuda-0-1', 'default-cuda-0-2' etc).

Closes #1225 

## Contributor Ch [... cut]