LiberTEM/LiberTEM #1262: Reconnect loop for LiberTEM using JupyterHub integration on Firefox 100.01 on Mac

uellue, 2022-06-01:
## How to reproduce

Vita was using LiberTEM with JupyterHub integration on Moellenstedt over VPN. Browser: Firefox 100.01 on Mac

Leave the tab unused for about 30 min (?).

## What happens

When accessing LiberTEM again after 30 min (?) of not using it, it enters a reconnect loop where the correct state only flashes briefly. Reloading the tab or closing and reopening didn't fix it.

LiberTEM cluster shutdown and restart fixes the issue.

## What should happen

The tab continues to work normally.

sk1p, 2022-06-01:
Checking the logs, this seems to be likely related:

```
Jun  1 12:23:23 iff588 run.sh: [2022-06-01 12:23:23,345] INFO [libertem.web.base.log_message:17] message: ANALYSIS_UPDATED (analysis=d0b30619-3acb-4b48-be6c-e387e02f2305)
Jun  1 12:23:23 iff588 run.sh: [I 2022-06-01 12:23:23.346 SingleUserNotebookApp log:189] 200 PUT /user/vmergner/libertem/api/compoundAnalyses/affb41bf-63fa-4227-b560-83d08b6e438c/analyses/d0b30619-3acb-4b48-be6c-e387e02f2305/ (vmergner@134.94.54.73) 3.19ms
Jun  1 12:23:24 iff588 run.sh: [I 2022-06-01 12:23:24.389 SingleUserNotebookApp log:189] 101 GET /user/vmergner/libertem/api/events/ (vmergner@134.94.54.73) 0.94ms
Jun  1 12:23:24 iff588 run.sh: [I 2022-06-01 12:23:24.389 SingleUserNotebookApp handlers:313] Trying to establish websocket connection to ws://localhost:46313/api/events/
Jun  1 12:23:24 iff588 run.sh: [I 2022-06-01 12:23:24.391 SingleUserNotebookApp handlers:321] Websocket connection established to ws://localhost:46313/api/events/
Jun  1 12:23:24 iff588 run.sh: [2022-06-01 12:23:24,408] INFO [libertem.web.config.get:22] ConfigHandler.get
Jun  1 12:23:24 iff588 run.sh: [2022-06-01 12:23:24,412] INFO [libertem.web.base.log_message:21] message: CONFIG
Jun  1 12:23:24 iff588 run.sh: [2022-06-01 12:23:24,414] INFO [libertem.web.connect.get:28] ConnectHandler.get
Jun  1 12:23:24 iff588 run.sh: [2022-06-01 12:23:24,414] INFO [libertem.web.base.log_message:21] message: INITIAL_STATE
Jun  1 12:23:24 iff588 run.sh: [2022-06-01 12:23:24,424] INFO [libertem.web.base.log_message:15] message: FINISH_JOB (job=3a7f1376-51a4-4c1a-b574-b779f87e6799)
Jun  1 12:23:24 iff588 run.sh: tornado.application - ERROR - Uncaught exception GET /api/events/ (::1)
Jun  1 12:23:24 iff588 run.sh: HTTPServerRequest(protocol='http', host='moellenstedt.iff.kfa-juelich.de', method='GET', uri='/api/events/', version='HTTP/1.1', remote_ip='::1')
Jun  1 12:23:24 iff588 run.sh: Traceback (most recent call last):
Jun  1 12:23:24 iff588 run.sh: File "/usr/local/anaconda2/envs/libertem/lib/python3.7/site-packages/tornado/websocket.py", line 956, in _accept_connection
Jun  1 12:23:24 iff588 run.sh: await open_result
Jun  1 12:23:24 iff588 run.sh: File "/usr/local/anaconda2/envs/libertem/lib/python3.7/site-packages/libertem/web/events.py", line 66, in open
Jun  1 12:23:24 iff588 run.sh: await self.engine.send_existing_job_results()
Jun  1 12:23:24 iff588 run.sh: File "/usr/local/anaconda2/envs/libertem/lib/python3.7/site-packages/libertem/web/engine.py", line 236, in send_existing_job_results
Jun  1 12:23:24 iff588 run.sh: job_id=job_id, analysis_id=analysis_id,
Jun  1 12:23:24 iff588 run.sh: File "/usr/local/anaconda2/envs/libertem/lib/python3.7/site-packages/libertem/web/messages.py", line 90, in start_job
Jun  1 12:23:24 iff588 run.sh: "details": self.state.job_state.serialize(job_id),
Jun  1 12:23:24 iff588 run.sh: File "/usr/local/anaconda2/envs/libertem/lib/python3.7/site-packages/libertem/web/state.py", line 318, in serialize
Jun  1 12:23:24 iff588 run.sh: job = self[job_id]
Jun  1 12:23:24 iff588 run.sh: File "/usr/local/anaconda2/envs/libertem/lib/python3.7/site-packages/libertem/web/state.py", line 309, in __getitem__
Jun  1 12:23:24 iff588 run.sh: return self.jobs[uuid]
Jun  1 12:23:24 iff588 run.sh: KeyError: '93e0a0b7-1b9a-404e-bddf-1efaa6913da1'
```

Somehow, the state about jobs is inconsistent.

uellue, 2023-04-19:
Could not reproduce on iff1094 with Firefox 112.0.1 and current LiberTEM deployment on Moellenstedt over JuNet. Closing until it appears again.

closed 2023-04-19 by uellue