LiberTEM/LiberTEM #1287: Release LiberTEM v0.10

sk1p, 2022-07-20:
# Release checklist

Issues and pull requests to be considered for this release:

* https://github.com/LiberTEM/LiberTEM/pull/1286
* https://github.com/LiberTEM/LiberTEM/pull/1267
* https://github.com/LiberTEM/LiberTEM/pull/1283

## Before (using a release candidate package)

* [ ] Review open issues and pull requests
* [ ] License review: no import of GPL code from MIT code
      `pydeps --only "libertem" --show-deps --noshow src\libertem | python scripts\licensecheck.py`
* [ ] Run full CI pipeline, including slow tests, on [Azure DevOps](https://dev.azure.com/LiberTEM/LiberTEM/_build?definitionId=3)
* [ ] Run tests for related packages w/ new LiberTEM version
    * [ ] LiberTEM-live
    * [ ] LiberTEM-holo
    * [ ] LiberTEM-blobfinder
    * [ ] ptychography40
* [ ] Handle deprecation, search the code base for `DeprecationWarning`
      that are supposed to be removed in that release.
* [ ] GUI dependency update with `npm install`
* [ ] Review https://github.com/LiberTEM/LiberTEM/security/dependabot and update dependencies
* [ ] Full documentation review and update, including link check using
      ``sphinx-build -b linkcheck "docs/source" "docs/build/html"``
* [ ] Run complete test suite, including slow tests that are deactivated by default
      and tests that require sample files.
* [ ] Update the expected version in notes on changes, i.e. from `0.3.0.dev0`
      to `0.3.0` when releasing version 0.3.0.
* [ ] Update and review change log in `docs/source/changelog.rst`, merging
      snippets in `docs/source/changelog/*/` as appropriate.
* [ ] Update the JSON files in the ``packaging/`` folder with author and project information
* [ ] Edit `pytest.ini` to exclude flaky tests temporarily from release builds
* [ ] Create a release candidate using `scripts/release`. See `scripts/release --help` for details.
* [ ] `Confirm that wheel, tar.gz, and AppImage are built for the release candidate on
      GitHub <https://github.com/LiberTEM/LiberTEM/releases>`_
* [ ] Confirm that a new version with the most recent release candidate is created in the
      `Zenodo.org sandbox <https://sandbox.zenodo.org/record/367108>`_ that is ready for submission.
* [ ] Install release candidate packages in a clean environment
      (for example:
      `python -m pip install -i https://test.pypi.org/simple/ --extra-index-url https://pypi.org/simple 'libertem==0.2.0rc11'`)
* [ ] Test the release candidate docker image
* [ ] For the GUI-related items, open in an incognito window to start from a clean slate
* [ ] Correct version info displayed in info dialogue?
* [ ] Link check in version info dialogue
* [ ] Make sure you have test files of all supported types available
    * [ ] Include floats, ints, big endian, little endian, complex raw data
* [ ] Open each test file
    * [ ] Are parameters recognized correctly, as far as implemented?
    * [ ] Any bad default values?
    * [ ] Does the file open correctly?
    * [ ] Have a look at the dataset info dialogue. Reasonable values?
* [ ] Perform all analyses on each test file.
    * [ ] Does the result change when the input parameters are changed?
    * [ ] All display channels present and looking reasonable?
    * [ ] Reasonable performance?
    * [ ] Use pick mode.
* [ ] Re-open all the files
    * [ ] Are the files listed in "recent files"?
    * [ ] Are the parameters filled from the cache correctly?
* [ ] Try opening all file types with wrong parameters
    * [ ] Proper understandable error messages?
* [ ] Pick one file and confirm keyboard and mouse interaction for all analyses
    * [ ] Correct bounds check for keyboard and mouse?
* [ ] Check what happens when trying to open non-existent files or directories in the GUI.
    * [ ] Proper understandable error message?
    * [ ] Possible to continue working?
* [ ] Shut down libertem-server while analysis is running
    * [ ] Shut down within a few seconds?
    * [ ] All workers reaped?
* [ ] Check what happens when trying to open non-existent files by scripting.
    * [ ] Proper understandable error message? TODO automate?
* [ ] Check what happens when opening all file types with bad parameters by scripting
    * [ ] Proper understandable error message? TODO automate?
* [ ] Run libertem-server on Windows, connect to a remote dask cluster running on Linux,
  open all file types and perform an analysis for each file type.
* [ ] Use the GUI while a long-running analysis is running
    * [ ] Still usable, decent response times?
* [ ] Confirm that pull requests and issues are handled as intended, i.e. milestoned and merged
  in appropriate branch.
* [ ] Final version bump: `./scripts/release bump v0.3.0 --tag`, push to github
* [ ] After pipeline finishes, write minimal release notes for the [release](https://github.com/liberTEM/LiberTEM/releases) and publish the GitHub release

## After releasing on GitHub

* [ ] Confirm that all release packages are built and release notes are up-to-date
* [ ] Install release package
* [ ] Confirm correct version info
* [ ] Confirm package upload to PyPI
* [ ] Confirm images and tags on https://hub.docker.com/r/libertem/libertem
* [ ] Publish new version on zenodo.org
* [ ] Update documentation with new links, if necessary
    * [ ] Add zenodo badge for the new release to Changelog page
* [ ] Send announcement message on mailing list
* [ ] Edit `pytest.ini` to include flaky tests again
* [ ] Bump version in master branch to next .dev0 (`./scripts/release bump v0.X.0.dev0 --commit`)
* [ ] Add to institutional publication databases
* [ ] Add the current LiberTEM version to [CVL](https://github.com/Chasdfracterisation-Virtual-Laboratory/CharacterisationVL-Software>) - add both the singularity and the .desktop file!

matbryan52, 2022-07-26:
I don't know where the master version of this checklist is stored, but we should add the necessary conda-related steps to it once that's in place.

sk1p, 2022-07-26:
> I don't know where the master version of this checklist is stored, but we should add the necessary conda-related steps to it once that's in place.

It's in the contribution docs

sk1p, 2022-07-26:
Release candidate v0.10.0rc1 is available from https://test.pypi.org/project/libertem/#description

sk1p, 2022-07-27:
Tests pass on Max OS X / x86_64 (without data access)

matbryan52, 2022-07-27:
> Release candidate v0.10.0rc1 is available from https://test.pypi.org/project/libertem/#description

Conda recipe at https://github.com/LiberTEM/condaforge-staged-recipes/blob/libertem/recipes/libertem/meta.yaml has been updated for the new package requirements and for the rc1 tarball on pypi-test. Tests pass for building on Ptycho (assuming empyre availability).

matbryan52, 2022-07-28:
Seems unlikely to be related to the release but I was looking at:

> Use the GUI while a long-running analysis is running. Still usable, decent response times?

And in fact I can fairly easily get the UI into a state where I can't tell if it's crashed or the workers are over-subscribed. If I wait long enough everything comes back into sync but for a few minutes (in this setup) there is little to show if the system is doing anything.

I think part of the problem is that the 'Analysis running...' text only seems to appear after the first result has arrived, which is unfortunate if the first result takes 60 seconds to arrive! This can be the case when opening a second analysis while a long analysis is already running, for example, where we don't even get the full UI loading into the second pane until a first frame is available. 

I guess more generally having a way of dynamically displaying the number of in_flight jobs might be useful (links to #1214 but is not exactly the same thing).

sk1p, 2022-07-28:
> I think part of the problem is that the 'Analysis running...' text only seems to appear after the first result has arrived, which is unfortunate if the first result takes 60 seconds to arrive! This [...]

Yeah, exactly - once a job is started, the busy indicator needs to start displaying, which is not the case yet. I noticed this too yesterday - I think it's one of the important usability issues of the web GUI.

sk1p, 2022-07-28:
I ran the libertem-live tests with LiberTEM v0.10.0rc1, and for both the `master` branch and the `dectris-wip` branch the merlin tests run through (dectris tests are not yet fully ported, so it's expected that they fail)

sk1p, 2022-07-28:
docker image `libertem/libertem:v0.10.0rc1` is pushed and tested (works both from docker and from singularity)

sk1p, 2022-07-28:
rc2 is currently baking...

sk1p, 2022-07-28:
v0.10 is baking :tada:

sk1p, 2022-07-28:
CVL: https://github.com/Characterisation-Virtual-Laboratory/CharacterisationVL-Software/pull/227

matbryan52, 2022-07-29:
PR is up for conda-forge https://github.com/conda-forge/staged-recipes/pull/19861

Build fails right now because empyre is not yet merged there but should be fine once that's the case.

sk1p, 2023-03-20:
I'll close this, as 0.10 was released quite a while ago and conda support isn't currently worked on actively