LiberTEM/LiberTEM #1309: Guess a square nav_shape if a square number of frames detected in the web interface

matbryan52, 2022-08-03:
For datasets which define their sig shape but not their nav shape (e.g. non-integrated Merlin), if a square number of frames is detected we could guess a square scan grid given how common that choice is in reality.

matbryan52, 2022-10-17:
Looking at the various datasets, I can see the cases where we fall back to `nav_shape = (image_count,)` in `detect_params` are the following:

- MIB
- TVIPS
- SEQ
- MRC
- K2IS

A small util function Python-side could be applied to these fall-back cases, giving a square nav shape if image_count is a square. This would avoid having to handle this on the Typescript side.

sk1p, 2022-10-17:
> A small util function Python-side could be applied to these fall-back cases, giving a square nav shape if image_count is a square. This would avoid having to handle this on the Typescript side.

Sounds good to me!

PR #1338 (2022-10-19): Choose square `nav_shape` if detected `num_frames` is square (web-only)
Fixes #1309 

Required updating tests for MIB and MRC `detect_params`. Quite difficult to test this for the other datasets affected as we don't necessarily have data recorded on a square nav grid (apart from mocking the data).

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [... cut]
closed 2022-10-19 by sk1p