LiberTEM/LiberTEM #1333: Multi-level progress bar for slow Datasets / UDFs

matbryan52, 2022-10-11:
When running heavy UDFs or reading from slow data sources (e.g. over the network), the `ctx.run_udf` progress bar can be stuck for a long time with no feedback to the user about what is happening as we wait on each partition result.

[Tqdm supports nested progress bars](https://github.com/tqdm/tqdm#nested-progress-bars) which could provide more fine-grained display about how the run is going, for example the current active partitions and their number of tilestacks/frames processed.

For single-process executors this is quite simple to implement, as we can update the progress bar directly, though for the Dask or Pipelined executor we need some IPC. In the Pipelined case we already implement IPC using queues, so with a bit of adaptation it could be used to update the main process about its progress. [Dask implements an API](https://distributed.dask.org/en/stable/logging.html#structured-logs) for event logging under topic labels (`worker.log_event(topic, message)`) - internally I think this is used to track task life cycles - and the `client` has a `subscribe_topic` method to trigger a callback on the main node when an event is logged on a worker.

I played around with the Dask version and was able to throw together the following prototype for about 40 lines of code, most of which is related to tqdm instantiation / destruction:

https://user-images.githubusercontent.com/78845903/195121065-99ac445c-da81-4725-bd2d-fefaa4ef0737.mp4

This was set to only signal progress about 20 times per partition as a rate limiter. I think it would also be good if the Task itself triggers the creation of its associated progress bar by sending a message to say "I am slow"; that way, for fast partitions / processing the only added overhead is measuring the tile processing speed and we won't need to send / respond to any events. While the extra logging / display will introduce some overhead once active, it should be minor compared to the already-slow processing speed we're trying to report on.

Clearly not a high priority feature, and it is less full-featured than the `opentelemetry` solution, but it could be quite transparent / simple for the user - only making itself visible when `progress=True` and the processing speed warrants it.

matbryan52, 2022-10-11:
One case to consider would be the output when there are many parallel workers; it would be too much to display 20+ progress bars at once. An option would be to display a few progress bars with one additional message / line stating the other active partitions, e.g.

```
75%|███████████████                                  |  15/20 [02:00<01:00,  6.03s/it]
Partition 16:  10%|████                              |   2/20 [02:00<01:00,  6.03s/it]
Partition 17:  30%|█████                             |   7/20 [02:00<01:00,  6.03s/it]
Partition 18:  50%|███████                           |  10/20 [02:00<01:00,  6.03s/it]
Also in progress: 19, 20, 21, 22, 23, 24, 25
```

sk1p, 2022-10-13:
As an alternative to multiple progress bars, it could be possible to have a single progress bar, but instead of dividing it into partitions, using the number of navigation items as a measure. Then, it can either be updated for full partitions in larger jumps, or for partial results in the slow case. What do you think? Do we gain anything crucial by having separate progress bars?

matbryan52, 2022-10-13:
> As an alternative to multiple progress bars, it could be possible to have a single progress bar, but instead of dividing it into partitions, using the number of navigation items as a measure. Then, [...]

We can test it! I think the communications requirements would be the same so the display part will be easy to prototype / modify after the fact.

My view on having per-partition / task display is that it gives us a bit more information about the state of the run - things like the number of tasks in progress, whether one task is stuck or slower than the others etc. If I requested a cluster with 8 workers but only ever have 4 partitions in progress then it might help me recognize something that is not obvious from a single progress bar.

matbryan52, 2022-10-24:
> As an alternative to multiple progress bars, it could be possible to have a single progress bar, but instead of dividing it into partitions, using the number of navigation items as a measure. Then, [...]

After having worked on this I fully agree, much simpler to stick to a single bar. Better compatibility with different terminals and more robust, too. The PR uses a single bar tracking frames with additional info about partitions.

PR #1341 (2022-11-07): Frame-level progress bar updates (+ comms system to enable it)
Closes #1333.

This modifies the progress bar to give more fine-grained progress updates when partitions are slow to compute:
```
Partitions 1(1)/8, Frames:  17%|███████                      | 176/1024 [00:11<00:56, 15.03it/s]
```
The new information includes the number of partitions in progress in parentheses `1(1)/8` and the total number of frames processed so far.  Additionally the progress bar in a Jupyter notebook will display as a JS widget rather than a text bar via `tqdm.auto`.

This is functioning for all executors except  [... cut]
closed 2022-11-07 by sk1p