LiberTEM/LiberTEM #1335: `Partition.make_slices()` can create very small 'final' partitions for odd shapes

matbryan52, 2022-10-18:
The `make_slices` method can sometimes generate oddly sized partitions if the input shape is not simply divisible by `num_partitions`

```python
    @classmethod
    def make_slices(cls, shape, num_partitions, sync_offset=0):
        ...
        num_frames = shape.nav.size
        f_per_part = max(1, num_frames // num_partitions)
        c0 = itertools.count(start=0, step=f_per_part)
        c1 = itertools.count(start=f_per_part, step=f_per_part)
        
        for (start, stop) in zip(c0, c1):
            if start >= num_frames:
                break
            stop = min(stop, num_frames)
            ...
```

For example: `shape = (8190, ...)` with `num_partitions = 4` generates slices for 5 partitions of size `(2047, 2047, 2047, 2047, 2)`, which is fairly sub-optimal! (and doesn't obey the `dataset.get_num_partitions()` method).

We're already importing `numpy` in this file, so an easy solution is the following using `np.linspace`, though the same functionality can be written using `itertools`:

```python
        num_frames = shape.nav.size
        boundaries = tuple(
                np.linspace(
                        0,
                        num_frames,
                        num=num_partitions + 1,
                        endpoint=True,
                        dtype=int
                    )
                )
        for (start, stop) in zip(boundaries[:-1], boundaries[1:]):
            ...
```

What I might have missed is a particular reason for the odd behaviour above ?

matbryan52, 2022-10-18:
Even though this should never happen in practice, it might be worth guarding against `num_partitions` > `num_frames` and `num_partitions <= 0`, for which `linspace` would raise.

sk1p, 2022-10-18:
I don't think there's a reason, other than "it's (was) good enough". Feel free to improve `make_slices`!

PR #1337 (2022-10-20): `make_slices` obeys `num_partitions` (avoids hanging frames)
Closes #1335 . Uses the implementation with `np.linspace`.

This required making a change to `MemoryPartition.get_tiles()` to propagate `tiling_intent` where available, which I think was actually a bug/oversight.

I also had to modify a number of test cases (including core tests in `tests/udf`) because they had been coded both for the old implementation of `make_slices` and the old implementation of `MemoryPartition.get_tiles()`. It seems like there were a few cases which previously were expected to raise with `pytest.raises` but now [... cut]
closed 2022-10-20 by sk1p