LiberTEM/LiberTEM #1339: Pipelined executor shutdown errors (cloudpickle teardown)

matbryan52, 2022-10-21:
I'm finding that `PipelinedExecutor` is failing to shutdown correctly (hangs after the analysis) with the following MWE. Difficult to diagnose but the tracebacks after `KeyboardInterrupt` seem to imply that an error was encountered on each worker during `PipelinedExecutor.close()`, specifically in the call to `SimpleMPWorkerQueue.put` to send `SHUTDOWN`:

One one machine:
```
KeyboardInterrupt: 
Exception ignored in: <function PipelinedExecutor.__del__ at 0x7f17011bf5b0>
Traceback (most recent call last):
  File "/home/bryan/Workspace/lt_fork/src/libertem/executor/pipelined.py", line 914, in __del__
  File "/home/bryan/Workspace/lt_fork/src/libertem/executor/pipelined.py", line 892, in close
  File "/home/bryan/Workspace/lt_fork/src/libertem/common/executor.py", line 490, in put
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/cloudpickle/cloudpickle_fast.py", line 73, in dumps
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/cloudpickle/cloudpickle_fast.py", line 633, in dump
AttributeError: 'NoneType' object has no attribute '__newobj__'
```
on another:
```
KeyboardInterrupt
Exception ignored in: <function PipelinedExecutor.__del__ at 0x7f74b1bcba60>
Traceback (most recent call last):
  File "/home/mat/Workspace/libertem_dev/libertem_fork/src/libertem/executor/pipelined.py", line 934, in __del__
  File "/home/mat/Workspace/libertem_dev/libertem_fork/src/libertem/executor/pipelined.py", line 910, in close
  File "/home/mat/Workspace/libertem_dev/libertem_fork/src/libertem/common/executor.py", line 490, in put
TypeError: 'NoneType' object is not callable
```

On both machines the call inside `put::490` is to the `cloudpickle` module. My guess is that during teardown the module is already gone by the time we get to the stage of closing down our workers, and as a result we cannot use `cloudpickle` to send the final `SHUTDOWN` message. An option would be a `.put_safe` method on `SimpleMPWorkerQueue` which only supports native python types without `cloudpickle`, and a corresponding second `except:` block in `get` which tries to use the object from the queue if either `cloudpickle` is torn down or it can't decode the object taken from the queue (in which case we assume it's a native python object).

The MWE:

```python
import tempfile
import pathlib
import numpy as np

import libertem.api as lt

if __name__ == '__main__':
    nav_shape = (32, 32)
    sig_shape = (4, 4)
    data = np.ones(nav_shape + sig_shape, dtype=np.float32)

    ctx = lt.Context.make_with('pipelined')

    with tempfile.TemporaryDirectory() as tpath:
        path = pathlib.Path(tpath) / 'test.raw'
        data.tofile(path)

        ds = ctx.load('raw',
                      path=path,
                      nav_shape=nav_shape,
                      sig_shape=sig_shape,
                      dtype=data.dtype)
        ds.set_num_cores(8)

        sum_a = ctx.create_sum_analysis(ds)
        sum_r = ctx.run(sum_a, progress=True)
```

matbryan52, 2022-10-21:
Another piece of info, registering the `executor.__del__` method to be run with the `atexit` module does correctly allow the script to exit and/or drop to a prompt (and later be exit()-ed), at least on this MWE. This is the case both with/without a special `put` to use during `close()`.

sk1p, 2022-10-24:
So, the root of the issue is not calling `Context.close` at the end of the script. Right now, we have a "last resort" `self.close` in `PipelinedExecutor.__del__` but, on interpreter shutdown, we don't have guarantees on basically anything that's going on around us. We can't rely on things not being cleaned up before us. So a possible fix of the script would be:

```python
import tempfile
import pathlib
import numpy as np

import libertem.api as lt

if __name__ == '__main__':
    nav_shape = (32, 32)
    sig_shape = (4, 4)
    data = np.ones(nav_shape + sig_shape, dtype=np.float32)

    with lt.Context.make_with('pipelined') as ctx:
        with tempfile.TemporaryDirectory() as tpath:
            path = pathlib.Path(tpath) / 'test.raw'
            data.tofile(path)

            ds = ctx.load('raw',
                          path=path,
                          nav_shape=nav_shape,
                          sig_shape=sig_shape,
                          dtype=data.dtype)
            ds.set_num_cores(8)

            sum_a = ctx.create_sum_analysis(ds)
            sum_r = ctx.run(sum_a, progress=True)
```

But, as it is not directly clear for users that this is the "blessed" style which is needed for standalone Python scripts, it could still be useful to have something that cleans up at process exit.

Registering something via `atexit` could work - one just has to be careful that it doesn't needlessly keep objects alive that would otherwise already be happily garbage-collected. 

Maybe something like this:

```python
import weakref
import tempfile
import pathlib
import numpy as np
import atexit

import libertem.api as lt

if __name__ == '__main__':
    nav_shape = (32, 32)
    sig_shape = (4, 4)
    data = np.ones(nav_shape + sig_shape, dtype=np.float32)

    ctx = lt.Context.make_with('pipelined')

    weak_ctx = weakref.ref(ctx)
    atexit.register(lambda: weak_ctx().close() if weak_ctx() is not None else None)

    with tempfile.TemporaryDirectory() as tpath:
        path = pathlib.Path(tpath) / 'test.raw'
        data.tofile(path)

        ds = ctx.load('raw',
                      path=path,
                      nav_shape=nav_shape,
                      sig_shape=sig_shape,
                      dtype=data.dtype)
        ds.set_num_cores(8)

        sum_a = ctx.create_sum_analysis(ds)
        sum_r = ctx.run(sum_a, progress=True)
```

That could automatically be registered in `Context.__init__`.

sk1p, 2022-10-24:
> Right now, we have a "last resort" `self.close` in `PipelinedExecutor.__del__` but, on interpreter shutdown, we don't have guarantees on basically anything that's going on around us

See also: https://docs.python.org/3/reference/datamodel.html#object.__del__

PR #1343 (2022-11-07): Make sure the `Context` cleans up after itself
Fixes #1339

It registers an atexit handler with a weak reference to the `Context`, and calls `Context.close` at process exit. Usually, using

    with Context(...) as ctx:
        ...

or explicitly calling `Context.close` for cleanup is preferred, but it is not obvious, and also not consistent with notebook usage, where a context manager is not practical.

Tests TBD - not sure yet how to properly test this. Sketch: maybe with a small function ran via `multiprocessing.Process(target=...)`, and two cases; 1) manual cleanup before interpreter shutd [... cut]
closed 2022-11-07 by sk1p