LiberTEM/LiberTEM #1372: Support non-canonical sparse arrays and arrays with non-zero fill value in `count_nonzero()`

uellue, 2022-12-21:
https://github.com/LiberTEM/LiberTEM/blob/1b823125adc681263cbbf828ef55a999ad513bdf/src/libertem/common/math.py#L37-L44

For example this:

```python
import sparse
from libertem.common.math import count_nonzero

a = sparse.COO(((1, ), (2, ), (3, )), data=[False], fill_value=False, shape=(4, 4, 4))
count_nonzero(a), count_nonzero(a.todense())
# (1, 0)

b = sparse.COO(((1, ), (2, ), (3, )), data=[True], fill_value=True, shape=(4, 4, 4))
count_nonzero(b), count_nonzero(b.todense())
# (1, 64)

jalajk24, 2023-01-31:
Can I work on this issue? I found it interesting to work on but don't know a lot about this

sk1p, 2023-02-01:
> Can I work on this issue? I found it interesting to work on but don't know a lot about this

Hi, thank you for expressing your interest! Feel free to work on this, I'll assign the issue to you.

jalajk24, 2023-02-01:
@sk1p can you plz elaborate this issue and drop some insight for how to approach this issue

sk1p, 2023-02-01:
I can try :) So the issue is that our helper `count_nonzero(a)` gives a different result than `count_nonzero(a.todense())`. This happens if `a` is a sparse array with a different fill value than `0`.

So to fix this issue, the first step would be to understand the above, i.e. understand how `sparse.COO` arrays work, how the helper `count_nonzero` works, and to reproduce the problem. This issue then needs some research on how to actually fix it (that's the main part of the work!).

PR #1711 (2025-03-27): Fix `count_nonzero()`
Closes #1372

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [N/A] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [N/A] I have added/updated documentation for all user-facing changes
* [x] I have added/updated test cases
* [N/A] I have included the [rebuilt production build of the client](https://libertem.github.io/LiberTEM/contributing.html?#building-the-client) (only if  [... cut]
commit 69820920 (2025-03-27): Fix `count_nonzero()`

Closes #1372
closed 2025-03-27 by sk1p