LiberTEM/LiberTEM #1396: Too many open files error while running slow tests

matbryan52, 2023-03-21:
While running the slow tests on the server I get the following errors related to too many open files (I think probably processes), coming from distributed.

```
2023-03-21 10:07:28,708 - tornado.application - ERROR - Exception in callback functools.partial(<bound method NannyMemoryManager.memory_monitor of <distributed.worker_memory.NannyMemoryManager object at 0x7fe4a3221420>>, <Nanny: tcp://134.94.164.101:42685, threads: 1>)
Traceback (most recent call last):
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/psutil/_common.py", line 443, in wrapper
    ret = self._cache[fun]
AttributeError: 'Process' object has no attribute '_cache'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/distributed/compatibility.py", line 161, in _run
    val = self.callback()
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/distributed/worker_memory.py", line 366, in memory_monitor
    memory = psutil.Process(process.pid).memory_info().rss
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/psutil/__init__.py", line 332, in __init__
    self._init(pid)
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/psutil/__init__.py", line 361, in _init
    self.create_time()
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/psutil/__init__.py", line 714, in create_time
    self._create_time = self._proc.create_time()
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/psutil/_pslinux.py", line 1645, in wrapper
    return fun(self, *args, **kwargs)
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/psutil/_pslinux.py", line 1855, in create_time
    ctime = float(self._parse_stat_file()['create_time'])
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/psutil/_pslinux.py", line 1645, in wrapper
    return fun(self, *args, **kwargs)
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/psutil/_common.py", line 446, in wrapper
    return fun(self)
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/psutil/_pslinux.py", line 1687, in _parse_stat_file
    data = bcat("%s/%s/stat" % (self._procfs_path, self.pid))
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/psutil/_common.py", line 776, in bcat
    return cat(fname, fallback=fallback, _open=open_binary)
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/psutil/_common.py", line 764, in cat
    with _open(fname) as f:
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/site-packages/psutil/_common.py", line 728, in open_binary
    return open(fname, "rb", buffering=FILE_READ_BUFFER_SIZE)
OSError: [Errno 24] Too many open files: '/proc/3684716/stat'
```

```
FAILED tests/template/test_clust_template.py::test_clust_default - OSError: [Errno 24] Too many open files: '~/miniconda3/envs/lt_fork/share/jupyter/kernels'
FAILED tests/template/test_fem_template.py::test_fem_analysis - OSError: [Errno 24] Too many open files
ERROR tests/server/test_cluster_status.py::test_tcp_cluster_details - KeyError: 'DASK_SCHEDULER_ADDRESS'
ERROR tests/server/test_token_auth.py::test_wrong_token_full[OPTIONS-/api/shutdown/] - AssertionError: server thread failed to start
ERROR tests/server/test_token_auth.py::test_wrong_token_full[OPTIONS-/api/config/] - AssertionError: server thread failed to start
ERROR tests/server/test_token_auth.py::test_wrong_token_full[OPTIONS-/api/config/cluster/] - AssertionError: server thread failed to start
ERROR tests/server/test_token_auth.py::test_wrong_token_full[OPTIONS-/api/config/connection/] - OSError: [Errno 24] Too many open files: '/proc/cpuinfo'
ERROR tests/server/test_token_auth.py::test_missing_token_full[GET-/] - OSError: [Errno 24] Too many open files: '/proc/cpuinfo'
ERROR tests/server/test_token_auth.py::test_missing_token_full[GET-/api/datasets/detect/] - OSError: [Errno 24] Too many open files: '/proc/cpuinfo'
ERROR tests/server/test_token_auth.py::test_missing_token_full[GET-/api/datasets/schema/] - OSError: [Errno 24] Too many open files: '/proc/cpuinfo'
ERROR tests/server/test_token_auth.py::test_missing_token_full[GET-/api/datasets/ds-uuid-here/] - OSError: [Errno 24] Too many open files: '/proc/cpuinfo'
ERROR tests/server/test_token_auth.py::test_missing_token_full[GET-/api/browse/localfs/] - OSError: [Errno 24] Too many open files: '/proc/cpuinfo'
...
...
```

Later I get another error in stderr:

```
Exception ignored in: <function BaseEventLoop.__del__ at 0x7fe5b656caf0>
Traceback (most recent call last):
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/asyncio/base_events.py", line 690, in __del__
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/asyncio/unix_events.py", line 68, in close
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/asyncio/selector_events.py", line 87, in close
  File "/home/bryan/miniconda3/envs/lt_fork/lib/python3.10/asyncio/selector_events.py", line 94, in _close_self_pipe
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_ssock'
sys:1: RuntimeWarning: coroutine 'NotebookClient.async_start_new_kernel' was never awaited
```

The environment is the following:

```
distributed == 2022.11.1
dask==2022.11.1
tornado==6.1
```

This is almost certainly not directly LiberTEM's fault (seems independent of which branch I'm on), but I'm making a record here just in case someone else starts seeing it.

sk1p, 2023-03-21:
As a workaround on Linux or Mac OS, you can run the following in the shell before running the tests: `ulimit -n 4096`. On systems where this is not globally restricted, this will raise the limit of open files.

sk1p, 2024-05-16:
This should actually be fixed at least since #1561- I'm currently running `pytest tests/ -m "slow and not dist" -n 12` again and will close the issue if that succeeds.

closed 2024-05-16 by sk1p