LiberTEM/LiberTEM #1410: Issues with unmanaged memory in Dask workers

uellue, 2023-04-19:
## How to reproduce

Run `pytest -m "not dist" -x -s -vvv tests/executor/test_functional.py::test_executors[dask_executor-DM]` on OS X with 8 GB RAM.

Possibly it can also be triggered on other systems with constrained RAM.

## What happens

Dask workers have a memory limit of 1 GB in that scenario. They exceeded it regularly, leading to failed tests or hangs.

```
2023-04-19 20:43:37,361 - distributed.worker.memory - WARNING - Unmanaged memory use is high. This may indicate a memory leak or the memory may not be released to the OS; see https://distributed.dask.org/en/latest/worker-memory.html#memory-not-released-back-to-the-os for more information. -- Unmanaged memory: 733.17 MiB -- Worker memory limit: 1.00 GiB
2023-04-19 20:43:37,459 - distributed.worker.memory - WARNING - Unmanaged memory use is high. This may indicate a memory leak or the memory may not be released to the OS; see https://distributed.dask.org/en/latest/worker-memory.html#memory-not-released-back-to-the-os for more information. -- Unmanaged memory: 751.01 MiB -- Worker memory limit: 1.00 GiB
2023-04-19 20:43:37,974 - distributed.worker.memory - WARNING - Worker is at 93% memory usage. Pausing worker.  Process memory: 0.94 GiB -- Worker memory limit: 1.00 GiB
2023-04-19 20:43:38,057 - distributed.nanny.memory - WARNING - Worker tcp://134.94.164.23:63006 (pid=14681) exceeded 95% memory budget. Restarting...
```

## What should happen

A Dask worker shouldn't have so much unmanaged RAM.

We should figure out what exactly is eating RAM here. The error only occurred with DM datasets so far.

matbryan52, 2023-04-20:
I can recreate this on windows by artifically limiting worker memory to 1 GiB. Will try to track down why.

matbryan52, 2023-04-20:
> Results from triaging: The error is associated with using a `kind="sig"` UDF (`StdDevUDF` in that case, reproducible with `SumUDF`). It has nothing to do with DM per se, just that the test dataset for DM has very large frames, almost 4kx4k. Such a frame has a size of 128 MB with dtype=float64. In that light, the memory use could possibly be decreased, but it is not absurdly high, given that `StdDevUDF` needs two such buffers and maybe some temporary variables.
> 
> Deferring to backlog to see if we can eliminate a copy somewhere!

OK, well found.

In parallel, at least in the case of LiberTEM-spawned clusters, we could try to prevent this by reducing the number of workers we create on systems with low RAM but many cores, and specifying a higher minimum memory allocation than the Dask default.

uellue, 2023-04-20:
> In parallel, at least in the case of LiberTEM-spawned clusters, we could try to prevent this by reducing the number of workers we create on systems with low RAM but many cores, and specifying a higher memory allocation than the Dask default.

I can include that with the cluster resources in #1353. For GPUs I have to adapt the cluster setup to the available memory anyway, and I could use the same logic to set up CPU workers.

uellue, 2025-05-15:
Still reproduced. Worker memory for test can be limited by adding `options={'memory_limit': "1 GB"}` to  https://github.com/LiberTEM/LiberTEM/blob/39ba3b5cdf10a41a3ddc72072639efda98259c03/conftest.py#L1042-L1047