LiberTEM/LiberTEM #1419: Configure `libertem-server` worker params from command line

matbryan52, 2023-04-21:
Right now we require the user to go through a create/connect page before loading data in `libertem-server`.

I would argue that in most cases people immediately click 'Connect' without looking at the content. We could offer the ability to pre-configure the cluster from the launch command, e.g.:

```bash
libertem-server -c 8 -g 1
```

which would create an 8 cpu / 1 gpu cluster and put you immediately on the select data file browser page. It might even be worth it to just create a maximum-resource cluster by default, and the current behaviour with the configuration page  becomes an option, too.

This could also extend to connecting to existing Dask clusters:

```bash
libertem-server --dask-scheduler 100.100.100.100:8080
```

The goal would to reduce the typical time-to-analysis.

matbryan52, 2023-08-10:
An additional feature here would be to actually expose the scheduler address when `libertem-server` creates a cluster, that way you have a one-command solution to set up a LiberTEM-configured cluster on a server somewhere, then connect to it remotely from another machine.

sk1p, 2023-08-15:
Agreed - defaulting to a max resource cluster is a good idea. We could then have a "reconfigure cluster" command in the status bar, to add/remove workers (which would in reality stand up a new cluster when applying).

With #1353 we should also have better defaults for the number of CUDA workers (and maybe detection of pinning via `taskset` or constrained docker CPU resources?), meaning one less reason to change the defaults.

PR #1535 (2023-10-27): Specify resources, load dataset from `libertem-server` CLI
Closes #1419 

Adds the following CLI options:

- `-c / --cpus <int>` specify the number of CPU workers
- `-g / --gpus <int>` specify the number of GPU workers
- `-o / --open-ds <path>` specify the dataset to open via URL fragment (#1518 )

Also improves the log message for the connection address (more prominent, adds a version with hostname), and improves the help text for `--insecure` (#1534 ).

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/mast [... cut]
closed 2023-10-27 by uellue