LiberTEM/LiberTEM #1469: Move grid matching code out of core LiberTEM?

sk1p, 2023-07-24:
The `gridmatching` (or more specifically, the `fullmatch`) code is the only place we depend on `hdbscan`. We have had a bunch of compatibility issues in the past, and it would be nice if normal LiberTEM development was not interrupted by those issues, and to rather contain the blast radius of such issues.

Should we move this code (including tests and examples) to another repository; maybe to `LiberTEM-blobfinder`?

sk1p, 2024-02-21:
@uellue @matbryan52 discussed today in the call

sk1p, 2024-02-21:
To elaborate, in addition to the mentioned compatibility issues, `hdbscan` also slows down our CI, as they don't publish wheels. `time pip install hdbscan` in a clean environment (and removing my local cached built wheel) takes ~30s, this is probably even slower on most of the CI runners.

uellue, 2024-02-21:
Currently working on a PR to move the code to Blobfinder. Looking good so far, checking docs etc.

PR #1600 (2024-04-18): Deprecate gridmatching and fullmatch
Depends on releasing
https://github.com/LiberTEM/LiberTEM-blobfinder/pull/83, will fail and should not be merged until then.

Closes #1469

The tests are removed, which allows removing the test fixtures and corresponding imports, which then allows testing for a `DeprecationWarning` on import.

## Contributor Checklist:

* [ ] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [ ] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/ [... cut]
closed 2024-04-18 by uellue