LiberTEM/LiberTEM #1476: Release v0.12

sk1p, 2023-07-26:
# Release checklist

Issues and pull requests to be considered for this release:

* #1441 
* #1458 
* #1475

## Before (using a release candidate package)

* [ ] Review open issues and pull requests
* [ ] License review: no import of GPL code from MIT code
      `pydeps --only "libertem" --show-deps --noshow src\libertem | python scripts\licensecheck.py`
* [ ] Run full CI pipeline, including slow tests, on [Azure DevOps](https://dev.azure.com/LiberTEM/LiberTEM/_build?definitionId=3)
* [ ] Run tests for related packages w/ new LiberTEM version
    * [ ] LiberTEM-live
    * [ ] LiberTEM-holo
    * [ ] LiberTEM-blobfinder
    * [ ] ptychography40
* [ ] Handle deprecation, search the code base for `DeprecationWarning`
      that are supposed to be removed in that release.
* [ ] GUI dependency update with `npm install`
* [ ] Review https://github.com/LiberTEM/LiberTEM/security/dependabot and update dependencies
* [ ] Full documentation review and update, including link check using
      ``sphinx-build -b linkcheck "docs/source" "docs/build/html"``
* [ ] Run complete test suite, including slow tests that are deactivated by default
      and tests that require sample files.
* [ ] Update the expected version in notes on changes, i.e. from `0.3.0.dev0`
      to `0.3.0` when releasing version 0.3.0.
* [ ] Update and review change log in `docs/source/changelog.rst`, merging
      snippets in `docs/source/changelog/*/` as appropriate.
* [ ] Update the JSON files in the ``packaging/`` folder with author and project information
* [ ] Edit `pytest.ini` to exclude flaky tests temporarily from release builds
* [ ] Create a release candidate using `scripts/release`. See `scripts/release --help` for details.
* [ ] `Confirm that wheel, tar.gz, and AppImage are built for the release candidate on
      GitHub <https://github.com/LiberTEM/LiberTEM/releases>`_
* [ ] Confirm that a new version with the most recent release candidate is created in the
      `Zenodo.org sandbox <https://sandbox.zenodo.org/record/367108>`_ that is ready for submission.
* [ ] Install release candidate packages in a clean environment
      (for example:
      `python -m pip install -i https://test.pypi.org/simple/ --extra-index-url https://pypi.org/simple 'libertem==0.2.0rc11'`)
* [ ] Test the release candidate docker image
* [ ] For the GUI-related items, open in an incognito window to start from a clean slate
* [ ] Correct version info displayed in info dialogue?
* [ ] Link check in version info dialogue
* [ ] Test GUI without internet access
* [ ] Make sure you have test files of all supported types available
    * [ ] Include floats, ints, big endian, little endian, complex raw data
* [ ] Open each test file
    * [ ] Are parameters recognized correctly, as far as implemented?
    * [ ] Any bad default values?
    * [ ] Does the file open correctly?
    * [ ] Have a look at the dataset info dialogue. Reasonable values?
* [ ] Perform all analyses on each test file.
    * [ ] Does the result change when the input parameters are changed?
    * [ ] All display channels present and looking reasonable?
    * [ ] Reasonable performance?
    * [ ] Use pick mode.
* [ ] Re-open all the files
    * [ ] Are the files listed in "recent files"?
    * [ ] Are the parameters filled from the cache correctly?
* [ ] Try opening all file types with wrong parameters
    * [ ] Proper understandable error messages?
* [ ] Pick one file and confirm keyboard and mouse interaction for all analyses
    * [ ] Correct bounds check for keyboard and mouse?
* [ ] Check what happens when trying to open non-existent files or directories in the GUI.
    * [ ] Proper understandable error message?
    * [ ] Possible to continue working?
* [ ] Shut down libertem-server while analysis is running
    * [ ] Shut down within a few seconds?
    * [ ] All workers reaped?
* [ ] Check what happens when trying to open non-existent files by scripting.
    * [ ] Proper understandable error message? TODO automate?
* [ ] Check what happens when opening all file types with bad parameters by scripting
    * [ ] Proper understandable error message? TODO automate?
* [ ] Run libertem-server on Windows, connect to a remote dask cluster running on Linux,
  open all file types and perform an analysis for each file type.
* [ ] Use the GUI while a long-running analysis is running
    * [ ] Still usable, decent response times?
* [ ] Confirm that pull requests and issues are handled as intended, i.e. milestoned and merged
  in appropriate branch.
* [ ] Final version bump: `./scripts/release bump v0.3.0 --tag`, push to github
* [ ] After pipeline finishes, write minimal release notes for the [release](https://github.com/liberTEM/LiberTEM/releases) and publish the GitHub release

## After releasing on GitHub

* [ ] Confirm that all release packages are built and release notes are up-to-date
* [ ] Install release package
* [ ] Confirm correct version info
* [ ] Confirm package upload to PyPI
* [ ] Confirm images and tags on https://hub.docker.com/r/libertem/libertem
* [ ] Publish new version on zenodo.org
* [ ] Update documentation with new links, if necessary
    * [ ] Add zenodo badge for the new release to Changelog page
* [ ] Send announcement message on mailing list
* [ ] Edit `pytest.ini` to include flaky tests again
* [ ] Bump version in master branch to next .dev0 (`./scripts/release bump v0.X.0.dev0 --commit`)
* [ ] Add to institutional publication databases
* [ ] Add the current LiberTEM version to [CVL](https://github.com/Characterisation-Virtual-Laboratory/CharacterisationVL-Software) - add both the singularity and the .desktop file!

sk1p, 2023-08-09:
v0.12.0.rc1 is in the oven: https://github.com/LiberTEM/LiberTEM/actions/runs/5811075507

sk1p, 2023-08-09:
> v0.12.0.rc1 is in the oven: https://github.com/LiberTEM/LiberTEM/actions/runs/5811075507

... and now available on https://test.pypi.org/project/libertem/

There's still a bug in the release script, that is hopefully fixed in #1485, which only affected tagging and pushing the container images: https://github.com/LiberTEM/LiberTEM/actions/runs/5811075507/job/15754094890 - rc1 should be ready for testing regardless.

sk1p, 2023-08-09:
Full CI run incl. slow tests here: https://dev.azure.com/LiberTEM/LiberTEM/_build/results?buildId=2477&view=results

matbryan52, 2023-08-10:
> ... Reviewing conda packages after release: ...

Conda-related actions should probably be included in the standard release checklist now (in the after section)

sk1p, 2023-08-10:
> Conda-related actions should probably be included in the standard release checklist now (in the after section)

Done in #1486

sk1p, 2023-08-10:
> License review: no import of GPL code from MIT code

There are two issues, mostly related to typing imports:

1) `libertem.common.executor` uses types from `libertem.udf.base` 
2) `libertem.common.progress` uses types from `libertem.udf.base` 

Might need to extract some interfaces to `libertem.common`. These are not new for 0.12 though, and I'm tempted to leave these as-is and fix them after the release. Opened https://github.com/LiberTEM/LiberTEM/issues/1487 for this.

sk1p, 2023-08-10:
>  Correct version info displayed in info dialogue?

The baked revision is `"unknown"` for some reason, also meaning the revision is not displayed in the web GUI - I'll debug this.

sk1p, 2023-08-10:
> Edit: seems to be a working-directory issue

... or, more specifically, the way new-style builds are run, they are isolated from the git repository, meaning when building the wheel, it's too late to check the revision, as that process is running somewhere else outside the repository. The sdist does contain the revision as expected. Hmm. I'd actually expect that the wheel is built using the sdist as input, but that doesn't seem to be the case, as disabling the revision baking for the wheel if the dest file exists doesn't fix the issue. Need to dive in a bit more...

matbryan52, 2023-08-10:
> Hmm. I'd actually expect that the wheel is built using the sdist as input...

I think you can specify different things to include in whl / sdist, so am not sure it works like that. At least when running `python -m build` the two stages are independent.

sk1p, 2023-08-10:
Yeah, currently (re?-)learning some of the packaging stuff, pep517 etc... maybe we can hook into the process earlier, before stuff is copied to the separate location.

Actually, it seems that the wheel _is_ built from the sdist (it's even called `/tmp/build-via-sdist-XXXX`), but the sdist is also subtly wrong... so when that's fixed, the wheel should be fine, too.

sk1p, 2023-08-10:
`baked_revision` fix in https://github.com/LiberTEM/LiberTEM/pull/1488

All of that said, I think it's time that we move our packaging to no longer use the legacy `setup.py` based process, as that is documented to be removed at some point in the future. Just a question what to use:

- Keep using `setuptools`, but with configuration in pyproject.toml?
- hatch? hatchling? (I think one is the front-end, the other the backend?)
- flit?
- poetry?!?
- pdm?

[Oof.](https://packaging.python.org/en/latest/key_projects/)

matbryan52, 2023-08-10:
> All of that said, I think it's time that we move our packaging to no longer use the legacy `setup.py` based process, as that is documented to be removed at some point in the future. Just a question what to use:

My initial feeling would be to first migrate as much as we can into `pyproject.toml` using `setuptools`, since that should be the path of least resistance /  minimal re-configuration, but if we see limitations (e.g. more and more of the code replaced by rust extensions ?) then consider something else ?

sk1p, 2023-08-10:
> My initial feeling would be to first migrate as much as we can into `pyproject.toml` using `setuptools`, since that should be the path of least resistance / minimal re-configuration, but if we see [...]

Sounds good, yeah. What I do see is that it could be that the `_baked_revision` stuff won't work as it does now, and maybe that's also a good thing. The information is technically redundant with the version tag, and I think the replacement is to have either the bare version tag, or a `0.XX.YY.dev0+some-revision` version scheme in development, which makes clear that there are commits on top of the version tag. Let's see how well `setuptools-scm` works these days.

About rust, I'm currently very comfortable with using maturin and separate packages for this, to completely separate these from the main build.

rc3 is now baking, and should have the revision stuff fixed.

matbryan52, 2023-08-10:
Not an issue _per se_, but I just had probably the longest numba warmup I've ever seen:

```
[2023-08-10 15:08:50,361] INFO [libertem.web.dataset.prime_numba_caches:100] warmup done, took 79.456s
```

after a fresh install on Windows 10, i5-1145G7, 16 GB RAM, NVMe disk. Subsequent warmups take about 15 seconds when selecting the same dataset in the web UI.

sk1p, 2023-08-10:
> Not an issue _per se_, but I just had probably the longest numba warmup I've ever seen:

[...] 

Oof! Is that with a virus scanner active? And if so, can you re-try with it deactivated?

matbryan52, 2023-08-10:
> Oof! Is that with a virus scanner active? And if so, can you re-try with it deactivated?

There is a scanner active, but no I can't deactivate it :( (work PC)

sk1p, 2023-08-14:
> * GUI dependency update with `npm install`

This will have to wait until after the release. It turns out that one of our main dependencies (`create-react-app`) hasn't seen an update in a while and looks abandoned upstream, so we need to move to something else. The react docs themselves mention next.js, but we don't really need a full framework, we only need the build system parts, like `create-react-app` provided.

The currently best option I see is to move to [vite](https://vitejs.dev/), which is advertised more or less as a (faster? easier to setup?) replacement for the webpack build stack we currently use. As far as I can see, that won't be the one with least code changes, but manually managing a webpack config (+ the rest of the build stack) would probably be more work over time.

sk1p, 2023-08-15:
rc4 is baking here: https://github.com/LiberTEM/LiberTEM/actions/runs/5865621227

uellue, 2023-08-15:
Tested the GUI Monte Carlo Style with different files and analyses, with and without CuPy. Looks good! In particular, the scheduling behavior under heavy load is much better than in the past. File browsing is snappy, and the available resources seem to be distributed equally between analyses running in parallel. Shutting down the server via the web GUI is also very responsive under load.

I'm not checking the items because it was a cursory test, but since there weren't so many changes I'd be confident to sign this off for this release.

matbryan52, 2023-08-16:
> Tested the GUI Monte Carlo Style with different files and analyses, with and without CuPy. Looks good! In particular, the scheduling behavior under heavy load is much better than in the past. File [...]

I also did a similar random selection of tests / files / multiple analyses at once (from a Windows machine connecting to Linux, and locally) and agree with all these points 👍

matbryan52, 2023-08-16:
Do we want to merge the PR with the changelog and misc updates ( #1486 ) before releasing ?

sk1p, 2023-08-16:
> Do we want to merge the PR with the changelog and misc updates ( #1486 ) before releasing ?

We can merge it before sending out the release notifications, and include the DOI badge, too.

closed 2023-08-17 by sk1p