LiberTEM/LiberTEM #1489: GUI support for multiple GPU workers per device

matbryan52, 2023-08-10:
We now support easy creation of multiple GPU workers per device through the `Context.make_with` endpoint. It would be useful to have the same thing for the web GUI for machines with a powerful card which won't be saturated by one worker process.

sk1p, 2023-08-10:
The UI could very roughly look like this:

- [ ] CUDA Device 0
- [x] CUDA Device 1 - number of workers: [    1 ]

With the "number of workers" numerical input being shown once the device is selected with the checkbox (maybe indented and below the row of the checkbox, though). Probably checkboxes checked by default, and a reasonable number of workers per cuda device? Which can also be improved upon once #1353 or it's replacement is landed, and we have some heuristic of choosing number of workers per device based on available memory.

sk1p, 2023-08-15:
Behavior, which should be encoded as test cases:

- output of the GPU selection widget is a list of device IDs, where the IDs can be repeated to represent multiple workers for one device
- deselecting a CUDA device removes the device ID from the  list of CUDA workers
- checking the checkbox again switches back to the previously selected number of workers for that specific device
- all checkboxes are disabled and unchecked in case `cupy` is not available, and the "field value" is an empty list in that case

sk1p, 2023-08-17:
I've settled on a simpler design:

[image]

... the checkboxes are not really necessary and complicate the state handling quite a bit. Better to implement it in a way that de-selecting a GPU is accomplished by setting its number of workers to zero.

PR #1499 (2023-08-21): GUI: multiple GPU workers per CUDA device
Fixes #1489 

~~Based on #1498 - will rebase once that's merged.~~

## TODO

- [x] Write/update tests for web API changes
- [x] Integrate [react-testing-library](https://www.robinwieruch.de/vitest-react-testing-library/) and write tests for form behavior, if possible
- [x] Finish implementing full behavior described in #1489 (especially allowing "empty string" intermediate state)

## Contributor Checklist:

* [ ] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x]  [... cut]
closed 2023-08-21 by sk1p