LiberTEM/LiberTEM #1490: Notebook export bug (from upstream `autopep8`)

matbryan52, 2023-08-10:
We're affected by this issue in `autopep8` when exporting a notebook:

https://github.com/hhatto/autopep8/issues/689

specifically in:
 
```
  File "D:\...\envs\lt_release012\Lib\site-packages\libertem\web\notebook_generator\template.py", line 36, in code_formatter
    return autopep8.fix_code(code)
           ^^^^^^^^^^^^^^^^^^^^^^^
```

From activity on the issue it looks like it will soon be handled ?

matbryan52, 2023-08-10:
Seems like a solution is to pin `pycodestyle` to 2.10.0.

sk1p, 2023-08-10:
> Seems like a temporary solution is to pin `pycodestyle` to 2.10.0

Yeah, probably `<2.11` is the correct pin, to not be overly strict

sk1p, 2023-08-10:
I'll take care of this one - theoretically the notebook tests should also reproduce the issue, I'll see if that is the case

PR #1492 (2023-08-10): Pin pycodestyle
Fixes #1490; workaround for https://github.com/hhatto/autopep8/issues/689

## Contributor Checklist:

* [ ] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [ ] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [ ] I have added/updated documentation for all user-facing changes
* [ ] I have added/updated test cases
* [ ] I have included the [rebuilt production build of the client](https://libertem.github.io/LiberTEM/ [... cut]
commit 5b77b09b (2023-08-10): Pin pycodestyle

Fixes #1490; workaround for https://github.com/hhatto/autopep8/issues/689
closed 2023-08-10 by sk1p