LiberTEM/LiberTEM #1491: Web GUI fails to open folder containing shortcuts on Windows with bad permissions

matbryan52, 2023-08-10:
I stumbled across a (very) edge case for the web GUI - trying to open a folder on Windows which contains a shortcut to a directory to which I don't have access, I get the following UI error. It is possible to close the error and continue working, but not to access the directory.

```
Error browsing directory: SyntaxError: Unexpected token '<', "<html><tit"... is not valid JSON
```

and in the terminal I get the following exception:

```
2023-08-10 14:50:54,569 - tornado.application - ERROR - Uncaught exception GET /api/browse/localfs/?path=C%3A%5CUsers%5Cmb265392%2F.. (::1)
HTTPServerRequest(protocol='http', host='localhost:9000', method='GET', uri='/api/browse/localfs/?path=C%3A%5CUsers%...%2F..', version='HTTP/1.1', remote_ip='::1')
Traceback (most recent call last):
  File "D:\...\envs\lt_release012\Lib\site-packages\tornado\web.py", line 1786, in _execute
    result = await result
             ^^^^^^^^^^^^
  File "D:\...\envs\lt_release012\Lib\site-packages\libertem\web\browse.py", line 19, in get
    listing = await executor.run_function(get_fs_listing, path)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\...\envs\lt_release012\Lib\site-packages\libertem\executor\base.py", line 87, in run_function
    return await sync_to_async(fn_with_args, self._pool)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\...\envs\lt_release012\Lib\site-packages\libertem\common\async_utils.py", line 58, in sync_to_async
    return await loop.run_in_executor(pool, fn)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\...\envs\lt_release012\Lib\concurrent\futures\thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\...\envs\lt_release012\Lib\site-packages\libertem\common\async_utils.py", line 27, in _wrapped
    return f(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^
  File "D:\...\envs\lt_release012\Lib\site-packages\libertem\executor\dask.py", line 598, in run_function
    return future.result()
           ^^^^^^^^^^^^^^^
  File "D:\...\envs\lt_release012\Lib\site-packages\distributed\client.py", line 327, in result
    raise exc.with_traceback(tb)
  File "D:\...\envs\lt_release012\Lib\site-packages\libertem\io\fs.py", line 82, in get_fs_listing
    s = os.stat(full_path)
    ^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Access is denied: 'C:\\Users\\Username.old'
```

sk1p, 2023-08-10:
It's weird that windows doesn't even allow you to `stat` that shortcut - and as `get_fs_listing`  includes the `stat` result object in the returned dicts, it's not so easy to include an entry for those files we can't even `stat`.

A quick fix would be to catch `PermissionError` here:

https://github.com/LiberTEM/LiberTEM/blob/0f92e0928c6980fb7a4ae292484e2c9a0894471d/src/libertem/io/fs.py#L79-L99

... and just `continue`, like we do for the suspected TOCTOU-stuff, meaning these shortcuts would just be hidden.

matbryan52, 2023-08-10:
> A quick fix would be to catch `PermissionError` here:
> ...
> ... and just `continue`, like we do for the suspected TOCTOU-stuff, meaning these shortcuts would just be hidden.

Agreed, want to include it in this release or consider it a bizarre edge case for later ?

sk1p, 2023-08-10:
> Agreed, want to include it in this release or consider it a bizarre edge case for later ?

If possible let's fix this now - I think otherwise it'll be forgotten.

matbryan52, 2023-08-10:
Fix works.

Though, it's actually more complex than I thought, this folder is not a shortcut but specifically a `.old` folder which is created when a Windows account is moved (it just has a shortcut icon). Normal shortcuts are just recognized as a `.lnk` file by the web gui, even to no-permission locations. (It then tries to load them as a RAW dataset :(

We could perhaps look into recognising shortcuts to other directories and allowing the user to follow them in the UI, but that's for another Issue.

sk1p, 2023-08-10:
> Though, it's actually more complex than I thought, this folder is not a shortcut but specifically a `.old` folder which is created when a Windows account is moved (it just has a shortcut icon). 

Interesting - so probably not so easy to test in CI, which I already suspected...

> Normal shortcuts are just recognized as a `.lnk` file by the web gui, even to no-permission locations. (It then tries to load them as a RAW dataset :(

Yeah, it's a bummer links like that aren't really integrated into the file system on Windows, like they are on sane operating systems (IIRC windows does have soft link support in NTFS, but that's not what's exposed to the users...)

> We could perhaps look into recognising shortcuts to other directories and allowing the user to follow them in the UI, but that's for another Issue.

It's a bit ugly, but possible to follow these [using the `win32com.client` module](https://stackoverflow.com/a/571573/540644) - just as a note for anyone who wants to pick this up :)

PR #1493 (2023-08-14): Catch `PermissionError` when doing `os.stat()` on file or directory
Fixes #1491 

Adding a test case for this is nontrivial, still thinking about it.

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [ ] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [ ] I have added/updated documentation for all user-facing changes
* [ ] I have added/updated test cases
* [ ] I have included the [rebuilt producti [... cut]