LiberTEM/LiberTEM #1501: Focus and planning for the v0.13 release

sk1p, 2023-08-15:
As you might have noticed, I worked on getting the GUI dependencies updated in #1498, so it makes sense to include a few GUI improvements in the next release. Namely:

- #1498
- #1499
- #1482 
- #1500 
- #1464 
- maybe adding support for opening data sets that consist of multiple files from the GUI? Including a proper API for #1224 

On the other hand, there's still some improvements for the live processing pipeline I'd like to work on, at least:

- #1484 

If we are working on the multi-file support, we might want to revive #1376 too, since that may also touch the data set loading APIs etc.

In addition, I want to continue the work on GUI integrations, like PR, and/or the jupyter-based work of @matbryan52 

Thoughts? Other ideas/priorities?

matbryan52, 2023-08-16:
In a similar theme (server / QoL) I'd add #1419 

More generally quality-of-life and ease-of-use might be worth some focus, in the style of #1379,  #1443. The GUI stuff fits into this, but also documentation, examples and simplifying APIs. Speaking to people around me I think we still suffer from the perception that LiberTEM is too complicated to use!

uellue, 2023-10-25:
What about releasing 0.13 with the current features, namely descan error compensation, progress bar, multiple GUI GPU workers, dm4 conversion and opening files through URL parameters? If you agree, I'd defer the rest to 0.14.

uellue, 2023-10-25:
Is #1224 ready to go, by the way? We seem to be kicking the ball forward here. :-D

sk1p, 2023-10-25:
> What about releasing 0.13 with the current features, namely descan error compensation, progress bar, multiple GUI GPU workers, dm4 conversion and opening files through URL parameters? If you agree, I'd defer the rest to 0.14.

I'm all for releasing more often!

> Is https://github.com/LiberTEM/LiberTEM/pull/1224 ready to go, by the way? We seem to be kicking the ball forward here. :-D

Should be kicked further IMHO, unless we have an urgent need to merge it now.

matbryan52, 2023-10-25:
Agreed that a release would be good.

I am however just working on a quick PR for the CLI to complement the dataset autoloading in the web interface. Hopefully I can get that up today.

uellue, 2023-11-07:
Closing here in favor of release issue #1542!

closed 2023-11-07 by uellue