LiberTEM/LiberTEM #1508: UDFs with multiple implementations / `UDF.get_method()` discrepancy

matbryan52, 2023-09-09:
As the UDF interface is currently written, a UDF has two ways to signal which processing method it implements:

https://github.com/LiberTEM/LiberTEM/blob/d839573b2752b26738df051177aaa8c5f6f3d964/src/libertem/common/udf.py#L67-L68

which is mostly used during tileshape negotiation. The default implementation is this:

https://github.com/LiberTEM/LiberTEM/blob/d839573b2752b26738df051177aaa8c5f6f3d964/src/libertem/udf/base.py#L1099-L1107

but also implicitly by implementing one of the processing `Protocol` (`UDFFrameMixin`, `UDFTileMixin`, `UDFPartitionMixin`). This second mechanism is used at runtime via an `isinstance` check in `UDFPartRunner._run_tile` to actually choose which method to call:

https://github.com/LiberTEM/LiberTEM/blob/d839573b2752b26738df051177aaa8c5f6f3d964/src/libertem/udf/base.py#L2104-L2120

I was caught out by this when working on #1304 writing a UDF which implements both `process_tile` and `process_frame` - I had assumed that by overriding `get_method` I could influence which processing method would be used based on runtime information (`self.params`, dataset shape etc) . But in fact because we order the `isinstance` checks as `tile`, `frame`, `partition`, this UDF always runs as `UDFTileMixin` independently of what `get_method` returns.

This also links to #1506 where a UDF might have multiple implementations depending on the runtime context (for performance or algorithmic reasons).

I think we could find a non-breaking way to both reconcile the possible discrepancy between `get_method` and the runtime protocol check, and at the same time allow a UDF to influence how it is used at runtime.

One possibility is to actually use `UDF.get_method()` inside `UDFPartRunner._run_tile`, and move the type checking onto the default implementation of `get_method()` itself, something like this:

```python
    def get_method(self) -> Literal['tile', 'frame', 'partition']:
        if isinstance(self, UDFTileMixin):
            return 'tile'
        elif isinstance(self, UDFFrameMixin):
            return 'frame'
        elif isinstance(self, UDFPartitionMixin):
            return 'partition'
        else:
            raise TypeError("UDF should implement one of the `process_*` methods")
```

then do a string comparison (or better still put our processing methods into an `Enum`) inside `UDFPartRunner._run_tile`.

We could also memoize the processing method somewhere during initialization, which would reduce the overhead from a lot of `isinstance` calls (small but in a large dataset with lots of tiles it could add up!).

There might be side-effects I haven't thought of, however! If not, I'll implement and use as the base for #1304.

sk1p, 2023-09-11:
> I was caught out by this when working on #1304 writing a UDF which implements both `process_tile` and `process_frame` - I had assumed that by overriding `get_method` I could influence which [...]

`get_method` is an implementation detail and not meant to be overridden. The [documented behavior](https://libertem.github.io/LiberTEM/udf/advanced.html#precedence) is indeed that you override one of the `process_*` methods, and if you override multiple, the method is chosen by precedence.

So:

> a UDF has two ways to signal which processing method it implements:[...]

Not really, implementing one of the protocols is the only documented way.

> I think we could find a non-breaking way to both reconcile the possible discrepancy between get_method and the runtime protocol check, and at the same time allow a UDF to influence how it is used at runtime.

Yes, because `get_method` is not part of the user-facing API, allowing multiple `process_*` methods is a feature addition and we can fully decide now how this should be handled. We can either improve/replace/... and document `get_method`, or implement this feature in any other shape we can come up with. In any way, if the user doesn't explicitly select a method, we can still fall back to the documented and backwards-compatible precedence scheme.

> One possibility is to actually use UDF.get_method() inside UDFPartRunner._run_tile, and move the type checking onto the default implementation of get_method() itself, something like this: [...]

The reason `_run_tile` is built like it is currently (I think you noticed this in the PR, I didn't have a detailed look yet) is type safety - using an `isinstance` means we get the `hasattr`-like behavior combined with statically checkable types. IIRC it used to call `get_method` and dispatch on the string result, which of course is not type safe at all.

> We could also memoize the processing method somewhere during initialization, which would reduce the overhead from a lot of isinstance calls (small but in a large dataset with lots of tiles it could add up!).

Agreed, we should be conscious of  per-frame/per-tile overheads here.

matbryan52, 2023-09-12:
> The reason `_run_tile` is built like it is currently (I think you noticed this in the PR, I didn't have a detailed look yet) is type safety - using an `isinstance` means we get the `hasattr`-like [...]

Makes sense, and I appreciated the way the `@runtime_checkable` was used to implement that!

In the PR the default `get_method` performs the same type check on behalf of `UDFPartRunner`, but `mypy` can't follow this through the dispatch in `_run_tile`. To satisfy the type checker I had to add an additional `assert isinstance(udf, Mixin)` after the dispatch.

I also added an addition check on the main node during `_prepare_run_for_dataset` that `get_method` returns a valid, implemented method, but clearly `mypy` can't follow that through, either. But this does mean that the UDF is rejected early based on the type check before we send out tasks (previously `get_method` would raise `TypeError` if the correct attribute was not present but didn't actually check against the prototols.

closed 2023-09-14 by sk1p