LiberTEM/LiberTEM #1534: `libertem-server` CLI `--insecure` implies `-h 0.0.0.0`

matbryan52, 2023-10-25:
As-is, using the `--insecure` flag doesn't actually allow you to connect to the `libertem-server` instance from other addresses than localhost. You have to combine it with `-h 0.0.0.0` to accept any connection or specify another fixed address.

Is there a reason why `--insecure` and `host='localhost'` (the default) cannot imply `host=0.0.0.0`, to save the user from having to know they need to modify the host parameter? My understanding of `--insecure` is that "I trust my network, I don't need token auth, I just want to look at my data", so it seems like a way to make accessing `libertem-server` a bit easier.

matbryan52, 2023-10-25:
The current help text for `--insecure`:

```
  --insecure                      allow to bind to non-localhost without token auth
```

which I don't think is actually true in the current implementation.

sk1p, 2023-10-25:
> which I don't think is strictly true in the current implementation (or could do with clarification).

It means that binding to non-localhost requires either token auth or `--insecure`:

```
$ libertem-server -h 0.0.0.0
Usage: libertem-server [OPTIONS]
Try 'libertem-server --help' for help.

Error: listening on non-localhost 0.0.0.0:9000 currently requires token authentication or --insecure
```

>Is there a reason why `--insecure` and `host='localhost'` (the default) cannot imply `host=0.0.0.0`, to save the user from having to know they need to modify the host parameter? My understanding of `--insecure` is that "I trust my network, I don't need token auth [or can't implement it locally], I just want to look at my data on my server".

Hmm, I think the reason is, I wanted this configuration to be as explicit as possible. `--insecure` means "allow an insecure configuration", it does not imply this configuration. I could be convinced that I'm too cautious, or maybe to add an explicit option `--insecure-listen-all` (I'm having a deja vu, we might have discussed something when we added the flag?)

sk1p, 2023-10-25:
> I've added more detail to the help message here: [e621c45](https://github.com/LiberTEM/LiberTEM/commit/e621c4538666a384862e75be89acd4b40060e281) , open as to the wording!
> 
> ```
>                   "Allow connections from non-localhost without token authorization. "
>                   "Applies only when combined with --host <address> "
>                   "(use `--insecure -h 0.0.0.0` to accept any connection)"
> ```

:+1: that works for me!

closed 2023-10-27 by matbryan52