LiberTEM/LiberTEM #1542: Release v0.13.0

uellue, 2023-11-07:
# Release checklist

Issues and pull requests to be considered for this release:

* #1540
* #1541 

## Step 0

* [ ] Create a release candidate using `scripts/release`. See `scripts/release --help` for details.

## Before (using a release candidate package)

* [x] Review open issues and pull requests
* [ ] License review: no import of GPL code from MIT code
      `pydeps --only "libertem" --show-deps --noshow src\libertem | python scripts\licensecheck.py`
* [ ] Run full CI pipeline, including slow tests, on [Azure DevOps](https://dev.azure.com/LiberTEM/LiberTEM/_build?definitionId=3)
* [ ] Run tests for related packages w/ new LiberTEM version
    * [ ] LiberTEM-live
    * [ ] LiberTEM-holo
    * [ ] LiberTEM-blobfinder
    * [ ] ptychography40
* [ ] Handle deprecation, search the code base for `DeprecationWarning`
      that are supposed to be removed in that release.
* [ ] GUI dependency update with `npm install`
* [ ] Review https://github.com/LiberTEM/LiberTEM/security/dependabot and update dependencies
* [ ] Full documentation review and update, including link check using
      ``sphinx-build -b linkcheck "docs/source" "docs/build/html"``
* [ ] Run complete test suite, including slow tests that are deactivated by default
      and tests that require sample files.
* [ ] Update the expected version in notes on changes, i.e. from `0.3.0.dev0`
      to `0.3.0` when releasing version 0.3.0.
* [ ] Update and review change log in `docs/source/changelog.rst`, merging
      snippets in `docs/source/changelog/*/` as appropriate.
* [ ] Update the JSON files in the ``packaging/`` folder with author and project information
* [ ] Edit `pytest.ini` to exclude flaky tests temporarily from release builds
* [ ] `Confirm that wheel, tar.gz, and AppImage are built for the release candidate on
      GitHub <https://github.com/LiberTEM/LiberTEM/releases>`_
* [ ] Confirm that a new version with the most recent release candidate is created in the
      `Zenodo.org sandbox <https://sandbox.zenodo.org/record/367108>`_ that is ready for submission.
* [ ] Install release candidate packages in a clean environment
      (for example:
      `python -m pip install -i https://test.pypi.org/simple/ --extra-index-url https://pypi.org/simple 'libertem==0.2.0rc11'`)
* [ ] Test the release candidate docker image
    * [ ] Confirm rc images and tags on https://ghcr.io/libertem/libertem
* [ ] For the GUI-related items, open in an incognito window to start from a clean slate
* [ ] Correct version info displayed in info dialogue?
* [ ] Link check in version info dialogue
* [ ] Test GUI without internet access
* [ ] Make sure you have test files of all supported types available
    * [ ] Include floats, ints, big endian, little endian, complex raw data
* [ ] Open each test file
    * [ ] Are parameters recognized correctly, as far as implemented?
    * [ ] Any bad default values?
    * [ ] Does the file open correctly?
    * [ ] Have a look at the dataset info dialogue. Reasonable values?
* [ ] Perform all analyses on each test file.
    * [ ] Does the result change when the input parameters are changed?
    * [ ] All display channels present and looking reasonable?
    * [ ] Reasonable performance?
    * [ ] Use pick mode.
* [ ] Re-open all the files
    * [ ] Are the files listed in "recent files"?
    * [ ] Are the parameters filled from the cache correctly?
* [ ] Try opening all file types with wrong parameters
    * [ ] Proper understandable error messages?
* [ ] Pick one file and confirm keyboard and mouse interaction for all analyses
    * [ ] Correct bounds check for keyboard and mouse?
* [ ] Check what happens when trying to open non-existent files or directories in the GUI.
    * [ ] Proper understandable error message?
    * [ ] Possible to continue working?
* [ ] Shut down libertem-server while analysis is running
    * [ ] Shut down within a few seconds?
    * [ ] All workers reaped?
* [ ] Check what happens when trying to open non-existent files by scripting.
    * [ ] Proper understandable error message? TODO automate?
* [ ] Check what happens when opening all file types with bad parameters by scripting
    * [ ] Proper understandable error message? TODO automate?
* [ ] Run libertem-server on Windows, connect to a remote dask cluster running on Linux,
  open all file types and perform an analysis for each file type.
* [ ] Use the GUI while a long-running analysis is running
    * [ ] Still usable, decent response times?
* [ ] Confirm that pull requests and issues are handled as intended, i.e. milestoned and merged
  in appropriate branch.
* [ ] Final version bump: `./scripts/release bump v0.3.0 --tag`, push to github
* [ ] After pipeline finishes, write minimal release notes for the [release](https://github.com/liberTEM/LiberTEM/releases) and publish the GitHub release

## After releasing on GitHub

* [ ] Confirm that all release packages are built and release notes are up-to-date
* [ ] Install release package
* [ ] Confirm correct version info
* [ ] Confirm package upload to PyPI
* [ ] Confirm images and tags on https://ghcr.io/libertem/libertem
* [ ] Publish new version on zenodo.org
* [ ] Update documentation with new links, if necessary
    * [ ] Add zenodo badge for the new release to Changelog page
* [ ] Send announcement message on mailing list
* [ ] Edit `pytest.ini` to include flaky tests again
* [ ] Bump version in master branch to next .dev0 (`./scripts/release bump v0.X.0.dev0 --commit`)
* [ ] Add to institutional publication databases
* [ ] Add the current LiberTEM version to [CVL](https://github.com/Chasdfracterisation-Virtual-Laboratory/CharacterisationVL-Software>) - add both the singularity and the .desktop file!
* [ ] Conda packaging: review PRs on https://github.com/conda-forge/libertem-feedstock/pulls

sk1p, 2023-11-07:
Full pipeline run @ https://dev.azure.com/LiberTEM/LiberTEM/_build/results?buildId=2557&view=results

sk1p, 2023-11-08:
rc1 is baking: https://github.com/LiberTEM/LiberTEM/actions/runs/6799706310

uellue, 2023-11-08:
The Zenodo sandbox is unstable currently. The RC doesn't show up there.

uellue, 2023-11-08:
Exception in https://github.com/LiberTEM/LiberTEM/actions/runs/6799706310/job/18487715324#step:7:3447:

```
Traceback (most recent call last):
  File "/home/runner/work/LiberTEM/LiberTEM/scripts/zenodo_upload", line 389, in main
    r = new_version(parent_id=parent)
  File "/home/runner/work/LiberTEM/LiberTEM/scripts/zenodo_upload", line 158, in new_version
    raise ZenodoException(prettystring(r))
ZenodoException: Status code: 400 
{'message': 'Referer checking failed - no Referer.', 'status': 400}

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/runner/work/LiberTEM/LiberTEM/scripts/zenodo_upload", line 394, in main
    r = get_deposition(deposition_id=parent)
  File "/home/runner/work/LiberTEM/LiberTEM/scripts/zenodo_upload", line 143, in get_deposition
    raise ZenodoException(prettystring(r))
ZenodoException: Status code: 403 
{'message': 'Permission denied.', 'status': 403}
done
```

uellue, 2023-11-08:
The Zenodo API didn't change since LiberTEM started: https://developers.zenodo.org/#changes

For that reason I'd assume that we can go ahead with the release. The main Zenodo page should behave as before.

uellue, 2023-11-08:
Python 3.8 with Cupy on Win 11:

```
================================== FAILURES ===================================
_____________________________ test_cancel_udf_job _____________________________

base_url = 'http://127.0.0.1:60960'
default_raw = <RawFileDataSet of float32 shape=(16, 16, 128, 128)>
http_client = <aiohttp.client.ClientSession object at 0x000001AFB3A05D90>
server_port = 60960
shared_state = <libertem.web.state.SharedState object at 0x000001AFBEDA2430>
local_cluster_url = 'tcp://localhost:57673', default_token = 'something_random'

    @pytest.mark.asyncio
    async def test_cancel_udf_job(
        base_url, default_raw, http_client, server_port, shared_state, local_cluster_url, default_token,
    ):
        await create_connection(base_url, http_client, local_cluster_url, default_token)
    
        print("checkpoint 1")
    
        # connect to ws endpoint:
        ws_url = f"ws://127.0.0.1:{server_port}/api/events/?token={default_token}"
        async with websockets.connect(ws_url) as ws:
            print("checkpoint 2")
            initial_msg = json.loads(await ws.recv())
            assert_msg(initial_msg, 'INITIAL_STATE')
    
            ds_uuid = "ae5d23bd-1f2a-4c57-bab2-dfc59a1219f3"
            ds_url = "{}/api/datasets/{}/?token={}".format(
                base_url, ds_uuid, default_token,
            )
    
            ds_data = _get_ds_params()
            async with http_client.put(ds_url, json=ds_data) as resp:
                assert resp.status == 200
                resp_json = await resp.json()
                assert_msg(resp_json, 'CREATE_DATASET')
    
            # same msg via ws:
            msg = json.loads(await ws.recv())
            assert_msg(msg, 'CREATE_DATASET')
    
            ca_uuid, ca_url = await create_update_compound_analysis(
                ws, http_client, base_url, ds_uuid, token=default_token,
            )
    
            analysis_uuid, analysis_url = await create_analysis(
                ws, http_client, base_url, ds_uuid, ca_uuid, token=default_token,
            )
    
            job_uuid, job_url = await create_job_for_analysis(
                ws, http_client, base_url, analysis_uuid, token=default_token,
            )
    
            await asyncio.sleep(0)  # for debugging, set to >0
    
            async with http_client.delete(job_url) as resp:
                assert resp.status == 200
                assert_msg(await resp.json(), 'CANCEL_JOB')
    
            # wait for CANCEL_JOB message:
            done = False
            num_seen = 0
            while not done:
                msg = json.loads(await ws.recv())
                num_seen += 1
                if msg['messageType'] == 'TASK_RESULT':
                    assert_msg(msg, 'TASK_RESULT')
                    assert msg['job'] == job_uuid
                elif msg['messageType'] == 'JOB_PROGRESS':
                    assert msg['job'] == job_uuid
                elif msg['messageType'] == 'CANCEL_JOB':
                    # this is the confirmation sent from the DELETE method handler:
                    assert_msg(msg, 'CANCEL_JOB')
                    assert msg['job'] == job_uuid
                    done = True
                else:
                    raise Exception("invalid message type: {}".format(msg['messageType']))
                if 'followup' in msg:
                    for i in range(msg['followup']['numMessages']):
                        # drain binary messages:
                        msg = await ws.recv()
    
>           assert num_seen < 4
E           assert 5 < 4

tests\server\test_cancel.py:103: AssertionError
```

uellue, 2023-11-08:
> Python 3.8 with Cupy on Win 11:

Very similar failure with Python 3.9, 3.10, 3.11.

uellue, 2023-11-08:
It doesn't happen if I run the test alone.

sk1p, 2023-11-09:
Release is baking: https://github.com/LiberTEM/LiberTEM/actions/runs/6812923418

sk1p, 2023-11-09:
Noticed directly after release: https://github.com/LiberTEM/LiberTEM/pull/1554 - we'll need a point release directly afterwards.

sk1p, 2023-11-09:
conda-forge PR for 0.13.0: https://github.com/conda-forge/libertem-feedstock/pull/4

sk1p, 2024-01-22:
CVL: https://github.com/Characterisation-Virtual-Laboratory/CharacterisationVL-Software/pull/229

closed 2024-01-22 by sk1p