LiberTEM/LiberTEM #1544: Make sure DECTRIS compound HDF5 files work

uellue, 2023-11-08:
Make sure https://er-c-data-mgmt.fz-juelich.de/browse/?path=/adhoc/arina_cryo4d_stem/maxleotest_master.h5 works in the web GUI without having to specify the dataset path.

Also, we could consider adding hdf5plugin as a dependency and not only an extra since datasets acquired with DECTRIS camera default settings require it.

sk1p, 2023-11-08:
Just tested this with h5py versions 3.0 to current, and all of them seem to fail to "see" / visit the dataset correctly.

uellue, 2023-12-18:
It seems that the master file just contains a list of separate dataset entries for each of the compound files:

```
 - entry : <HDF5 group "/entry" (4 members)>
		 - data : <HDF5 group "/entry/data" (105 members)>
			 - data_000001 : <HDF5 dataset "data": shape (10000, 96, 96), type "<u2">
					 - image_nr_high : 10000
					 - image_nr_low : 1
			 - data_000002 : <HDF5 dataset "data": shape (10000, 96, 96), type "<u2">
					 - image_nr_high : 20000
					 - image_nr_low : 10001
			 - data_000003 : <HDF5 dataset "data": shape (10000, 96, 96), type "<u2">
					 - image_nr_high : 30000
					 - image_nr_low : 20001
			 - data_000004 : <HDF5 dataset "data": shape (10000, 96, 96), type "<u2">
					 - image_nr_high : 40000
					 - image_nr_low : 30001
```

The total scan is 1024x1024, and it looks like stacking all these separate entries together would give you that. I couldn't find an entry that combines the files. I'll try a few things to see if we can make it work somehow.

uellue, 2023-12-18:
https://www.dectris.com/en/support/downloads/header-docs/hdf5/

uellue, 2023-12-18:
https://github.com/h5py/h5py/blob/master/examples/eiger_use_case.py

uellue, 2023-12-18:
Following this example, I got it to work by creating a proper virtual HDF5 dataset for the whole thing.

Code to create the master entry: [`create_true_master.ipynb`](https://er-c-data-mgmt.fz-juelich.de/browse/?path=/adhoc/arina_cryo4d_stem/create_true_master.ipynb)

Master entry: [master.h5](https://er-c-data-mgmt.fz-juelich.de/browse/?path=/adhoc/arina_cryo4d_stem/master.h5)

To open in LiberTEM, open with 1024x1024 nav shape.

uellue, 2023-12-18:
In conclusion, it could be worth it to explore an Eiger-NeXus dataset derived from HDF5 that creates the temporary HDF5 virtual dataset in a temporary file and interprets the pixel mask and flatfield correction.

For the time being we can build these virtual datasets with the code in the notebook in a separate file for each acquisition.