LiberTEM/LiberTEM #1558: Event loop teardon problem during testing (py3.8)

matbryan52, 2023-11-11:
Noticed while testing the conda-forge PR (https://github.com/conda-forge/libertem-feedstock/pull/5) but actually is present in a normal test run here on GH. I think we haven't seen this because we don't test python 3.8 as part of the normal Github actions :+1: 

https://github.com/LiberTEM/LiberTEM/blob/9f2d912e51db301a9f77a915799980a911391697/.github/workflows/ci.yml#L19-L21

I ran the CI on Python 3.8 in my fork and have the same failure there as with conda-forge:

https://github.com/matbryan52/LiberTEM/actions/runs/6833496847/job/18585312301?pr=7

Not sure why yet, though.

matbryan52, 2023-11-11:
```
FAILED tests/template/test_fft_template.py::test_fft_analysis - RuntimeError: Event loop is closed
FAILED tests/template/test_fft_template.py::test_pick_fft_analysis - RuntimeError: Event loop is closed
= 2 failed, 1966 passed, 652 skipped, 900 deselected, 10 xfailed, 23581 warnings in 303.72s (0:05:03)
```

during notebook preprocessing:

```
    def test_pick_fft_analysis(hdf5_ds_2, tmpdir_factory, lt_ctx, local_cluster_url):
        datadir = tmpdir_factory.mktemp('template_tests')
    
        conn = {'connection': {'type': 'tcp', 'address': local_cluster_url}}
        path = hdf5_ds_2.path
        dataset = _get_hdf5_params(path)
    
        params = {
            'x': 4,
            'y': 4,
            'real_rad': 4,
            'real_centerx': 8,
            'real_centery': 8,
        }
    
        analysis = [{
                "analysisType": 'PICK_FFT_FRAME',
                "parameters": params,
        }]
    
        notebook = notebook_generator(conn, dataset, analysis, save=True)
        notebook = io.StringIO(notebook.getvalue())
        nb = nbformat.read(notebook, as_version=4)
        ep = ExecutePreprocessor(timeout=600)
>       ep.preprocess(nb, {"metadata": {"path": datadir}})
```

sk1p, 2023-11-11:
Hm, I remember some issues with this before - it was related to nested asyncio things IIRC. Maybe changes in `pytest-asyncio` event loop handling since the last release have broken this?

sk1p, 2023-11-11:
I was trying to reproduce this locally, and running just `tests/template/` takes a loong time. Right now it's stuck in teardown of the shared local dask cluster, I'm not sure it'll ever recover even:

```
Thread 0x7F28A0FDE740 (idle): "MainThread"
    wait (threading.py:306)
    wait (threading.py:558)
    wait (distributed/utils.py:403)
    sync (distributed/utils.py:414)
    sync (distributed/utils.py:351)
    close (distributed/deploy/cluster.py:223)
    local_cluster_url (conftest.py:1060)
    _teardown_yield_fixture (_pytest/fixtures.py:911)
    finish (_pytest/fixtures.py:1024)
```

matbryan52, 2023-11-11:
> Hm, I remember some issues with this before - it was related to nested asyncio things IIRC. Maybe changes in `pytest-asyncio` event loop handling since the last release have broken this?

What I can't understand is how this would be a Python 3.8-only issue, both on conda-forge and here on GH actions 3.7, 3.9/10/11 are all fine.

sk1p, 2023-11-11:
> What I can't understand is how this would be a Python 3.8-only issue - both on conda-forge and here on GH actions 3.7, 3.9/10/11 are all fine.

Yeah, that is weird... 3.9 and 3.8 are using the same versions of `pytest-asyncio` and `nest-asyncio`, which would be suspicious candidates for causing this.

Here's a full diff of the versions installed by tox, as output by `pip freeze`:

```
--- /tmp/py38.txt	2023-11-11 21:58:29.849342973 +0100
+++ /tmp/py39.txt	2023-11-11 21:58:26.221302823 +0100
@@ -1,11 +1,10 @@
-y38: commands[0]> pip freeze
+py39: commands[0]> pip freeze
 aiohttp==3.8.6
 aiosignal==1.3.1
 asttokens==2.4.1
 async-timeout==4.0.3
 attrs==23.1.0
 autopep8==2.0.4
-backcall==0.2.0
 beautifulsoup4==4.12.2
 bleach==6.1.0
 bqplot==0.12.42
@@ -14,16 +13,16 @@
 click==8.1.7
 cloudpickle==3.0.0
 comm==0.2.0
-contourpy==1.1.1
+contourpy==1.2.0
 coverage==7.3.2
 cycler==0.12.1
 Cython==0.29.36
-dask==2023.5.0
+dask==2023.11.0
 debugpy==1.8.0
 decorator==5.1.1
 defusedxml==0.7.1
 Deprecated==1.2.14
-distributed==2023.5.0
+distributed==2023.11.0
 empyre==0.3.2
 exceptiongroup==1.1.3
 execnet==2.0.2
@@ -41,7 +40,7 @@
 importlib-resources==6.1.1
 iniconfig==2.0.0
 ipykernel==6.26.0
-ipython==8.12.3
+ipython==8.17.2
 ipywidgets==8.1.1
 jedi==0.19.1
 Jinja2==3.1.2
@@ -54,11 +53,11 @@
 jupyterlab-widgets==3.0.9
 kiwisolver==1.4.5
 lazy_loader==0.3
-libertem @ file:///home/alex/source/LiberTEM/.tox/.tmp/package/27/libertem-0.14.0.dev0.tar.gz#sha256=43b0ff4167ac6edaf589ffc31b08404517023dbd55c2e7bfbe4e8dfe5803ae79
+libertem @ file:///home/alex/source/LiberTEM/.tox/.tmp/package/28/libertem-0.14.0.dev0.tar.gz#sha256=77e22b0cda68a85a1213900f74adb475d45b149f24a400266a7124649b8174ab
 llvmlite==0.41.1
 locket==1.0.0
 MarkupSafe==2.1.3
-matplotlib==3.7.3
+matplotlib==3.8.1
 matplotlib-inline==0.1.6
 mistune==3.0.2
 msgpack==1.0.7
@@ -68,20 +67,18 @@
 nbformat==5.9.2
 ncempy==1.11.0
 nest-asyncio==1.5.8
-networkx==3.1
+networkx==3.2.1
 numba==0.58.1
-numexpr==2.8.5
-numpy==1.24.4
+numexpr==2.8.7
+numpy==1.26.1
 opentelemetry-api==1.21.0
 packaging==23.2
-pandas==2.0.3
+pandas==2.1.3
 pandocfilters==1.5.0
 parso==0.8.3
 partd==1.4.1
 pexpect==4.8.0
-pickleshare==0.7.5
 Pillow==10.0.1
-pkgutil_resolve_name==1.3.10
 platformdirs==4.0.0
 pluggy==1.3.0
 prompt-toolkit==3.0.40
@@ -100,14 +97,13 @@
 pytest-xdist==3.4.0
 python-dateutil==2.8.2
 pytz==2023.3.post1
-PyWavelets==1.4.1
 PyYAML==6.0.1
 pyzmq==25.1.1
 referencing==0.30.2
 rpds-py==0.12.0
-scikit-image==0.21.0
+scikit-image==0.22.0
 scikit-learn==1.3.2
-scipy==1.10.1
+scipy==1.11.3
 six==1.16.0
 sortedcontainers==2.4.0
 soupsieve==2.5
@@ -116,7 +112,7 @@
 stack-data==0.6.3
 tblib==3.0.0
 threadpoolctl==3.2.0
-tifffile==2023.7.10
+tifffile==2023.9.26
 tinycss2==1.2.1
 tomli==2.0.1
 toolz==0.12.0
```

sk1p, 2023-11-11:
I have a suspicion: the following change seems to cause similar behavior on Python 3.9:

```
diff --git a/test_requirements.txt b/test_requirements.txt
index 31fc98fb..e84d66b6 100644
--- a/test_requirements.txt
+++ b/test_requirements.txt
@@ -9,3 +9,5 @@ websockets
 pytest-benchmark
 ipykernel
 joblib
+dask==2023.5.0
+distributed==2023.5.0
```

sk1p, 2023-11-11:
So, it seems to be an interaction between `distributed` and `jupyter-core`; the following works around the issue:

```
diff --git a/test_requirements.txt b/test_requirements.txt
index 31fc98fb..d63bba80 100644
--- a/test_requirements.txt
+++ b/test_requirements.txt
@@ -9,3 +9,4 @@ websockets
 pytest-benchmark
 ipykernel
 joblib
+jupyter_core<5; python_version == '3.8'
```

(don't ask me what's going on - I just noticed that it was hanging (or just very slow) with a stack trace that includes `jupyter_core`...)

matbryan52, 2023-11-11:
> So, it seems to be an interaction between `distributed` and `jupyter-core`

Well done!

> (don't ask me what's going on - I just noticed that it was hanging (or just very slow) with a stack trace that includes `jupyter_core`...)

😁 Well spotted in any case. At least this should be possible to fix on the conda-forge PR without changing the sdist. Will try that in the morning.

sk1p, 2024-01-23:
As an update, I've changed a bit how the event loop is set up in testing (as part of #1572), as the newer versions of `pytest-asyncio` kept complaining about how we did things. Now, the `event_loop` fixture is only overridden on Python 3.7, as that is still on an older `pytest-asyncio` version. And of course Windows has it's own thing, so we override the event loop policy for Python >= 3.8. I'm just adding this note if we hit a similar issue in the conda builds again.

sk1p, 2024-02-27:
In #1603 this is now passing on Linux - Windows needs something extra...

matbryan52, 2024-02-27:
> In #1603 this is now passing on Linux - Windows needs something extra...

OK, I will see if I can reproduce on our server.

uellue, 2024-02-27:
Running the "not dist" tests without data in Win 11 with fresh Python 3.8 `pip install -e . -r test_requirements.txt` gave the following:

[notdist.freeze.txt](https://github.com/LiberTEM/LiberTEM/files/14423203/notdist.freeze.txt)
[notdist.log](https://github.com/LiberTEM/LiberTEM/files/14423204/notdist.log)

closed 2024-02-28 by sk1p