LiberTEM/LiberTEM #1577: Revisit event loop blocking functions

sk1p, 2024-01-24:
Remainders from #1572:

  - [ ] `unsnooze` -> `get_executor`: slow on some machines, so it might need to be `async` (or can it run in a thread?)
  - [ ] `shutdown`: closing the executor could take a while?
  - [ ] `make_executor`

Also, the cluster spin-up could be more lazy, like only wait for a fraction of workers to come up, instead of all of them, and already start servicing real user requests.

sk1p, 2024-02-26:
> * [ ]  `unsnooze` -> `get_executor`: slow on some machines, so it might need to be `async` (or can it run in a thread?)
> * [ ]  `shutdown`: closing the executor could take a while?
> * [ ]  `make_executor`
> 
> Regarding `unsnooze` and `make_executor` - both are basically blocked by this issue:
> https://github.com/LiberTEM/LiberTEM/blob/5d4d2de2256113b3760fabede1a316716fadaa6b/src/libertem/web/server.py#L244-L252

This is now taken care of as part of #1578 - `shutdown` probably can't be async, as that is called too late in the whole process. `shutdown` is generally a bit broken right now - if it's called from a different thread than where the loop is running, we can't reliably wait for the cancellation to actually go through (and if the event loop is dead, it's pointless anyways)

uellue, 2025-05-15:
Is this issue still relevant?