LiberTEM/LiberTEM #1617: Support for new EMPAD files

uellue, 2024-03-26:
The files in https://er-c-data-mgmt.fz-juelich.de/browse/?path=/adhoc/libertem/libertem-test-data/EMPAD/Magical_DataSet#browse don't work with with current EMPAD reader.

sk1p, 2024-04-02:
Loading via `ctx.load("empad", ...)` gives:

```
[...]
File ~/source/LiberTEM/src/libertem/io/dataset/base/file.py:166, in get_array_from_memview()
    162 assert len(mem) % itemsize == 0, \
    163     "len(mem) must fit the dtype"
    164 assert len(mem) // itemsize % self.num_frames == 0, \
    165     "len(mem) must fit the number of frames"
--> 166 assert len(mem) // itemsize // self.num_frames % (
    167     slicing.frame_size + (self._frame_header + self._frame_footer) // itemsize
    168 ) == 0, "len(mem) must fit the sig shape"
    169 arr_uncut = np.frombuffer(mem, dtype=self._native_dtype).reshape(
    170     (self.num_frames, -1)
    171 )
    172 arr = arr_uncut[:, slicing.frame_offset:slicing.frame_offset + slicing.frame_size]

AssertionError: len(mem) must fit the sig shape
```

Probable cause: the raw file has shape `(128, 16)`, and the XML file contains the following:

```
  <scan_parameters mode="search">
    <scan_resolution_x>16</scan_resolution_x>
    <scan_resolution_y>128</scan_resolution_y>
    <scan_center_x>0.5</scan_center_x>
    <scan_size>1.0</scan_size>
    <scan_center_y>0.5</scan_center_y>
  </scan_parameters>
  <scan_parameters mode="acquire">
    <scan_resolution_x>256</scan_resolution_x>
    <scan_resolution_y>256</scan_resolution_y>
    <scan_center_x>0.5</scan_center_x>
    <scan_size>1.0</scan_size>
    <scan_center_y>0.5</scan_center_y>
  </scan_parameters>
```

So the `mode="search"` parameters are actually the correct ones. I'll try to track down how this happens, and we should probably inform TFS about this - but now there are these kind of files around. As a work around for this, we could match the file size with the different scan parameters and select the one that matches.

uellue, 2024-04-02:
If we have an enumerable number of options we can perhaps check which ones match the actual raw data file size?

uellue, 2024-04-03:
> If we have an enumerable number of options we can perhaps check which ones match the actual raw data file size?

:-D I should have read your comment before more carefully: 

>  As a work around for this, we could match the file size with the different scan parameters and select the one that matches.

sk1p, 2024-04-08:
> we should probably inform TFS about this

Internally, we have opened a support case for this issue, as it is definitely happening at our Spectra microscope.

PR #1620 (2024-04-09): Fixup for malformed EMPAD data
For some EMPAD data sets, the raw file corresponds to the shape of the scan parameters with `mode="search"`, not `mode="acquire"`. That means we were expecting more data than actually available.

Note that this means we are more strict in what we accept, in particular this checks (if we are opening via the XML file) that the raw file is exactly as large as specified in the XML file (one of the two `scan_parameters`). Previously, it was probably possible to have a raw file that is larger than the specified scan in the XML file, and we would have probably accepted  [... cut]
closed 2024-04-09 by sk1p