LiberTEM/LiberTEM #1623: Release planning for 0.14

uellue, 2024-04-10:
We have quite a good set of changes in the current master, in particular fixing CoM descan error compensation for live processing, enabled by valid data masking, and supporting EMPAD files saved in search mode. For that reason we can think about releasing soon. This would also enable valid data masking and descan error compensation in iCoM, relevant for Dan.

Should we first release a new version of blobfinder to include #1600 in 0.14, or defer that?

If you agree, I can start reviewing the open issues and PRs, addressing or kicking them down the line as needed.

# Release checklist

Issues and pull requests to be considered for this release:

* ??? #1600
* ??? #1482 
* #1553

## Step 0

* [ ] Create a release candidate using `scripts/release`. See `scripts/release --help` for details.

## Before (using a release candidate package)

* [ ] Review open issues and pull requests
* [ ] License review: no import of GPL code from MIT code
      `pydeps --only "libertem" --show-deps --noshow src\libertem | python scripts\licensecheck.py`
* [ ] Run full CI pipeline, including slow tests, on [Azure DevOps](https://dev.azure.com/LiberTEM/LiberTEM/_build?definitionId=3)
* [ ] Run tests for related packages w/ new LiberTEM version
    * [ ] LiberTEM-live
    * [ ] LiberTEM-holo
    * [ ] LiberTEM-blobfinder
    * [ ] ptychography40
    * [ ] LiberTEM-iCoM
* [ ] Handle deprecation, search the code base for `DeprecationWarning`
      that are supposed to be removed in that release.
* [ ] GUI dependency update with `npm install`
* [ ] Review https://github.com/LiberTEM/LiberTEM/security/dependabot and update dependencies
* [ ] Full documentation review and update, including link check using
      ``sphinx-build -b linkcheck "docs/source" "docs/build/html"``
* [ ] Run complete test suite, including slow tests that are deactivated by default
      and tests that require sample files.
* [ ] Update the expected version in notes on changes, i.e. from `0.3.0.dev0`
      to `0.3.0` when releasing version 0.3.0.
* [ ] Update and review change log in `docs/source/changelog.rst`, merging
      snippets in `docs/source/changelog/*/` as appropriate.
* [ ] Update the JSON files in the ``packaging/`` folder with author and project information
* [ ] Edit `pytest.ini` to exclude flaky tests temporarily from release builds
* [ ] `Confirm that wheel, tar.gz, and AppImage are built for the release candidate on
      GitHub <https://github.com/LiberTEM/LiberTEM/releases>`_
* [ ] Confirm that a new version with the most recent release candidate is created in the
      `Zenodo.org sandbox <https://sandbox.zenodo.org/record/367108>`_ that is ready for submission.
* [ ] Install release candidate packages in a clean environment
      (for example:
      `python -m pip install -i https://test.pypi.org/simple/ --extra-index-url https://pypi.org/simple 'libertem==0.2.0rc11'`)
* [ ] Test the release candidate docker image
    * [ ] Confirm rc images and tags on https://ghcr.io/libertem/libertem
* [ ] For the GUI-related items, open in an incognito window to start from a clean slate
* [ ] Correct version info displayed in info dialogue?
* [ ] Link check in version info dialogue
* [ ] Test GUI without internet access
* [ ] Make sure you have test files of all supported types available
    * [ ] Include floats, ints, big endian, little endian, complex raw data
* [ ] Open each test file
    * [ ] Are parameters recognized correctly, as far as implemented?
    * [ ] Any bad default values?
    * [ ] Does the file open correctly?
    * [ ] Have a look at the dataset info dialogue. Reasonable values?
* [ ] Perform all analyses on each test file.
    * [ ] Does the result change when the input parameters are changed?
    * [ ] All display channels present and looking reasonable?
    * [ ] Reasonable performance?
    * [ ] Use pick mode.
* [ ] Re-open all the files
    * [ ] Are the files listed in "recent files"?
    * [ ] Are the parameters filled from the cache correctly?
* [ ] Try opening all file types with wrong parameters
    * [ ] Proper understandable error messages?
* [ ] Pick one file and confirm keyboard and mouse interaction for all analyses
    * [ ] Correct bounds check for keyboard and mouse?
* [ ] Check what happens when trying to open non-existent files or directories in the GUI.
    * [ ] Proper understandable error message?
    * [ ] Possible to continue working?
* [ ] Shut down libertem-server while analysis is running
    * [ ] Shut down within a few seconds?
    * [ ] All workers reaped?
* [ ] Check what happens when trying to open non-existent files by scripting.
    * [ ] Proper understandable error message? TODO automate?
* [ ] Check what happens when opening all file types with bad parameters by scripting
    * [ ] Proper understandable error message? TODO automate?
* [ ] Run libertem-server on Windows, connect to a remote dask cluster running on Linux,
  open all file types and perform an analysis for each file type.
* [ ] Use the GUI while a long-running analysis is running
    * [ ] Still usable, decent response times?
* [ ] Confirm that pull requests and issues are handled as intended, i.e. milestoned and merged
  in appropriate branch.
* [ ] Final version bump: `./scripts/release bump v0.3.0 --tag`, push to github
* [ ] After pipeline finishes, write minimal release notes for the [release](https://github.com/liberTEM/LiberTEM/releases) and publish the GitHub release

## After releasing on GitHub

* [ ] Confirm that all release packages are built and release notes are up-to-date
* [ ] Install release package
* [ ] Confirm correct version info
* [ ] Confirm package upload to PyPI
* [ ] Confirm images and tags on https://ghcr.io/libertem/libertem
* [ ] Publish new version on zenodo.org
* [ ] Update documentation with new links, if necessary
    * [ ] Add zenodo badge for the new release to Changelog page
* [ ] Conda packaging: review PRs on https://github.com/conda-forge/libertem-feedstock/pulls
* [ ] Send announcement message on mailing list
* [ ] Edit `pytest.ini` to include flaky tests again
* [ ] Bump version in master branch to next .dev0 (`./scripts/release bump v0.X.0.dev0 --commit`)
* [ ] Add to institutional publication databases
* [ ] Add the current LiberTEM version to [CVL](https://github.com/Chasdfracterisation-Virtual-Laboratory/CharacterisationVL-Software>) - add both the singularity and the .desktop file!

sk1p, 2024-04-11:
> We have quite a good set of changes in the current master, in particular fixing CoM descan error compensation for live processing, enabled by valid data masking, and supporting EMPAD files saved in search mode. For that reason we can think about releasing soon. This would also enable valid data masking and descan error compensation in iCoM, relevant for Dan.
> If you agree, I can start reviewing the open issues and PRs, addressing or kicking them down the line as needed.

Agreed!

IMHO:

> Should we first release a new version of blobfinder to include #1600 in 0.14, or defer that?
> * ??? [Deprecate gridmatching and fullmatch #1600](https://github.com/LiberTEM/LiberTEM/pull/1600)
> * [WIP: add test environment without any extras installed LiberTEM-blobfinder#87](https://github.com/LiberTEM/LiberTEM-blobfinder/pull/87)

Yes, include (blobfinder release is almost done)

> * ??? [GUI support for the features added in the CoM UDF #1482](https://github.com/LiberTEM/LiberTEM/issues/1482)

Postpone

> * [Fix license to GPL-3 or later in Zenodo template #1553](https://github.com/LiberTEM/LiberTEM/issues/1553) [Update license to GPL 3.0 or later #1624](https://github.com/LiberTEM/LiberTEM/pull/1624)

See review

sk1p, 2024-04-11:
Manual "thorough" run of GitHub Actions successful: https://github.com/LiberTEM/LiberTEM/actions/runs/8644927225

sk1p, 2024-04-11:
Data tests, including slow ones, ran successfully: https://dev.azure.com/LiberTEM/LiberTEM/_build/results?buildId=2740&view=results

sk1p, 2024-04-18:
Before cutting an RC, I think I'll adjust the release script a bit, to match what we have done for blobfinder (and maybe include some changes to also put the rc on the "real" pypi, as that removes some of the complexity)

sk1p, 2024-04-18:
For the UI testing, main focus should be on the snoozing feature, I don't think we had substantial changes other than that. I'll also re-check the `libertem-live-server` CoM + valid data masking stuff, to make sure we hit the goal there.

sk1p, 2024-04-25:
There is some instability with our tests, for example [this run here](https://dev.azure.com/LiberTEM/LiberTEM/_build/results?buildId=2755&view=logs&j=4d6f0f97-9d62-53bc-79d2-96c85fc226be&t=b5ec37f2-4755-55f4-9304-bcf3d588aca8) or [this one[(https://dev.azure.com/LiberTEM/LiberTEM/_build/results?buildId=2754&view=logs&j=87893246-87fc-5a29-d06a-4b8b42ba2dd7&t=4d42f0d3-3212-51bb-b595-3bee5051e738), seemingly hanging. Doesn't seem to be limited to notebook tests, but that's where I've seen it the first time. Might be something to look into before the release.

sk1p, 2024-04-25:
> There is some instability with our tests, for example [this run [...]

matbryan52, 2024-04-26:
> There is some instability with our tests, for example [this run [...]

Not much to go on in those logs, is it just the `/data` tests that you've noticed this for? Could it be network or load on the runner?

sk1p, 2024-04-29:
> Not much to go on in those logs, is it just the `/data` tests that you've noticed this for?

Yeah, no recent hangs on GitHub Actions.

> Could it be network or load on the runner?

Unclear, couldn't properly debug this last time, I just kicked the agent containers, which one has to do from time to time.

sk1p, 2024-04-30:
> I'll also re-check the `libertem-live-server` CoM + valid data masking stuff, to make sure we hit the goal there.

The valid data mask and related changes in `CoMUDF` look good. Once #1633 lands, we are ready to build a release candidate.

sk1p, 2024-05-16:
rc1 is baking: https://github.com/LiberTEM/LiberTEM/actions/runs/9108024491

sk1p, 2024-05-16:
Hmm, git revision does not display in the about dialog of the GUI (seen while testing the docker image) - I'll investigate.

sk1p, 2024-05-16:
> Hmm, git revision does not display in the about dialog of the GUI (seen while testing the docker image) - I'll investigate.

`_baked_revision.py` is present in the wheel, but not in the docker image. I think the proper fix is to use `hatchling` to build the package, like we are already doing for `LiberTEM-{live,blobfinder,...}`. I'll open an issue for this, to be solved for next release.

sk1p, 2024-05-16:
rc2 is building: https://github.com/LiberTEM/LiberTEM/actions/runs/9115164958

sk1p, 2024-05-16:
Zenodo is err'ing out with a 404 here, I'll try if a re-run as rc3 fixes things https://github.com/LiberTEM/LiberTEM/actions/runs/9115164958/job/25061353503#step:7:3405

sk1p, 2024-05-18:
Conda-forge PR should be ready to go: https://github.com/conda-forge/libertem-feedstock/pull/7