LiberTEM/LiberTEM #1648: Installation with `conda` on Python 3.10

uellue, 2024-05-21:
## How to reproduce

On Debian Bookworm:

```
conda create -n ltrelease310 python=3.10
conda activate ltrelease310
conda install -c conda-forge libertem[bqplot] cupy
libertem-server
```

## What happens

```
(ltrelease310) weber@iff007:~$ libertem-server                                                                                       
Traceback (most recent call last):                                                                                                   
  File "/home/weber/miniconda3/envs/ltrelease310/bin/libertem-server", line 11, in <module>                                          
    sys.exit(main())
  File "/home/weber/miniconda3/envs/ltrelease310/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/weber/miniconda3/envs/ltrelease310/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/weber/miniconda3/envs/ltrelease310/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/weber/miniconda3/envs/ltrelease310/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/weber/miniconda3/envs/ltrelease310/lib/python3.10/site-packages/libertem/web/cli.py", line 89, in main
    from .server import is_localhost
  File "/home/weber/miniconda3/envs/ltrelease310/lib/python3.10/site-packages/libertem/web/server.py", line 26, in <module>
    from .shutdown import ShutdownHandler
  File "/home/weber/miniconda3/envs/ltrelease310/lib/python3.10/site-packages/libertem/web/shutdown.py", line 4, in <module>
    from .state import SharedState
  File "/home/weber/miniconda3/envs/ltrelease310/lib/python3.10/site-packages/libertem/web/state.py", line 21, in <module>
    from libertem.io.writers.results import formats  # NOQA
  File "/home/weber/miniconda3/envs/ltrelease310/lib/python3.10/site-packages/libertem/io/writers/results/formats.py", line 1, in <module>
    import h5py
  File "/home/weber/miniconda3/envs/ltrelease310/lib/python3.10/site-packages/h5py/__init__.py", line 33, in <module>
    from . import version
  File "/home/weber/miniconda3/envs/ltrelease310/lib/python3.10/site-packages/h5py/version.py", line 15, in <module>
    from . import h5 as _h5
  File "h5py/h5.pyx", line 1, in init h5py.h5
ImportError: /home/weber/miniconda3/envs/ltrelease310/lib/python3.10/site-packages/h5py/defs.cpython-310-x86_64-linux-gnu.so: undefined symbol: H5Pget_fapl_direct
```

## Differential diagnosis

* `conda install -c conda-forge h5py` in a clean 3.10 environment allows importing h5py.
* `conda install -c conda-forge libertem` and `conda install -c conda-forge libertem[bqplot]` allows running libertem-server.
* `conda install -c conda-forge cupy h5py` allows importing h5py.
   * Subsequently adding libertem with `conda install -c conda-forge cupy h5py libertem[bqplot]` allows running libertem-server

matbryan52, 2024-05-22:
I can reproduce this on an Ubuntu 20 server.

Installing `libertem cupy` pulls in:

```
  h5py               conda-forge/linux-64::h5py-3.7.0-nompi_py310h06dffec_100
  hdf5               pkgs/main/linux-64::hdf5-1.12.1-h2b7332f_3
```

While `cupy h5py` pulls in:

```
  h5py               conda-forge/linux-64::h5py-3.11.0-nompi_py310hf054cd7_101
  hdf5               conda-forge/linux-64::hdf5-1.14.3-nompi_h4f84152_101
```

Both pull in:

```
  cudatoolkit        conda-forge/linux-64::cudatoolkit-11.8.0-h4ba93d1_13
  cupy               conda-forge/linux-64::cupy-13.1.0-py310h189a05f_4
```

This is likely related : https://github.com/conda-forge/h5py-feedstock/issues/103

Something we depend on is causing `h5py` or `hdf5` to be pinned, and then `conda` is installing an incompatible other half. I tried tracking down what is doing this but the solver logs are hard to follow.

By the way, the optional dependency syntax `libertem[bqplot]` is unsupported by `conda` and is just silently ignored in the install process, see https://github.com/conda/conda/issues/7502

uellue, 2024-05-22:
Thank you for replicating and tracking this down further!

I guess this one is a bit out of our hands since we only have a few restrictions in our dependencies where I can't imagine how they would cause this. Probably leave this Issue here for re-checking later and for documentation purposes?

Good to know about the dependency syntax, funny that it doesn't throw an error...

uellue, 2024-05-22:
Running `conda update -c conda-forge hdf5 h5py` after `conda install -c conda-forge libertem cupy` allows starting `libertem-server` and at least basic usage of it.

matbryan52, 2024-05-22:
Re-trying later in the day installed a working combination of `h5py`, `hdf5` this time. See the installed packages diff here : https://www.diffchecker.com/cKXscjJ6/

I would guess it might be a release or a CDN glitch ?

matbryan52, 2024-06-27:
I think this can be closed, install on 3.10 seems OK.

closed 2024-06-27 by matbryan52