LiberTEM/LiberTEM #1649: Permissive licensing?

uellue, 2024-06-07:
Discussion with JK Weiss, TESCAN:

If LiberTEM was under a permissive license like MIT, it would allow integration in TESCAN's software. That would be of concrete interest for TESCAN.

Since we have a lot of code contributed by a small number of contributors, and we can probably contact a large percentage of our contributors, re-licensing would be a realistic option.

uellue, 2024-06-07:
Assigning to the next milestone to make sure we follow up in a timely manner.

uellue, 2024-06-07:
```
    install_requires=[
        "numpy", # 3-Clause BSD
        "scipy>=1.4.1", # 3-Clause BSD
        "sparse", # 3-Clause BSD License
        "distributed>=2.19.0", # 3-Clause BSD License
        "dask!=2023.6.1",  # 3-Clause BSD License
        "click", # 3-Clause BSD License
        "tornado>=5", # Apache 2.0
        "matplotlib", # Matplotlib only uses BSD compatible code, and its license is based on the [PSF](https://docs.python.org/3/license.html) license
        "pillow", # Like PIL, Pillow is licensed under the open source HPND License (permissive)
        "h5py", # 3-Clause BSD License
        "psutil", # 3-Clause BSD License
        # https://github.com/LiberTEM/LiberTEM/issues/763
        # https://github.com/LiberTEM/LiberTEM/issues/783
        "numba>=0.51", # BSD 2-Clause "Simplified" License
        "ncempy>=1.10", # !!! GPL-3
        'pywin32!=226;platform_system=="Windows"', # 3-Clause BSD License
        # FIXME pull request #259
        # https://github.com/LiberTEM/LiberTEM/pull/259#discussion_r251877431
        'scikit-image', # 3-Clause BSD License
        'cloudpickle', # 3-Clause BSD License
        'jsonschema', # MIT License
        'scikit-learn', # 3-Clause BSD License
        'tqdm', # MIT "it's complicated" https://github.com/tqdm/tqdm/blob/master/LICENCE
        'threadpoolctl>=3.0', # 3-Clause BSD License
        'nbformat', # 3-Clause BSD License
        'nbconvert', # 3-Clause BSD License
        'autopep8', # MIT License
        'empyre>=0.3.0', # !!! EMPyRe is licensed under [GPLv3](https://iffgit.fz-juelich.de/empyre/empyre/-/blob/master/LICENSE).
        'defusedxml', # Python Software Foundation License (PSFL)
        'typing-extensions',  # Python Software Foundation License
        'opentelemetry-api', # Apache 2.0
        'tblib', # BSD 2-Clause License
        'tomli', # MIT License
        'sparseconverter>=0.3.3', # MIT License
        'numexpr!=2.8.6',  # MIT License
        'jupyter_ui_poll',  # MIT License
        'ipympl',  # 3-Clause BSD License
    ],
    extras_require={
        # NumPy interfacing issue on Win 11, Python 3.10
        # Likely caused by other packages constraining NumPy version to <1.23
        # FIXME recheck later again
        'torch': 'torch<1.12', # 3-Clause BSD License
        # Typing issue in hdbscan, maybe
        # https://github.com/scikit-learn-contrib/hdbscan/issues/600
        # FIXME check if error in tests\test_fullmatch.py persists in a later
        # hdbscan release
        'hdbscan': [ 
            "hdbscan<=0.8.30;( python_version=='3.11' and platform_system=='Windows')", # 3-Clause BSD License
            "hdbscan;( python_version!='3.11' or platform_system!='Windows')",
        ],
        'cupy': 'cupy', # MIT License
        'bqplot': [
            'bqplot', # MIT License
            'bqplot-image-gl', # MIT License
            'ipython'], # 3-Clause BSD License
        'hdf5plugin': 'hdf5plugin', # MIT License It's complicated https://github.com/silx-kit/hdf5plugin/blob/main/LICENSE
        'tracing': [
            'opentelemetry-distro', # Apache 2.0
            'opentelemetry-exporter-otlp', # Apache 2.0
        ],
    },
```

`empyre` and `ncempy` are GPL, the rest is permissive. Looks feasible!

uellue, 2024-06-07:
Discussion with @sk1p: We could use this opportunity to spin out the web GUI and perhaps other parts in separate modules.

sk1p, 2024-07-31:
Btw. this also concerns the LiberTEM-live repo, but there it is much easier because of the lower number of contributors.

@matbryan52 what do you think about this? Your opinion is very important for deciding if we want to go forward with re-licensing or not, before contacting other individual contributors. Both Dieter and me would vote to go forward with the re-licensing, if there aren't any show stoppers.

> `ncempy` [is] GPL

The `io` part of `ncempy` is MIT licensed, by the way!

matbryan52, 2024-08-02:
> @matbryan52 what do you think about this? Your opinion is very important for deciding if we want to go forward with re-licensing or not, before contacting other individual contributors. Both Dieter and me would vote to go forward with the re-licensing, if there aren't any show stoppers.

Absolutely, I've always assumed this would happen one day with the push to put things into `libertem.common` where possible.

Sorry for the slowness replying!

sk1p, 2024-08-05:
> Absolutely, I've always assumed this would happen one day with the push to put things into `libertem.common` where possible.

Thanks for the reply! @uellue I'll assign the issue to you if you don't mind - of course I'll help wherever I can.

> Sorry for the slowness replying!

No worries!

uellue, 2024-09-12:
List of contributors, tracking agreement with MIT relicensing:

- [x] Alexander Clausen
- [x] Dieter Weber
- [x] Matthew Bryan
- [ ] Karina Ruzaeva
- [ ] Vadim Migunov
- [ ] Sivert J.V. Dagenborg
- [ ] Anand Baburajan
- [ ] Abijith Bahuleyan
- [ ] Jan Caron
- [ ] Rahul Chandra
- [ ] Shankhadeep Dey
- [ ] Sayandip Halder
- [ ] Daniel S. Katz
- [ ] Barnaby D.A. Levin
- [ ] Magnus Nord
- [ ] Colin Ophus
- [ ] Simon Peter
- [ ] Levente Puskás
- [ ] Jay van Schyndel
- [ ] Jaeweon Shin
- [ ] Sai Sunku
- [ ] Håkon Wiik Ånes
- [ ] Knut Müller-Caspary
- [ ] Rafal E. Dunin-Borkowski

uellue, 2024-09-12:
Hi @kruzaeva @woozey @sivborg @anandbaburajan @twentyse7en @jan-car @irahulcse @iamshankhadeep @sayandip18 @danielskatz @bdalevin @magnunor @cophus @probonopd @DYVZ8X @ozej8y @jaewshin @saisunku @hakonanes @rafaldb 

I'm contacting you since you contributed to the LiberTEM project. Many thanks for being a part of this project! If you are wondering what your contribution was you can have a look at our [acknowledgments](https://libertem.github.io/acknowledgements.html#creators) or ask us.

From several companies, including Direct Electron, Nanomegas and TESCAN, we received feedback that they would be very interested to use LiberTEM if it was MIT licensed. That would be very much in line with our goal to make electron microscopy software and hardware more open and interoperable, and to provide and maintain a shared infrastructure to make life easier for everybody.

Since most of LiberTEM, including your contributions, were previously licensed under the GPL-3, I'd like to ask for your consent to re-license your contributions to any of the repositories in the [LiberTEM organization](https://github.com/LiberTEM/) under the MIT license. A very simple "yes" comment or e-mail would be sufficient if you agree.

If you don't agree or if you have further questions, you are very welcome to comment here or contact me at [d.weber@fz-juelich.de](mailto:d.weber@fz-juelich.de).

Again, many thanks for your contribution, and I hope to hear back from you!

Dieter

Iamshankhadeep, 2024-09-12:
Yea

irahulcse, 2024-09-12:
Yes

*Regards,*

*Rahul Chandra*

*Email: ***@***.*** ***@***.***>*

ozej8y, 2024-09-12:
YesSent from my jPhone

twentyse7en, 2024-09-12:
Yes

uellue, 2024-09-12:
Knut agreed by personal e-mail, thx! :-)

saisunku, 2024-09-12:
Yes

magnunor, 2024-09-12:
Yes

sivborg, 2024-09-12:
Yes

kruzaeva, 2024-09-12:
yes

hakonanes, 2024-09-12:
Yes yes

jaewshin, 2024-09-12:
Yes

woozey, 2024-09-12:
Yes

Great job pushing it forward 👍🏻👍🏻

uellue, 2024-09-13:
Rafal agreed in personal communication. :-)

uellue, 2024-09-25:
Sayandip agreed in personal communication

uellue, 2024-09-30:
Colin agreed in personal communication

uellue, 2024-10-23:
Barnaby agreed in personal communication. That means we have consent from all creators. I'm currently working on the `empyre` dependency.

PR #1689 (2024-11-20): Relicense to MIT
This is built on top of #1687, on which it depends.

* Remove Holo code that was copied from HyperSpy under the GPL
* Relicense LiberTEM under MIT license

We will need to release LiberTEM-holo and handle the migration and deprecation before merging this. Marked as draft for that reason.

It also may require a rebuilt client since `client/src/About.tsx` was changed.
    
Closes #1649

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a chang [... cut]
commit 56d3b61f (2024-11-20): Relicense LiberTEM under MIT license

Closes #1649
closed 2024-11-20 by sk1p