LiberTEM/LiberTEM #1657: v0.14.1 point release

sk1p, 2024-06-18:
Issues/PRs that should be included:

-  #1656

@uellue @matbryan52 would any of you like to take charge of this release?

uellue, 2024-06-18:
@sk1p sorry to bother, but what were the steps for a point release again? Sth about setting the version by hand?

We didn't merge anything of importance since the last release, so no need for a stable branch.

sk1p, 2024-06-18:
> @sk1p sorry to bother, but what were the steps for a point release again? Sth about setting the version by hand?

Yup, either setting the version to 0.14.1.dev0 by hand, or using the release script to "bump backwards" with `--force` (IIRC)

> We didn't merge anything of importance since the last release, so no need for a stable branch.

👍

matbryan52, 2024-06-18:
> @uellue @matbryan52 would any of you like to take charge of this release?

I am happy to test and help but I haven't ever been close to the active part of release cutting before, would be coming from a ways behind. If @uellue you want to look at it together I am there.

uellue, 2024-06-18:
Yes, `--force` made it downgrade. Thx! Now waiting for the thorough tests.

uellue, 2024-06-18:
Cluster analysis doesn't like NumPy 2 on 3.9: https://github.com/LiberTEM/LiberTEM/actions/runs/9567900368/job/26376754929

--> #1658

matbryan52, 2024-06-19:
> Cluster analysis doesn't like NumPy 2 on 3.9: https://github.com/LiberTEM/LiberTEM/actions/runs/9567900368/job/26376754929
> 
> --> #1658

Looks like the scikit-image release fixed the cluster analysis problem on py39 : https://github.com/LiberTEM/LiberTEM/actions/runs/9567900368/job/26404990589

uellue, 2024-06-19:
I also confirmed that LiberTEM still works with NumPy < 2, since we recommend it. In a call until 11:00, then I will do short docs update.

matbryan52, 2024-06-19:
Just a warning for now, I imagine this predates numpy2.

```
src\libertem\io\dataset\npy.py:11
  ~\Workspace\libertem_fork\src\libertem\io\dataset\npy.py:11: DeprecationWarning: `np.compat`, which was used during the Python 2 to 3 transition, is deprecated since 1.26.0, and will be removed
    from numpy.compat import long, asstr
 ```

uellue, 2024-06-19:
> Just a warning for now, I imagine this predates numpy2.
>
> ```
> src\libertem\io\dataset\npy.py:11
>   ~\Workspace\libertem_fork\src\libertem\io\dataset\npy.py:11: DeprecationWarning: `np.compat`, which [...]

I'm preparing a PR to integrate the upstream changes which fix this. :-)

matbryan52, 2024-08-02:
Think we're done with the point release ? Conda release is good, and we haven't seen any fires that I'm aware of

closed 2024-08-02 by matbryan52