LiberTEM/LiberTEM #1672: Update and maintain lower version bound for `numba`

sk1p, 2024-08-19:
Even if not strictly needed at all times, we should keep the low bound of `numba` (maybe other packages?) to at least a version compatible with the lowest supported Python version.

Without this change, `pip` ([or `uv pip`](https://github.com/astral-sh/uv/issues/6209)) may decide to select an ancient version which fits, but may no longer install/run/build/... on current Python (because `numba` doesn't have an upper bound in `python_requires`, for good reason - it might be compatible!).

In [the case encountered](https://dev.azure.com/LiberTEM/LiberTEM/_build/results?buildId=2841&view=logs&j=4ddc71c5-afbc-5035-84c0-b16e02ac0e54&t=02668421-4727-569b-1360-de4b4bf56e3a), this is because `numpy` released `2.1`, which `numba==0.60.0` is explicitly incompatible with - so if the version resolution tries `numpy==2.1.0` first (`uv` behavior), it will select an old `numba`, whereas plain old `pip` will select `numpy==2.0.1` and current `numba`. 

I wonder if we can add some resilience to this - maybe try `uv pip install --resolution=lowest` in CI?

sk1p, 2024-08-19:
> I wonder if we can add some resilience to this - maybe try `uv pip install --resolution=lowest` in CI?

Doesn't seem to be feasible yet - this requires our transitive dependencies to also specify proper lower constraints. Currently, generating a lock file and cleanly installing on Python 3.11 for example requires the following extras:

```
bleach>6
ipython>7
decorator>4
pyzmq>24
kiwisolver>=1.4.5
```

Let's assume we put that into `r.in` in the LiberTEM source tree (and add `-e .` for flavor), then we can do `uv pip compile r.in --resolution=lowest` and `uv pip install` the result. I would prefer, if we do this, to not need any additional constraints, as they could hide real problems.

This is an approximate patch needed to constrain our direct dependencies:

```diff
diff --git a/pyproject.toml b/pyproject.toml
index 260bae39..a1c2e520 100644
--- a/pyproject.toml
+++ b/pyproject.toml
@@ -41,31 +41,31 @@ classifiers = [
     "Topic :: Scientific/Engineering :: Visualization",
 ]
 dependencies = [
-    "autopep8",
+    "autopep8>=2",
     "click",
     "cloudpickle",
-    "dask!=2023.6.1",
+    "dask!=2023.6.1,>=2023.5.0",
     "defusedxml",
-    "distributed>=2.19.0",
+    "distributed>=2023.5.0",
     "empyre>=0.3.0",
-    "h5py",
-    "ipympl",
-    "jsonschema",
-    "jupyter_ui_poll",
-    "matplotlib",
-    "nbconvert",
+    "h5py>=3.8",
+    "ipympl>0.9",
+    "jsonschema>=4",
+    "jupyter_ui_poll>=1",
+    "matplotlib>=3.9",
+    "nbconvert>=7.2",
     "nbformat",
     "ncempy>=1.10",
-    "numba>=0.51",
-    "numexpr!=2.8.6",
-    "numpy",
+    "numba>=0.60",
+    "numexpr!=2.8.6,>2.9",
+    "numpy>=1.23.5",
     "opentelemetry-api",
-    "pillow",
+    "pillow>10",
     "psutil",
     "pywin32!=226;platform_system==\"Windows\"",
-    "scikit-image",
-    "scikit-learn",
-    "scipy>=1.4.1",
+    "scikit-image>=0.22",
+    "scikit-learn>=1.3",
+    "scipy>=1.10.0",
     "sparse",
     "sparseconverter>=0.3.3",
     "tblib",
```

matbryan52, 2024-08-21:
> Even if not strictly needed at all times, we should keep the low bound of `numba` (maybe other packages?) to at least a version compatible with the lowest supported Python version.

I would agree, and since `numba` is both a fairly cornerstone package for us and tends to constrain the supported Python versions it would be a good thing to actively control in our dependencies. At a minimum to avoid the `uv` issue you describe.

> This is an approximate patch needed to (minimally) constrain our direct dependencies:

There are some very high pins in there ! Notably `matplotlib >= 3.9` (released in May 2024). I don't quite understand how this list was generated ?

sk1p, 2024-08-21:
> There are some very high pins in there ! Notably `matplotlib >= 3.9` (released in May 2024). I don't quite understand how this list was generated ?

Basically by adding/updating constraints until `uv pip compile --resolution=lowest` and installing those pinned versions succeeds. Indeed `matplotlib` looks over-constrained, 3.4 should be sufficient (might need to run the tests with these versions, though!).

PR #1677 (2024-09-21): Pin minimum numba versions for each Python version
This should fix the `uv` dependency resolution issue, where `uv` picks `numpy==2.1.x` and a very old, incompatible `numba` version that wasn't yet constrained to `numpy` versions.

Also add an unconditional `numba` pin so installation on newer Python in the future won't try to pick an ancient `numba` version.

Closes #1672 

## Contributor Checklist:

* [ ] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [ ] I have added [a changelog entry](https://github. [... cut]
closed 2024-09-21 by sk1p