LiberTEM/LiberTEM #1675: Loading .mrc files from Direct Electron Broken

CSSFrancis, 2024-09-19:
I was looking for how the metadata file is loaded for loading .mrc file from Direct Electron and I'm a bit confused by how this currently works. 

It seems like the navigation (real space) dimensions are just doubling the length in the header of the .mrc file rather than getting these values from the "_info.txt" file.

A maybe not so minimal example but I was writing a test which just takes a small dataset and then tries to load it into libertem. 

```python 

import libertem.api as lt
import pytest
import os
import glob
import time

class TestLoadingLiberTEM:
    @pytest.mark.parametrize("file_format", ["MRC", "DE5", "HSPY"])
    @pytest.mark.server
    def test_save_4DSTEM(self, client, file_format):
        if not os.path.exists("D:\Temp"):
            os.mkdir("D:\Temp")
        temp_dir = "D:\Temp"
        client["Frames Per Second"] = 100
        client["Scan - Enable"] = "On"
        client.scan["Size X"] = 16
        client.scan["Size Y"] = 8
        client["Autosave Movie"] = "On"
        client["Autosave Movie File Format"] = file_format
        client["Autosave Directory"] = temp_dir
        client.start_acquisition(1)
        while client.acquiring:
            time.sleep(.1)
        movie = glob.glob(temp_dir + "/*movie."+file_format.lower())[0]
        dataset = lt.Context().load("auto", path = movie)
        assert tuple(dataset.shape) == (8,16, 1024, 1024) # works for .hspy and .de5 but not for .mrc

```

CSSFrancis, 2024-09-19:
If you point me to where the navigation shape is determined, I can test/ submit PR.

sk1p, 2024-09-20:
Thanks for the report! We didn't yet implement loading the `_info.txt` metadata. Right now, we are ["guessing"](https://github.com/LiberTEM/LiberTEM/blob/b83c8310e78fbf8e638dd4913d2dd2d91d038fa2/src/libertem/io/dataset/mrc.py#L181-L204) square shapes from the number of images, see also `make_2D_square` here: https://github.com/LiberTEM/LiberTEM/blob/b83c8310e78fbf8e638dd4913d2dd2d91d038fa2/src/libertem/common/math.py#L52-L79

I think that logic just might be broken in this case, but anyways, it would result in an unhelpful `(128, 1024, 1024)` in this case if it were working as intended.

See also: https://github.com/LiberTEM/LiberTEM/pull/1338

So, two things to fix: 

1) Read the `_info.txt` sidecar if it exists (for DE-written data sets)
2) Fix `make_2D_square` not to invent an invalid shape

cc @matbryan52

sk1p, 2024-09-23:
>     2. Fix `make_2D_square` not to invent an invalid shape

I checked, and it doesn't seem to be the fault of `make_2D_square` - looks like the issue is further up the stack:

```
In [3]: make_2D_square((8*16,))
Out[3]: (128,)
```

... and `detect_params` does only feed `shape[0]` into `make_2D_square`.

CSSFrancis, 2024-09-23:
@sk1p What does the detect_params function do? Do you read the file shape from the header or something else? 

For reference this is the header format used in rosettasciio 

https://github.com/hyperspy/rosettasciio/blob/9626ee9614ee31c74302aca19d6033beba3054b9/rsciio/mrc/_api.py#L45

CSSFrancis, 2024-09-23:
@sk1p I guess this is using the ncempy package.  Is there a reason (besides probably the liscense 😅) for not using the .mrc file reader in rosettasciio.  

That file reader should read distributed and already reads the .metadata file. I also spent quite a bit of time making sure that the memmap function works well there.  We can think about reliscensing there or I can write a similar file reader for libertem.

sk1p, 2024-09-23:
> @sk1p I guess this is using the ncempy package.

Yup

> Is there a reason (besides probably the liscense 😅) for not using the .mrc file reader in rosettasciio.
> That file reader should read distributed and already reads the `_info.txt` file.

Other than the license (our I/O code has ~always been both GPL and MIT, so we could never depend on GPL code there), two things:

1) `rosettasciio` wasn't a thing IIRC when we made our reader, and we already depended on `ncempy`
2) We don't have a hard dependency on `distributed` for our I/O code - I previously thought that all lazy loading w/ `rosettasciio` goes through `dask` or `distributed`, but that may be wrong/outdated?

>  I also spent quite a bit of time making sure that the memmap function works well there.

Nice! Is this independent of the `distributed` code? Is this a format-specific API, or is there a generic "give me an mmap" function? If this was a generic API, that would basically allow us to support all formats in `rosettasciio` that support memory mapping with little effort.

For other formats (probably any that need some kind of decoding step), we would need a `read_slice(...)` kind of function, so we can read the data in very fine-grained chunks (what we call tiles in LiberTEM).

Also, until memory mapping is part of [the documented API](https://hyperspy.org/rosettasciio/api/index.html#interfacing-the-rosettasciio-plugins) I'm a bit hesitant to start depending on it. What are the stability guarantees of such undocumented funtions?

> We can think about reliscensing there or I can write a similar file reader for libertem.

Both would be fine for me, although duplicating it in both is a bit sad in the long term. Re-licensing `rosettasciio` to MIT or BSD would definitely motivate me to contribute, and work towards ditching some of our custom I/O code.

CSSFrancis, 2024-09-23:
> Other than the license (our I/O code has ~always been both GPL and MIT, so we could never depend on GPL code there), two things:
> 
> 1. `rosettasciio` wasn't a thing IIRC when we made our reader, and we already depended on `ncempy`

Yea I assumed this was the case. 

> 2. We don't have a hard dependency on `distributed` for our I/O code - I previously thought that all lazy loading w/ `rosettasciio` goes through `dask` or `distributed`, but that may be wrong/outdated?

There is a hard dependency on `dask` which might be a bit hard to get away from.  Although I'd be curious to hear your thoughts against it :).  In the .mrc file loader we could just return the `memmap` array for the basic operation.  For things like the .mib reader, operating on the memory mapped file has singificant performance [implications](https://github.com/hyperspy/rosettasciio/pull/174#pullrequestreview-1686535220) at times. 

Then for loading binary data such that it is compatible with [distributed computing](https://github.com/hyperspy/rosettasciio/blob/9626ee9614ee31c74302aca19d6033beba3054b9/rsciio/utils/distributed.py#L115) it's just so useful.  This is also somewhat based on some code that @uellue suggested a bit ago

> Nice! Is this independent of the `distributed` code? Is this a format-specific API, or is there a generic "give me an mmap" function? If this was a generic API, that would basically allow us to [...]

See the link above :). I've been trying to aim for two implementations of each code. One which just uses `mmap` and one that supports distributed computing using `dask`. We could just have the first instance return a memmap object in most cases although for some (like the .mib) files it's better to use `dask`

> For other formats (probably any that need some kind of decoding step), we would need a `read_slice(...)` kind of function, so we can read the data in very fine-grained chunks (what we call tiles in [...]

In that case it's probably just a matter of you saying what parts should be documented/ shouldn't be changed.  On top of that we get need integration tests which aren't that hard.  

> Both would be fine for me, although duplicating it in both is a bit sad in the long term. Re-licensing `rosettasciio` to MIT or BSD would definitely motivate me to contribute, and work towards [...]

Hmm let me try to restart that discussion, I think if we have a motivation that could help things move along much quicker than previously.

sk1p, 2024-09-23:
> There is a hard dependency on `dask` which might be a bit hard to get away from. Although I'd be curious to hear your thoughts against it :).

It's not so much "against it", but I'm for exposing a more generally usable interface, if possible. The fact is that it is very easy to put `dask` on top of memory maps (or whatever I/O functionality you have). We have some functionality to build dask arrays out of LiberTEM `DataSet`s, for example. But once you have a `dask` array, you are bound to it - on top of that, you can only build with components that are explicitly or implicitly compatible with `dask`. Memory maps aren't the be-all, end-all, either, but that's another discussion I think.

In LiberTEM, we always had a separation between the concern "execute some code somewhere" (what we call the executor) and I/O (`DataSet` and its related classes). This is needed for live processing, where we are using a specialized executor, but is also good for experimentation and flexibility - for example, it would not be hard to write an executor based on `joblib`, and I think @matbryan52 once wrote one based on Ray. Generally, we are not "bound to the whims" of a single project, except maybe for the very core pydata libraries like `numpy`, `cupy`, etc...

> In the .mrc file loader we could just return the `memmap` array for the basic operation. For things like the .mib reader, operating on the memory mapped file has singificant performance [implications](https://github.com/hyperspy/rosettasciio/pull/174#pullrequestreview-1686535220) at times.

I could roughly reproduce the numbers from that comment - some comments on this:

1) As the operation is repeated, it's serviced from the file system cache - after the first round, it's literally memory - if you replace your `test` memory map with `np.zeros_like(test)`, you get more or less exactly the same result. This has more to do with how `np.flip`, caches and memory work, rather than how I/O or harddrives work
2) It would be a good idea to choose a realistic size (i.e. 256x256 or 512x512 frames), as the performance very much depends on that (you get longer runs of sequential reads for larger frames, and the perf difference gets smaller)
3) The comparison is a bit unfair, as you are not actually _using_ the result of `np.flip`. All `np.flip` does is an update of the `strides` of the array, so an operation that will depend mostly on other properties of the array, like the number of dimensions, not  the data in the array (this takes like a microsecond or two on my system). When actually doing an operation on the result, things look differently:

[image]

Oops, suddenly the performance numbers are inverted! I think the TL;DR is that performance is complicated, and micro benchmarks are very hard to get right.

> Then for loading binary data such that it is compatible with [distributed [...]

:+1: yeah, going from memory map to `dask` array is easy.

> See the link above :). I've been trying to aim for two implementations of each code. One which just uses `mmap` and one that supports distributed computing using `dask`. We could just have the first instance return a memmap object in most cases 

Great! Would be useful as a general API indeed.

> although for some (like the .mib) files it's better to use `dask`

Disagree, see above.

> In that case it's probably just a matter of you saying what parts should be documented/ shouldn't be changed.

Okay. I think in general, the docs could use some love, to have them represent reality a bit more closely. For example, `digitalmicrograph.file_reader` does not return a `dict` at all, but a list of dicts. I understand why, but it's either not implementing the interface correctly, or the interface documentation is wrong (I haven't looked at other readers in detail).

This is probably something we should tackle once the licensing stuff is out of the way?

> Hmm let me try to restart that discussion, I think if we have a motivation that could help things move along much quicker than previously.

:+1:

CSSFrancis, 2024-09-23:
@sk1p I'm happy to be proven wrong, although I think there is a bit of a misunderstanding.  You're not testing the same thing above.   When you preform the `flip` function on a memory mapped array in numpy it returns a memory mapped array. 

This isn't very performant at all because now you have to "jump" around on disk in order to read the data.  It's the difference of reading into memory and then flipping vs changing the read pattern from the disk.

sk1p, 2024-09-23:
> You're not testing the same thing above. When you preform the `flip` function on a memory mapped array in numpy it returns a memory mapped array.

`test` in the screenshot above _is_ a memory mapped array, so I'm not sure why I'm not testing the same thing. I'm just chaining on a `sum` because `flip` is, essentially, a lazy function. Using a `.copy()` would probably have a similar effect (can't test right now).

> This isn't very performant at all because now you have to "jump" around on disk in order to read the data.
> It's the difference of reading into memory and then flipping vs changing the read pattern from the disk.

It's true that reading in a non-sequential way from a memory map _might_ not perform well if the file is not cached yet, but that's not what that benchmark measures. If the file fits into main memory (which it does with the 40^4 f32 example), you get at most one slow run from disk, which will average out under `%%timeit`. All the following runs will read from file system cache, and will be indistinguishable from operating on a "normal" array. You'd have to flush the file system cache as part of the benchmark as well to cause I/O and properly show the effect of the read order in the repeated runs.

When I'm back at my work computer tomorrow I'll cook up an example to show what I mean.

> The results are highly dependent on the type of disk/ set up you have but worth consideration.

... yup, including also operating system - Linux properly performs read-ahead for many access patterns, where Windows performance often tanks completely (of course also depending on SSD vs. HDD etc.). Which is one reason we have different I/O backends for many file formats we support, and used buffered I/O on Windows by default...

CSSFrancis, 2024-09-24:
> `test` in the screenshot above _is_ a memory mapped array, so I'm not sure why I'm not testing the same thing. I'm just chaining on a `sum` because `flip` is, essentially, a lazy function. Using a [...]

Oh that is my bad I didn't realize you were testing a non memmapped file...  :) I'm very curious to see what the actual result should be.

> It's true that reading in a non-sequential way from a memory map _might_ not perform well if the file is not cached yet, but that's not what that benchmark measures. If the file fits into main [...]

I guess I'll have to wait till tomorrow. Although we usually aren't loading the same data from disk over and over again when ooperating on 4D STEM data (or if we are it's not small enough to stay in the file system chache). I guess the place where I found this to be a rather large bottle neck was trying to load large (2-3 TB) sized binary files.  In that case slicing (tiling?) the data along both the x,y, time dimensions vs only along the x dimensions resulted in a huge change in read speed because of how much jumping around on disk was occuring.  

> ... yup, including also operating system - Linux properly performs read-ahead for many access patterns, where Windows performance often tanks completely (of course also depending on SSD vs. HDD [...]

Yea that's something that is pretty severely lacking from rosettasciio...

CSSFrancis, 2024-09-24:
> `test` in the screenshot above _is_ a memory mapped array, so I'm not sure why I'm not testing the same thing. I'm just chaining on a `sum` because `flip` is, essentially, a lazy function. Using a `.copy()` would probably have a similar effect (can't test right now).

Also I believe the `np.array()` function is essentially a copy in that it casts it to an in memory array. (correct?)

sk1p, 2024-09-24:
> Also I believe the `np.array()` function is essentially a copy in that it casts it to an in memory array. (correct?)

True, if you keep the default parameters.

> It's true that reading in a non-sequential way from a memory map might not perform well if the file is not cached yet, but that's not what that benchmark measures. If the file fits into main memory [...]
When I'm back at my work computer tomorrow I'll cook up an example to show what I mean.

With the given array/file size, I sadly can't make Linux behave properly, and don't have time to continue today. Basically, the idea is to call `test._mmap.madvise(mmap.MADV_DONTNEED)` as part of the benchmark where `test` is your numpy memmap object. Then you can observe system behavior (I/O load - I'm using `iostat` for that) to see if it has an effect:

```python
%%timeit
test._mmap.madvise(mmap.MADV_DONTNEED)
np.sum(np.flip(np.array(test[1,:,:,:]),axis=-2), axis=(-2, -1))
```

For me, this doesn't work properly, which could be because we are accessing the same memory again directly afterwards - the documentation is a bit unclear if this is an asynchronous operation or not.

So I guess it would be better to run a test that is a bit more realistic (make a larger file, drop the cache, simulate a longer running realistic access pattern over a larger region of the file).

> Although we usually aren't loading the same data from disk over and over again when ooperating on 4D STEM data (or if we are it's not small enough to stay in the file system chache).

I think it depends a bit on what you are doing - if it's a batch reconstruction, I would agree, but for interactive exploration, going through the whole data set multiple time is definitely a common access pattern, and we do have some servers running here that have enough RAM for the smaller data sets (<=100GB). For larger-than-RAM data sets, we also have a direct I/O method, that bypasses the cache.

> I guess the place where I found this to be a rather large bottle neck was trying to load large (2-3 TB) sized binary files. In that case slicing (tiling?) the data along both the x,y, time [...]

Oh yeah, especially at that size, having a traditional HDD and adversarial access patterns really hurt. I would really suggest to at least get SATA SSD storage in that case!

But anyways, this is getting a bit off-topic from the original issue :)

uellue, 2024-09-24:
> Also I believe the `np.array()` function is essentially a copy in that it casts it to an in memory array. (correct?)

Yes, `np.array()` copies by default unless instructed otherwise, while `np.asarray()` only copies if needed, by default.

Regarding the discussion on IO backends, a good LiberTEM file reader implements three reading protocols:

* Memory mapping. On Linux this gives the best performance for files that fit the file system cache.
* Buffered I/O. This is mostly necessary on Windows since memory mapping absolutely tanks on a HDD there, like @sk1p already wrote. We are speaking 1 MB/s instead of 100 MB/s. Linux performs an aggressive and effective read-ahead, so you get pretty much the HDD performance there.
* Direct I/O for files that are too large for the file system cache. This is absolutely necessary to get the full performance of today's SSDs on huge files, since the system will start thrashing from futile attempts to keep the data in the cache. Here we are talking about less than 1 GB/s creating heavy system load vs. maxing out a SSD RAID with ease and spending most time waiting on the SSDs or performing useful computations.

As @sk1p wrote, all three reading protocols should support reading small slices to work with our tiling mechanism, and they should also communicate what tile shapes they can read efficiently so that LiberTEM can make smart choices for the tiling scheme.

We do have readers that don't implement all interfaces, but they are to be considered second class citizens because their performance is fundamentally limited. 

In conclusion, the current file readers in RosettaSciIO can't replace the ones in LiberTEM. The other direction would probably be easier.

I'd be very happy to see a more universal implementation, and [we've already worked on a lazy array concept](https://github.com/hyperspy/hyperspy/issues/2784#issuecomment-872977598) that could be the basis for this. We just didn't have the bandwidth to tackle this and our current readers work just fine, mostly, so no pressing need. Essentially, one would need a library that can reshape, slice and stack virtual arrays without actually reading the data, and for each file format an implementation to read such sets of slices from the actual files. A bit similar to Dask in concept, but capable of efficiently handling many small chunks and virtually slicing and stacking instead of handling actual arrays.

If this interface can efficiently extract arbitrary subsets of a dataset and is, as far as possible, [independent of the actual I/O method](https://sans-io.readthedocs.io/how-to-sans-io.html), it could be a universal reader that works as a basis for both LiberTEM and Dask/array based frameworks.

General structure:

* `lazyslice`: Abstract array handling code, producing array index read ranges, universal
* `fileformat`: Convert array index ranges to I/O ranges in the files, format-specific
* `iobackend`: Read these I/O ranges as instructed, universal
* `fileformat`: extract and convert the desired data from the I/O snippets, format-specific

Fine points will be handling page alignment for Direct I/O, parallel async I/O, and merging I/O ranges where the gaps are small.

Probably it should be written in Rust to allow working with many small slices efficiently. Are there people in RosettaSciIO who are comfortable with Rust or would like to be in the future? :-)