LiberTEM/LiberTEM #1701: Improve partitioning of "sig-heavy" data

sk1p, 2025-02-21:
Sig-heavy meaning here data that has shapes like `(10, 4096, 4096)` instead of the usual 4D-STEM shapes where we have many more scan points. Right now, working on such a data set gives a message like the following:

```
[...]/site-packages/libertem/io/dataset/base/partition.py:75: RuntimeWarning: dataset contains fewer frames than partitions, setting num_part to 1 to allow processing (use fewer workers?)
  warnings.warn(
```

For reference, this was introduced in https://github.com/LiberTEM/LiberTEM/pull/1337 

As we have quite a lot of work to do in some cases per-frame, that means we are missing out in parallelization here; it could be 10x faster at least (maybe even more if we have mostly GPU workers). In other cases, we may have little work per-frame, meaning splitting "small" partitions into even smaller ones results in more overheads.

There are different approaches:

1) Have a tweakable (maybe when loading data) to let the user influence the partitioning (fixes the issue for advanced users)
2) By default, go the other way and split the data into single-frame partitions (set [`num_partitions=num_frames` instead of `num_partitions=1`](https://github.com/LiberTEM/LiberTEM/pull/1337/files#diff-e4757907cb1d6345ded30677b85509ddeea92ea5be743c62b48f0945e37f4c36R70-R76))
3) Have a heuristic decide this
4) Have the UDF communicate this (needs a new mechanism similar to tile shape negotiation)

For the short term, I'm tending towards 2), and maybe with a later addition of 1) to allow to override this decision.

sk1p, 2025-02-21:
I'll actually go ahead with a minimal PR for option 2 for discussion and possibly benchmarking of overheads etc.

matbryan52, 2025-02-26:
I think I chose the `num_partitions = 1` because it followed the existing heuristic that a partition should be around 512 MB, so even if you have many workers it didn't make sense from an overheads perspective to split a small dataset up beyond this. In the case of an image stack the existing heuristic is not the right one, and can only really be decided from the work that needs to be done on the data. I can imagine in some cases that setting `num_partitions == num_workers` is optimal, too !

The reality is that any 4D-STEM dataset which was small enough to run into the `num_partitions = 1` branch is trivially quick to compute on, anyway, so it doesn't really matter that this will now launch `num_frames` tasks instead. You'll have to wait for the overhead but this should be a few seconds maximum, then the trivial processing time.

I do still think that having a `num_partitions` override somewhere would be useful, for when our existing heuristic fails. More than once I've found myself monkeypatching a dataset like `ds.get_num_partitions = lambda *x: 4` in order to achieve something specific, I've always found it a little odd that there was no way to do this officially.

sk1p, 2025-03-12:
This is now done via #1702

closed 2025-03-12 by sk1p